esphome/issues #5265: Touch screen binary sensor button locations are no longer screen rotation relative

ThatSlacker, 2023-12-21:
### The problem

Previously when used with a touch screen the binary sensor component would define X and Y based on the rotation of that touch screen.  Now even though the display is rotated, the binary sensor is not.  

I have items rendered to the screen in the location where binary sensors should be and while those items appear correct, pressing them produces the wrong behavior.

### Which version of ESPHome has the issue?

2023.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

ESP32-2432S028R

### Component causing the issue

touchscreen

### Example YAML snippet

```yaml
display:
  - platform: ili9xxx
    id: maindisplay
    model: ili9341
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 90
    lambda: |-
      it.fill(id(Color::BLACK));
      it.printf(20, 0, id(arimo15), "Effect: %s", id(effect_value).c_str());
      it.filled_circle(168, 51, 23, id(color_blue));
      it.filled_circle(168, 103, 23, id(color_red));
      it.filled_circle(168, 155, 23, id(color_yellow));
      it.filled_circle(168, 207, 23, id(color_green));
      it.print(55, 63, id(mdi_large), TextAlign::CENTER, "󰛨"); // Light on
      it.print(55, 167, id(mdi_large), TextAlign::CENTER, "󰹏"); // Light off

      it.print(258, 53, id(mdi_medium), TextAlign::CENTER, "󰸰");  // Fireworks
      it.print(258, 105, id(mdi_medium), TextAlign::CENTER, "󰙴"); // Twinkle
      it.print(258, 157, id(mdi_medium), TextAlign::CENTER, "󰈸"); // Fire
      it.print(258, 209, id(mdi_medium), TextAlign::CENTER, "󰐅"); // Christmas

touchscreen:
  platform: xpt2046
  display: maindisplay
  spi_id: touch
  cs_pin: GPIO33
  interrupt_pin: GPIO36
  update_interval: 50ms
  threshold: 400
  calibration_x_min: 3890
  calibration_x_max: 210
  calibration_y_min: 170
  calibration_y_max: 3730

binary_sensor:
  - platform: touchscreen
    name: LED On
    x_min: 12
    x_max: 117
    y_min: 28
    y_max: 126
    filters:
      - delayed_on_off: .2s

  - platform: touchscreen
    name: LED Off
    x_min: 12
    x_max: 117
    y_min: 132
    y_max: 230
    filters:
      - delayed_on_off: .2s

  - platform: touchscreen
    name: Blue
    x_min: 123
    x_max: 212
    y_min: 28
    y_max: 74
    filters:
      - delayed_on_off: .2s

  - platform: touchscreen
    name: Red
    x_min: 123
    x_max: 212
    y_min: 80
    y_max: 126
    filters:
      - delayed_on_off: .2s
```

### Additional information

This worked up until the 2023.12.0 update.  I'm assuming this was changed as a part of the touchscreen update (https://github.com/esphome/esphome/pull/4596) but I don't see anything in the notes about needing to update binary sensor positions

Zoogara, 2023-12-21:
Yeah the documentation hasn't been updated, but I think there is a new option `swap_xy: true` which may solve it.

Mentioned in #5251

ThatSlacker, 2023-12-22:
I should have mentioned that I tried that one already.  Setting `swap_xy : true` by itself causes the upper left corner of the screen (rotated upper left that is) to blink on and off without fixing the issue.  

If I combine that with `mirror_x: true` I get a blinking corner, 3/4 of my screen is rendered upside down, and the problem persists.

If I further add `mirror_y: true` then I get the left portion of my screen duplicated across the right portion and rendered upside down with no change to the buttons.

Dropping `mirror_x: true` leaves the duplication and the blinking but now the duplicated portion of the screen is rendered right side up.  The buttons remain unchanged still.

Going down to only `mirror_y: true` rotates the display by 180 degrees.  It doesn't have any of the odd behavior but it is now 180 degrees out of phase with what the button placement is.  

Setting all three values to false renders everything at the default screen orientation - but flipped along the X axis.  

Moving to only  `mirror_x: true` removes all screen rotation.  The right-most set of buttons are rendered off the right side of the screen but now that the screen is no longer rotated the buttons that do appear on screen finally work.  This just mostly demonstrates that the rendering and the buttons are no longer connected.

ThatSlacker, 2023-12-22:
I'm now realizing that you were attaching that property to `touchscreen` and I was attaching that rotation to `display`.  

Attaching it to the correct property I can get the buttons to be upside down from what they should be.  If I combine that with a rotation of 270 I can get to a "working" state by flipping my display upside down.  But similar to guillempages in the Discord channel (https://discord.com/channels/429907082951524364/1119046373543780363/1185591149377900664) I can't seem to find a combination of options that gets me the "correct" state.

Zoogara, 2023-12-22:
Yeah - `mirror_x:` and `mirror_y:` for the display is truly borked...

I followed the documented instructions for 180 degree rotation in hardware and got similar gibberish to you tests.

And yes - sorry was referring to `touchscreen:` for the `swap_xy:` options, although it isn't in the documentation, just in that issue I linked.

ThatSlacker, 2023-12-22:
Thanks for the workaround!  Fortunately my display can be flipped upside down (thankfully I haven't mounted it to anything) so that one gets me up and running.  Hopefully for anyone that can't do that there's a real fix soon.

nielsnl68, 2023-12-22:
Correct, atm you need to set them separately from each other. We working on a solution that it will make it possible again.

ThatSlacker, 2023-12-23:
Just to make sure that it's clear - setting them separately does not fix the issue.  By setting them separately **and** turning the screen upside down I can work around the issue.  But anyone with a fixed screen (e.g. attached to a wall) is currently broken with no way of fixing it.  The workaround would render their display upside down.

If combining them together allows for 90 degree rotations again then we're fine and you can ignore this. Just wanted to make sure the right things are being tested in the fix.

nielsnl68, 2023-12-23:
@ThatSlacker could you send me the whole YAML script that needs to setup the display so i can try it here.

ThatSlacker, 2023-12-23:
```
substitutions:
  devicename: shopledcontrol
  networkname: shopledcontrol
  name: Shop LED control panel

globals:
  - id: wifi_iconstring
    type: std::string
    restore_value: no
    initial_value: '"󰤮"'
  - id: brightness_value
    type: std::string
    restore_value: no
    initial_value: '"0"'
  - id: effect_value
    type: std::string
    restore_value: no
    initial_value: '"None"'
  - id: recent_touch
    type: bool
    restore_value: no
    initial_value: "true"

esphome:
  name: $devicename
  friendly_name: $name
 
# The ESP32-2432S028R uses a standard ESP32-WROVER, so we use the esp32dev defintion
esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${devicename}
    password: !secret ap_wifi_password

captive_portal:

logger:
  baud_rate: 0  # (UART logging interferes with cse7766)

api:

ota:

text_sensor:
  - platform: version
    name: ${name} ESPHome Version
  - platform: wifi_info
    ip_address:
      name: ${name} IP Address
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# ============================================================ 
# ESPHome Display related setup
#
# Create a font to use, add and remove glyphs as needed. 
font:
  - file: 'fonts/materialdesignicons-webfont.ttf'
    id: mdi_large
    size: 48
    glyphs: [
[... 265 lines cut ...]
        - light.turn_on: backlight
        - lambda: |-
            id(recent_touch) = true;
        - script.execute: update_ui_values

text_sensor:
  - platform: homeassistant
    id: shopledeffect
    entity_id: light.shop_led_strip
    attribute: effect
  - platform: homeassistant
    id: shopledbrightness
    entity_id: light.shop_led_strip
    attribute: brightness

sensor:
  - platform: wifi_signal # Reports the WiFi signal strength in %
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    id: wifi_signal_pct
    update_interval: 60s
    entity_category: "diagnostic"
```

nielsnl68, 2023-12-23:
when i use the following setting on my ili9341 display the touchscreen works.
As you can see i replaced `rotation:` for `transform` in the display component.
Made sure that the values of the calibration settings are `calibration_x_min: 210` and  `calibration_x_max: 3890`
And use also the `transform` settings to mirror the y position. Because on my display the 0,0 position of the display is mirrored on the touchscreen.
For testing i show a yellow dot at the position where you touch the display.

```yaml
display:
  - platform: ili9xxx
    id: my_display
    model: ili9341
    cs_pin: 5
    dc_pin: 12
    reset_pin: 35
    auto_clear_enabled: false
    data_rate: 40MHz
    dimensions: 320x240
    transform:
      mirror_y: false
      mirror_x: false
      swap_xy: true

    lambda: |-
      it.fill(id(Color::BLACK));
     
      it.printf(20, 0, id(arimo15), "Effect: %s", id(effect_value).c_str());
      it.filled_circle(168, 51, 23, id(color_blue));
      it.filled_circle(168, 103, 23, id(color_red));
      it.filled_circle(168, 155, 23, id(color_yellow));
      it.filled_circle(168, 207, 23, id(color_green));
      it.print(55, 63, id(mdi_large), TextAlign::CENTER, "󰛨"); // Light on
      it.print(55, 167, id(mdi_large), TextAlign::CENTER, "󰹏"); // Light off

      it.print(258, 53, id(mdi_medium), TextAlign::CENTER, "󰸰");  // Fireworks
      it.print(258, 105, id(mdi_medium), TextAlign::CENTER, "󰙴"); // Twinkle
      it.print(258, 157, id(mdi_medium), TextAlign::CENTER, "󰈸"); // Fire
      it.print(258, 209, id(mdi_medium), TextAlign::CENTER, "󰐅"); // Christmas
      it.print(296, 0, id(mdi_small), id(wifi_iconstring).c_str());

      auto touch = id(my_toucher)->get_touch();
      if (touch.has_value()) 
        it.filled_circle(touch.value().x, touch.value().y, 10, color_yellow);

touchscreen:
  - platform: xpt2046
    display: my_display
    id: my_toucher
    update_interval: 50ms
    cs_pin: 18
    #interrupt_pin: 5
    threshold: 300
    calibration_x_min: 210
    calibration_x_max: 3890
    calibration_y_min: 170
    calibration_y_max: 3730
    transform:
      mirror_x: false
      mirror_y: true
      swap_xy: true
```

I hope this helps.