esphome/issues #5266: Changed sensor name after update

heggi, 2023-12-22:
### The problem

My temperature sensor changed the name from `esphome_temperature_test_temperature_1` to `esphome_web_394f6f_temperature_1` after firmware update to 2023.12.1
Firmware's sensor name stays the same (`esphome_temperature_test_firmware`)
Other ESP devices didn't change any sensor's names.

### Which version of ESPHome has the issue?

2023.12.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1_mini

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-394f6f
  friendly_name: ESPHome Temperature Test

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxx"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:
    
dallas:
  - pin: D4
    update_interval: 10s

sensor:
  - platform: dallas
    index: 0
    name: "Temperature #1"
```

tsgwiro1, 2023-12-22:
I have the same Issue. The issue cam with 2023.12.0

I have several ESPHome devices that did experience the mentioned change.

Examples:
`number.wp_solar_monitor_spez_dichte` changed to `number.wp_solar_monitor_spez_dichte_2`
`sensor.wp_solar_monitor_rucklauf` changed to `sensor.wp_solar_monitor_r_cklauf`
`sensor.wp_solar_monitor_pump_last_turn_on_2` changed to `sensor.wp_solar_monitor_pump_last_turn_on`
`switch.wp_zwe2_controller_main_power_on_off` changed to `switch.wp_zwe2_controller_main_power_on_off_2`

As you can see several typs sensor / switches / numbers are affected.

snipplets of the configuration
`switch:
  - platform: template
    name: "Main Power On/Off"
    id: zwe2_main_on_off
    restore_mode: RESTORE_DEFAULT_ON
    optimistic: true
    device_class: SWITCH`

`number:
  - platform: template
    name: "spez. Dichte"
    optimistic: true
    restore_value: true
    unit_of_measurement: "kg/m3"
    id: sdi
    entity_category: "config"
    mode: box
    min_value: 960
    max_value: 1060
    step: 1.0`

`sensor:
  - platform: dallas
    address: 0x43000000196fe828
    name: "Rücklauf"
    id: rl
    resolution: 10
    filters:
      - sliding_window_moving_average:
          window_size: 2
          send_every: 2`

OGB1952, 2023-12-22:
I have a few dozen entities that got broken in this way with 12.0.  Ended up with duplicates with different names and a lot of broken automations where the entity was shown as not available (because it had been renamed by the update).  Tried 12.1 thinking it would fix it, it didn't, so I reverted to 11.6 and all was OK (although I now have 91 orphaned entities).
For some reason 12.2 updated itself early this morning, woke up to the same problem, lots of entities showing as not available, same name change problem, same entity duplication problem.
Reverted to 11.6 again, updated all and everything is back as it was, but there are still large numbers of orphaned entities that I need to remove once I'm sure things aren't going to break again.
All I can suggest is sticking with 11.6 until such time as the issue is resolved, perhaps with 12.3?

tsgwiro1, 2023-12-22:
What would be the way to revert to an older version?

OGB1952, 2023-12-22:
Go to "system", then "backups".  Find the last ESPHome 11.6 backup and click it.  In the window that pops up select the two tick boxes then click "Restore".  It will take a few minutes to re-install version 11.6.
When this has finished, go to the ESPHome and click "Update all" at the top left.  It will take some time to complete as it has to compile and load all the restored ESPHome code to every device.  When completed you should be back where you were, but the entity list will have lots of orphaned entities created by the version 12.0 etc mess up.  For now I'm leaving these but will clean them up once there looks to be a fixed version of ESPHome available.

tsgwiro1, 2023-12-22:
Great thank you very much, that worked!

OGB1952, 2023-12-22:
Glad you're back running again.  I suggest ignoring ESPHome updates until such time as the this gets fixed and tested properly.  The problem seems to be related to a breaking change in the entity naming protocol that seems to cause unforeseen side effects with some entity names.  I can't pretend to understand why, I'm a hardware person, but this is what I've managed to pick up from various discussions by those who understand this stuff.

Bloodpack, 2023-12-22:
same here, using ESP-Home as an addon in Home Assistant and all my Sensors got renamed

it happened first on 2023.12 > i fixed all the sensor names 
updated to 2023.12.3 all renamed again

example:

from `nodemcu_wohnzimmer_onboard_led` to `nodemcu_wohnzimmer_onboard_led_2` and so on......

all automations are broken again

fabiobrolo, 2023-12-22:
I'm having the exact same issue in 2023.12.0, 2023.12.1, 2023.12.2 and 2023.12.03.

lynara, 2023-12-22:
same issue here, also downgraded back to 2023.11.6

OGB1952, 2023-12-22:
I think I've found what may be the cause of this problem.  It seems that the process of dealing with special characters changes the entity unique_id.  HA uses the unique-id to reference entities, so if this changes it no longer recognises an entity, which in turn leads to the "entity not available" warning and any integration or automation that uses that entity to break.
Looking at an example in my setup, an entity that had the unique_id "esp32c3-sensor1switchbedroom_ac_on_18c" was changed during the update to 12.x to "esp32c3-sensor1switchbedroom_ac_on_18_c" (with an extra underscore before the "c").
HA then treats this as a new entity, one that isn't used by anything, hence the apparent duplicates in the entity list and the broken integrations.

gednet, 2023-12-23:
I have the same issues. All of entities with characters used in polish language have different entitie names after update to version 2023.12.3. For example:

in ESPHome (device_name: esp32-salon)
sensor humidity
**name: "Obsługa salonu wilgotność"**

entitie name in version 2023.11.6
**sensor.esp32_salon_obsuga_salonu_wilgotno**

entitie name in version 2023.12.3
**sensor.esp32_salon_obs_uga_salonu_wilgotno**

In some entities a had "_2" at end of the name.

lynara, 2023-12-23:
For me this is happening for these entities:

```
name: "Akt. Leistung"
id: akt_leistung

name: "Akt. Leistung Heizstab"
id: heizstab_power

name: "Akt. Leistung Heizung"
id: heizung_power
```
After updating to 12.x the entities get suffixed with "_2" eventhought they have an id specified. 
I suspect it is because of the dot in the name because all the other entities without any special chars are fine.

Sab3rRid3r, 2023-12-24:
Same here some Sensors changed like
from 2023.11.6 
wpc05_wm_elektr_warmwasser_total

to 2023.12.3 
wm_elektr_warmwasser_total

OGB1952, 2023-12-24:
I found this quite a lot of work to fix, even after doing the "delete the config and then restore" process.  This process only partly worked, some entities did seem to get associated with their original functions, many did not.  I ended up (after a lot of editing) with 91 re-named entities that appear to be working, plus 91 old entities with their original names that were no longer working.  Some of the re-named entities worked OK without me having to do anything, most didn't and I had to edit integrations and automations to get them to work again.  Took me around 4 hours yesterday morning to get back to a working HA system after the havoc wreaked by this update.
At the moment I've not deleted all the duplicated entities, just hidden them, in case there is a problem I've not yet spotted.

derekpurdy, 2023-12-27:
i had a change with an extra underscore for all my devices
example:
"sump_pump_power_monitor_current"
became
"sump_pump_power_monitor__current"
for mqtt topics

lynara, 2024-02-29:
has this ever been addressed? haven't update since...

OGB1952, 2024-03-01:
> has this ever been addressed? haven't updated since...

The only real fix is to manually edit everything affected by the forced entity name changes I believe.  Can take some time, certainly wasted a few hours of my time getting things back so they worked.

lynara, 2024-03-01:
Hi, so I updated from 2023.11.6 to 2024.2.1 just now and didn't have the issue anymore.
Seems like it got addresses somewhere down the road.