esphome/issues #5268: Entities dissapeared after 5810 pull

Csontikka, 2023-12-22:
### The problem

https://github.com/esphome/esphome/pull/5810

A lot (switch) entities comes from the esphome has been disappeared. Not renamed, disappeared!
Maybe the problem with the string sanitizing procedure does not handle right all of the used characters
The affected entities contains hungarian characters.
Entities not containing hungarian characters works as expected

example yaml in details:

The special chars:
öőúáűüóíÖÜÓŐÚÉÁŰÍ

### Which version of ESPHome has the issue?

2023.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.1

### What platform are you using?

ESP8266

### Board

esp32,

### Example YAML snippet

```yaml
- platform: gpio
    pin: GPIO27
    name: Csepegtet**ő** (elől)
    id: relay5
    on_turn_on:
    - script.execute: script_relay5
    on_turn_off:
    - script.stop: script_relay5
```

### Anything in the logs that might be useful for us?

```txt
in HA log there is nothing relevant
```

OGB1952, 2023-12-22:
I have a similar problem but when looking at the entity list I found they had been duplicated and the new versions had a completely different name.  This causes any integration using them to display the "entity not available" message, as the integration is looking for the old name and won't automatically pick up the new name.
For example, one of the (dozens) of entities broken by updating from 11.6 (to either 12.0, 12.1 or 12.2) used to be named  switch.bedroom_ac_on_18degc
Following the 12.0 (or 12.1 or 12.2) update this changed to switch.esp32c3_sensor1_bedroom_ac_on_18_c
I have dozens of entities that have been renamed like this and the only way I could get back to a working HA system was to revert to version 11.6 and then "update all" .  This fixes all the broken integrations but has left 91 orphaned entities, all those new ones created when ESPHome was updated.

neutrino220, 2023-12-25:
I have the same since 2023.12.0
[image]