esphome/issues #5269: Sonoff mini default config not compile

txitxo0, 2023-12-22:
### The problem

Using the provided [config in docs for sonoff mini](https://devices.esphome.io/devices/Sonoff-Mini-Relay#basic-config), there is an error regarding the pin 13. "Pin 13 is used in multiple places"

So, the proble should be in here:
```
status_led:
  pin:
    number: GPIO13
    inverted: true

output:
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True

light:
  # the 4 lines below define the Blue LED light on Sonoff Mini, to expose in HomeAssistant remove line "internal: true"
  - platform: monochromatic
    name: ${device_name}_blueled
    output: blue_led
    internal: true # hides the Blue LED from HomeAssistant
    ```

![image](https://github.com/esphome/issues/assets/12279324/d6b2424f-15dc-4640-a210-d3f765418552)

### Which version of ESPHome has the issue?

2023.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP8266

### Board

Sonoff mini

### Component causing the issue

esp8266_pwm

### Example YAML snippet

```yaml
status_led:
  pin:
    number: GPIO13
    inverted: true

output:
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True

light:
  # the 4 lines below define the Blue LED light on Sonoff Mini, to expose in HomeAssistant remove line "internal: true"
  - platform: monochromatic
    name: ${device_name}_blueled
    output: blue_led
    internal: true # hides the Blue LED from HomeAssistant
```

### Anything in the logs that might be useful for us?

```txt
Pin 13 is used in multiple places
```

Codex-, 2023-12-28:
https://github.com/esphome/esphome-devices/commit/16cd2a2d4ee144a4076379f55a62bdecb6d321ba

config has been updated it appears

ssieb, 2023-12-28:
Yes, unfortunately...  That config doesn't even make sense.

closed 2023-12-28 by ssieb

Codex-, 2023-12-29:
@ssieb oh? Is it superceded by something else? What would you recommend?

rcruikshank, 2024-02-11:
I have the same issue.  But simply commented out the status_led section. Seemed to work and I never see the status led anyway.