esphome/issues #5270: Unable to compile LED controller

jehy, 2023-12-22:
### The problem

Component yaml:

```yaml
substitutions:
  name: "led-controller-1-ws2812b"
  friendly_name: "Athom LED Controller"
  project_name: "athom.ls-4p-3wire"
  project_version: "1.1"
  button_toggle: "true"
  led_restore_mode: RESTORE_DEFAULT_OFF
  number_of_leds: '60'
  led_rgb_order: GRB
  led_chipset: WS2812B

esp8266:
  board: esp8285
  restore_from_flash: true
  framework:
    version: 2.7.4

api:

ota:

logger:
  baud_rate: 0

web_server:
  port: 80

captive_portal:

binary_sensor:
  - platform: status
    name: "${friendly_name} Status"

  - platform: gpio
    pin:
      number: 0
      mode: INPUT_PULLUP
      inverted: true
    name: "${friendly_name} Power Button"
    disabled_by_default: true
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at least 0.2s
        then:
          - light.toggle: leds
      - timing:
          - ON for at least 4s
        then:
          - button.press: Reset

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime"

button:
  - platform: factory_reset
    name: Restart with Factory Default Settings
    id: Reset

power_supply:
  - id: relay
    pin: GPIO12

wled:
[... 156 lines cut ...]
  
  This feature requires framework version 2.7.4 or lower. Please see note on documentation for FastLED.
```

Never happened before 2023.12

### Which version of ESPHome has the issue?

2012.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2012.12

### What platform are you using?

ESP8266

### Component causing the issue

light.fastled_clockless

jehy, 2023-12-22:
Seems like build with 2.7.4 was fixed in https://github.com/esphome/esphome/pull/4777 but was broken in 2023.12

jehy, 2023-12-22:
Okay, I tried neopixel and I suddenly discovered that it took me two minutes to migrate to it, that's almost a drop in replacement. Can we somehow make it more obvious in compile error messages?

jehy, 2023-12-22:
[image]
By the way, docs say that using old framework should still work, but that's not the case.

DefenestratingWizard, 2024-09-10:
still an issue on 2024.8.3, can attach logs if needed

jerano, 2024-11-02:
Still the same problem! And please stop recommending other components, my LED strip is not supported by them.

Why can't the FastLED component be upgraded to something newer than v3.3.2 ?
I would assume the issue with RMT that prevented esphome from using v3.3.3 would have been fixed by now when they're at v3.9.2 ?!