esphome/issues #5274: Touchscreen tt21100 no longer working after 2023.12.3 update

woodshoes, 2023-12-22:
### The problem

Upgraded from 2023.11.6 to 2023.12.3 on my ESP32-S3-BOX (the old one) 
My touchscreen does not work - the screen or the home button directly below.  
Reverted back to 2023.11.6 and it started working again.  I know there is a breaking change for the touchscreen component as listed [here](https://github.com/esphome/esphome/pull/4596) but it doesn't look like I need to make any changes from my config?
Nothing fancy going on for the box, just using it as a display with some touchscreen buttons to turn office lights on/off.

### Which version of ESPHome has the issue?

2023.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

ESP32-S3-BOX

### Component causing the issue

touchscreen

### Example YAML snippet

```yaml
---
esphome:
  name: flexbox
  friendly_name: ESP32 S3 Box
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
    version: "1.0"
  min_version: 2023.10.5
  includes:
    - common.h

esp32:
  board: esp32s3box
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG
  logs:
    component: ERROR

api: 
  encryption:
    key: 3IzrqZs4csVkrt7/lmr238Z+HRP/vAZCbgqGCOe8zsQ=

ota:
  password: !secret ota_password

dashboard_import:
[... 381 lines cut ...]
    id: home_pressed
    entity_id: input_boolean.flexbox_home_pressed
    internal: True    

image:
  - file: "images/Graphic-Design-Background-320x240.jpg"
    id: background_image
    resize: 320x240
    type: rgba

time:
  - platform: homeassistant
    id: time_ha
    timezone: America/Chicago

psram:
  mode: octal
  speed: 80MHz
```

### Anything in the logs that might be useful for us?

```txt
Even with VERY_VERBOSE logging turned on (and removing the component: ERROR) I see nothing in the logs for touchscreen events.
```

nielsnl68, 2023-12-22:
Sadly i do not own a display with the tt21100 controller. 
Would you help me out and create a small script with only the `display` and the `touchscreen` ?

and change the yaml code for the touchscreen in:

```yaml
esphome:
  name: flexbox
  friendly_name: ESP32 S3 Box
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
    version: "1.0"
  min_version: 2023.10.5

esp32:
  board: esp32s3box
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG

api: 
  encryption:
    key: 3IzrqZs4csVkrt7/lmr238Z+HRP/vAZCbgqGCOe8zsQ=

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  manual_ip:
    static_ip: 192.168.25.74
    gateway: 192.168.25.254
    subnet: 255.255.255.0
    dns1: 192.168.1.210

psram:
  mode: octal
  speed: 80MHz

spi:
  clk_pin: GPIO7
  mosi_pin: GPIO6

# some HW Components of the S3-Box
i2c:
  scl: GPIO18
  sda: GPIO8
  scan: true

display:
  - platform: ili9xxx
    model: S3BOX
    cs_pin: GPIO5
    dc_pin: GPIO4
    reset_pin: GPIO48
    id: lcd

touchscreen:
  platform: tt21100
  id: my_touchscreen
  # address: 0x24
  interrupt_pin: GPIO3
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "id=%d, x=%d, y=%d, x_raw=%d, y_raw=%0d",
              touch.id,
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );     
  on_release:
    - lambda: |-
          ESP_LOGI("cal", "touches released",
```

Can you show the result of the log information when you touch and release the screen at the 4 corners?

thanks

woodshoes, 2023-12-22:
Hello
I had to "finish" your yaml above - not sure if you wanted more in it?  I made the last section read:

  on_release:
    - lambda: |-
          ESP_LOGI("cal", "touches released"
          );

After that, it flashed OK - but there is NOTHING in the logs about touchscreen events.  Attached  is a log file I took over USB - and I was touching all over that screen with no result.

Wireless flash of YAML from esphome 2023.12.3
[logs_flexdisplay_run.txt](https://github.com/esphome/issues/files/13757049/logs_flexdisplay_run.txt)

USB logs
[flexdisplay_logs (2).txt](https://github.com/esphome/issues/files/13757034/flexdisplay_logs.2.txt)

and the new YAML
[flexdisplay.txt](https://github.com/esphome/issues/files/13757056/flexdisplay.txt)

nielsnl68, 2023-12-22:
can you try using the same with the Arduino framework?

woodshoes, 2023-12-23:
Done.  Had to remove a few lines for it to compile - but it is the exact same result.  Nothing in the logs for touch events. 
Logs after reboot [flexdisplay_logs (3).txt](https://github.com/esphome/issues/files/13757465/flexdisplay_logs.3.txt)
New yaml 
[flexdisplay-arduino.txt](https://github.com/esphome/issues/files/13757470/flexdisplay-arduino.txt)

nielsnl68, 2023-12-23:
is this really all what you get in your log? I see only setup part but not the rest of the log.

clydebarrow, 2023-12-23:
Change log level to VERBOSE and see what appears in the log.
```
logger:
  hardware_uart: USB_SERIAL_JTAG
  level: VERBOSE
```

clydebarrow, 2023-12-23:
Hmmm, looks like this was removed from the original code:

```
  // Trigger initial read to activate the interrupt
  this->store_.touch = true;
```
I infer from this that it is necessary to read the chip once before any interrupts are fired.

nielsnl68, 2023-12-23:
> // Trigger initial read to activate the interrupt
>   this->store_.touch = true;

okay, restored this lines.

woodshoes, 2023-12-23:
OK - not sure what to do now, do you still want to see logs w/ VERBOSE?  It doesn't show touch events, not even with VERY_VERBOSE.  How can I test the "this->store_.touch = true;"
Thanks

clydebarrow, 2023-12-24:
Add to your yaml:

```
external_components:
  - source: github://pr#6008
    components: [ tt21100 ]
```

That will pull in the fix. Report back and if that fixes it I will merge the PR.

```

woodshoes, 2023-12-26:
OK This has made progress but it still is not working correctly.  After pulling in the PR, the touch screen only works once.  It never registers again after the first touch.  
If I press the "home" button _before_ I touch the screen, it works _repeatedly_ as shown in the logs here 
[flexdisplay_logs (4).txt](https://github.com/esphome/issues/files/13771102/flexdisplay_logs.4.txt)

But once I touch the display anywhere (besides the home button) it logs once but then no touches register again, not even a home button press.  
[flexdisplay_logs (5).txt](https://github.com/esphome/issues/files/13771110/flexdisplay_logs.5.txt)

And not sure if y'all want to see the yaml but its here 
[flexdisplay (1).txt](https://github.com/esphome/issues/files/13771115/flexdisplay.1.txt)

Thanks everyone!

woodshoes, 2023-12-26:
And just noticed that the PR was pulled in - upgraded to 2023.12.5 (removed the PR pull from yaml) - same results.  Home button works until one screen touch - then it doesn't work any more AND the touch screen does not respond or log any  more events.

nielsnl68, 2023-12-26:
Do you have any means to monitor the pin GIO3, somehow? I like to know how this pin behaves while you touch the screen.

woodshoes, 2023-12-26:
Off the top of my head no, but could try to figure it out.  What would I need?

nielsnl68, 2023-12-26:
You could start simply with a LED and a resister

woodshoes, 2023-12-27:
Welp - looks like I wont be able to be much help on the hardware troubleshooting side.  This device is already assembled with the screen and such (ESP32-S3-BOX).  I got it partially apart but didnt see any pinout access for GPIO3.  Anything else we can try?  It works perfectly in 2023.11.6.

nielsnl68, 2023-12-27:
okay np, thanks for looking into it anyway.

tommck, 2023-12-31:
I have one of these that I'm trying to setup for the first time.  Should I jump in here?

clydebarrow, 2023-12-31:
I'm going to have another look at this and will need testers - don't have the S3Box (only the -3 which is a different touch chip) to test on here. Stand by.

cptskippy, 2024-01-26:
I think I am experiencing the same issue as @woodshoes.

> After pulling in the PR, the touch screen only works once.

The binary sensor works perfectly...

```
i2c:
  sda: GPIO8
  scl: GPIO18
  scan: true

touchscreen:
  - platform: tt21100
    address: 0x24
    interrupt_pin: GPIO3
    on_touch:
      - logger.log: "Touchscreen:: Touched"

binary_sensor:
  - platform: tt21100
    index: 0
    name: "Home"
```

...UNTIL you touch the touchscreen.  As soon as you touch the touchscreen the on_touch event fires and then the tt21100 stops responding.

```
[13:24:33][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:33][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:36][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:36][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:37][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:37][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:37][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:38][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:38][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:38][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:39][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:39][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:39][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:40][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:40][D][binary_sensor:036]: 'Home': Sending state ON
[13:24:40][D][binary_sensor:036]: 'Home': Sending state OFF
[13:24:41][D][main:396]: Touchscreen:: Touched
```

> Do you have any means to monitor the pin GIO3, somehow? I like to know how this pin behaves while you touch the screen.

@nielsnl68 I looked at [the schematic for the board](file:///C:/Users/peter/Downloads/PCB_ESP32-S3-BOX_V2.4_20210916.pdf) and there is [a test pad for GPIO3 that's accessible](https://i.imgur.com/To0ktnU.png).  I probed it while using the device and it's high (3.3v) normally and when pressing the binary sensor, but as soon as the touchscreen is touched it drops low (0v) and stays there.

When the device is power cycled or reset then GPIO3 returns to high.

@clydebarrow let me know if there's any other tests you'd like me to try.

nielsnl68, 2024-01-26:
could you try this as external_components:  https://github.com/esphome/esphome/pull/5997

This should have the fix.

cptskippy, 2024-01-26:
> could you try this as external_components: [esphome/esphome#5997](https://github.com/esphome/esphome/pull/5997)

I tried this:

```
external_components:
  - source: github://pr#5997
    components: [ tt21100 ]
    #components: [ touchscreen ]
```

And it fails to build:
```
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/voice-assistant-1/src/esphome/components/tt21100/touchscreen/tt21100.o
Compiling .pioenvs/voice-assistant-1/src/main.o
src/esphome/components/tt21100/touchscreen/tt21100.cpp: In member function 'virtual void esphome::tt21100::TT21100Touchscreen::setup()':
src/esphome/components/tt21100/touchscreen/tt21100.cpp:67:42: error: 'class esphome::display::Display' has no member named 'get_native_width'; did you mean 'get_width'?
       this->x_raw_max_ = this->display_->get_native_width();
                                          ^~~~~~~~~~~~~~~~
                                          get_width
src/esphome/components/tt21100/touchscreen/tt21100.cpp:70:42: error: 'class esphome::display::Display' has no member named 'get_native_height'; did you mean 'get_height'?
       this->x_raw_max_ = this->display_->get_native_height();
                                          ^~~~~~~~~~~~~~~~~
                                          get_height
*** [.pioenvs/voice-assistant-1/src/esphome/components/tt21100/touchscreen/tt21100.o] Error 1
========================= [FAILED] Took 47.80 seconds =========================
```

clydebarrow, 2024-01-27:
> As soon as you touch the touchscreen the on_touch event fires and then the tt21100 stops responding.

Try removing the interrupt pin definition. This will make it fall back to polling. Should hopefully work until a better fix comes along.

nielsnl68, 2024-01-27:
sorry you need to add:

```yaml
external_components:
  - source: github://pr#5997
    components: [ display, touchscreen, tt21100 ]
```

woodshoes, 2024-01-27:
Had the same problem compiling as cptskippy - 
If I do:
external_components:
  - source: github://pr#5997
    components: [ display,  tt21100 ]
    
 Note no touchscreen component - it does compile, but then the touchscreen only works once. 

If I add all three like nielsnl68 says (display, touchscreen, tt21100) , it fails right away during compile.

INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/flexdisplay.yaml...
ERROR Unable to import component tt21100.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/ea27d8aa/esphome/components/tt21100/binary_sensor/__init__.py", line 7, in <module>
    from ..touchscreen import TT21100Touchscreen, TT21100ButtonListener
  File "/data/external_components/ea27d8aa/esphome/components/tt21100/touchscreen/__init__.py", line 5, in <module>
    from esphome.components import i2c, touchscreen
  File "/data/external_components/ea27d8aa/esphome/components/touchscreen/__init__.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
ERROR Unable to import component touchscreen.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1128, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/ea27d8aa/esphome/components/touchscreen/__init__.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
ERROR Unable to import component touchscreen.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1128, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
[... 154 lines cut ...]
  y_max: 160
binary_sensor.touchscreen: [source <unicode string>:89]
  
  Platform not found: 'binary_sensor.touchscreen'.
  platform: touchscreen
  name: Lamp
  x_min: 40
  x_max: 140
  y_min: 161
  y_max: 240
binary_sensor.touchscreen: [source <unicode string>:95]
  
  Platform not found: 'binary_sensor.touchscreen'.
  platform: touchscreen
  name: FanLight
  x_min: 180
  x_max: 280
  y_min: 161
  y_max: 240
touchscreen: [source <unicode string>:148]
  
  Component not found: touchscreen.
  platform: tt21100
  id: my_touchscreen
  interrupt_pin: GPIO3

nielsnl68, 2024-01-28:
can you give me the touchscreen part of your YAML? then i test it local (altrough i dont have a display with the tt21100 myself). But compiling should be okay.

nielsnl68, 2024-01-28:
Okay, just tested the PR with the Current version and found out that the PR has more changes that cant be loaded via the extended_components. (that a bummer) But i just updated the PR so it can be used with the current version.

cptskippy, 2024-01-28:
@nielsnl68 I just tried building with the latest PR and it's not working for me but seems to be a different issue than @woodshoes.

```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/voice-assistant-1.yaml...
ERROR Unable to import component touchscreen:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/ea27d8aa/esphome/components/touchscreen/__init__.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
ERROR Unable to import component tt21100.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/ea27d8aa/esphome/components/tt21100/binary_sensor/__init__.py", line 7, in <module>
    from ..touchscreen import TT21100Touchscreen, TT21100ButtonListener
  File "/data/external_components/ea27d8aa/esphome/components/tt21100/touchscreen/__init__.py", line 5, in <module>
    from esphome.components import i2c, touchscreen
  File "/data/external_components/ea27d8aa/esphome/components/touchscreen/__init__.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
INFO Detected timezone 'America/Los_Angeles'
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

touchscreen: [source <unicode string>:169]
  
  Component not found: touchscreen.
  - platform: tt21100
    address: 36
    interrupt_pin: GPIO3
    on_touch: 
      - logger.log: Touchscreen:: Touched
binary_sensor.tt21100: [source <unicode string>:316]
  
  Platform not found: 'binary_sensor.tt21100'.
  platform: tt21100
  name: Home
  index: 0
  on_press: 
    then: 
      - globals.set: 
          id: display_rotation
          value: !lambda |-
            return !id(display_rotation);
```

nielsnl68, 2024-01-28:
Can you do me a favor and add 

```yaml
external_components:
  - source: github://pr#5997
    components: [ display, touchscreen, tt21100 ]
    refresh: 0sec
```

cptskippy, 2024-01-28:
> Can you do me a favor and add
> 
> ```yaml
> external_components:
>   - source: github://pr#5997
>     components: [ display, touchscreen, tt21100 ]
>     refresh: 0sec
> ```

That allows it to compile, however it is still becoming unresponsive after one touch.  Also it is now showing a touchscreen event in the log:

```
[12:11:47][D][touchscreen:114]: Touch status: 0/7: raw:( 160,  192) calc:(   0,    0)
[12:11:47][D][main:402]: Touchscreen:: Touched
```

This is my config:

```
## BEGIN - Touchscreen
external_components:
  - source: github://pr#5997
    components: [ display, touchscreen, tt21100 ]
    refresh: 0sec    
        
i2c:
  sda: GPIO8
  scl: GPIO18
  scan: true

touchscreen:
  - platform: tt21100
    address: 0x24
    interrupt_pin: GPIO3
    on_touch:
      - logger.log: "Touchscreen:: Touched"
    on_release:
      - logger.log: "Touchscreen:: Released"
## END - Touchscreen
```

cptskippy, 2024-01-28:
> Try removing the interrupt pin definition. This will make it fall back to polling. Should hopefully work until a better fix comes along.

@clydebarrow the yaml won't compile without the interrupt, it's required for the tt21100.

nielsnl68, 2024-01-29:
My excuses, i forget to change one place where the interrupt pin was expected to be there. Now it is checked for existence.

When you still get stuck after the first touch, do you have time to dig into it with me to see where it is hanging from? I do not own a S3box so it is hard to test for me local.
 
For now please recompile and set logging to verbose. to see what happens.

cptskippy, 2024-01-29:
> When you still get stuck after the first touch, do you have time to dig into it with me to see where it is hanging from?

I'm going on holiday this week but I'll do what I can until then.

I stripped down my yaml so that has the minimum functionality:

```
substitutions:
  device_name: voice-assistant-1
  device_verbose_name: "Voice Assistant 1"

  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_hotspot_password: !secret wifi_hotspot_password

  api_encryption_key: !secret api_encryption_key
  ota_password: !secret ota_password

packages:
  device: !include templates/esp32.s3.template.yaml
  base: !include templates/wifi.template.yaml

logger:
  level: VERBOSE

psram:
  mode: octal
  speed: 80MHz

external_components:
  - source: github://pr#5997
    components: [ display, touchscreen, tt21100 ]
    refresh: 0sec

## BEGIN - Touchscreen
i2c:
  sda: GPIO8
  scl: GPIO18
  scan: true

touchscreen:
  - platform: tt21100
    address: 0x24
    interrupt_pin: GPIO3
    #reset_pin: GPIO48
    on_touch:
      - logger.log: "Touchscreen:: Touched"
    on_release:
      - logger.log: "Touchscreen:: Released"
## END - Touchscreen

## BEGIN - LCD Backlight
output:
  - platform: ledc
    id: gpio_45
    pin: GPIO45

light:
  - platform: monochromatic
    name: "LCD Backlight"
    id: lcd_backlight
    output: gpio_45
    restore_mode: ALWAYS_ON
    disabled_by_default: true
    default_transition_length: 0s
## END - LCD Backlight

## BEGIN - LCD
spi:
  clk_pin: GPIO7
[... 182 lines cut ...]
[10:45:10][D][binary_sensor:036]: 'Home': Sending state OFF
[10:45:10][D][main:202]: Home:: Released
[10:45:10][V][tt21100:095]: Button report: Len=14, ID=3, Time=39965, Value=[1], Signal=[00B4][0000][0000][0000]
[10:45:10][D][binary_sensor:036]: 'Home': Sending state ON
[10:45:10][D][main:196]: Home:: Touched
[10:45:10][V][tt21100:095]: Button report: Len=14, ID=3, Time=40115, Value=[0], Signal=[0000][0000][0000][0000]
[10:45:10][D][binary_sensor:036]: 'Home': Sending state OFF
[10:45:10][D][main:202]: Home:: Released
[10:45:11][V][tt21100:095]: Button report: Len=14, ID=3, Time=40465, Value=[1], Signal=[012B][0000][0000][0000]
[10:45:11][D][binary_sensor:036]: 'Home': Sending state ON
[10:45:11][D][main:196]: Home:: Touched
[10:45:11][V][tt21100:095]: Button report: Len=14, ID=3, Time=40565, Value=[0], Signal=[0000][0000][0000][0000]
[10:45:11][D][binary_sensor:036]: 'Home': Sending state OFF
[10:45:11][D][main:202]: Home:: Released
[10:45:12][V][tt21100:109]: Touch report: Len=17, ID=1, Time=41478, LargeObject=0, RecordNum=0, RecordCounter=0, NoiseEffect=2
[10:45:12][V][tt21100:120]: Touch 0: Type=0, Tip=0, EventId=0, TouchId=20, X=146, Y=132, Pressure=24, MajorAxisLen=0, Orientation=0
[10:45:12][D][touchscreen:114]: Touch status: 0/7: raw:( 146,  132) calc:(   0,    0)
[10:45:12][D][main:218]: Touchscreen:: Touched
[10:45:33][D][binary_sensor:036]: 'Settings': Sending state ON
[10:45:33][D][main:144]: Settings:: Pressed
[10:45:33][D][binary_sensor:036]: 'Settings': Sending state OFF
[10:45:33][D][main:150]: Settings:: Released
```

As soon as the on_touch event fires the TT21100 Touchscreen becomes unresponsive and doesn't report any further events.  However the "Settings" button demonstrates that the device is still functioning.

nielsnl68, 2024-01-30:
@woodshoes , can you tell what is going one with this repeat repaid of the following section? Is that you doing or is this something executed by the **tt21100**?

nielsnl68, 2024-01-30:
As @clydebarrow already suggested earlier. Please disable the "interupt_pin" for now. 
And connect pin gpo3 to the a `bineary_sensor:` so we can monitor that pin for later use.

and update your testscript with:

```yaml
substitutions:
  device_name: voice-assistant-1
  device_verbose_name: "Voice Assistant 1"

  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_hotspot_password: !secret wifi_hotspot_password

  api_encryption_key: !secret api_encryption_key
  ota_password: !secret ota_password

packages:
  device: !include templates/esp32.s3.template.yaml
  base: !include templates/wifi.template.yaml

logger:
  level: VERBOSE

psram:
  mode: octal
  speed: 80MHz

external_components:
  - source: github://nielsnl68/esphome#nvds-t221100-fix2
    components: [ display, touchscreen, tt21100 ]
    refresh: 0sec

## BEGIN - Touchscreen
i2c:
  sda: GPIO8
  scl: GPIO18
  scan: false

touchscreen:
  - platform: tt21100
  - update_interval: 50ms
    address: 0x24
    # interrupt_pin: GPIO3
    # reset_pin: GPIO48
    on_touch:
      - logger.log: "Touchscreen:: Touched"
    on_release:
      - logger.log: "Touchscreen:: Released"
## END - Touchscreen

## BEGIN - LCD Backlight
output:
  - platform: ledc
    id: gpio_45
    pin: GPIO45

light:
  - platform: monochromatic
    name: "LCD Backlight"
    id: lcd_backlight
    output: gpio_45
    restore_mode: ALWAYS_ON
    disabled_by_default: true
    default_transition_length: 0s
## END - LCD Backlight

## BEGIN - LCD
spi:
  clk_pin: GPIO7
  mosi_pin: GPIO6

display:
  - platform: ili9xxx
    id: lcd
    model: s3box
    cs_pin: GPIO5
    dc_pin: GPIO4
    reset_pin: GPIO48
    update_interval: never
    lambda: |-
      // Draw a line from [0,0] to [100,50]
      it.line(0, 0, 319, 239);
## END - LCD

binary_sensor:
  - platform: gpio
    id: settings
    name: "interupt pin"
    pin:
      number: GPIO3
      mode: INPUT_PULLUP
      inverted: true

  - platform: tt21100
    name: "Home_button"
    index: 0
    on_press:
      - logger.log: "Home:: Touched"
    on_release:
      - logger.log: "Home:: Released"
```

cptskippy, 2024-01-30:
@nielsnl68 there's an issue with that branch preventing compilation:

```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/voice-assistant-1.yaml...
INFO Updating https://github.com/nielsnl68/esphome.git@None
ERROR Unable to import component touchscreen:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/e67b79ba/esphome/components/touchscreen/__init__.py", line 6, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
ERROR Unable to import component tt21100.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/e67b79ba/esphome/components/tt21100/binary_sensor/__init__.py", line 7, in <module>
    from ..touchscreen import TT21100Touchscreen, TT21100ButtonListener
  File "/data/external_components/e67b79ba/esphome/components/tt21100/touchscreen/__init__.py", line 5, in <module>
    from esphome.components import i2c, touchscreen
  File "/data/external_components/e67b79ba/esphome/components/touchscreen/__init__.py", line 6, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

touchscreen: [source <unicode string>:36]
  
  Component not found: touchscreen.
  - platform: tt21100
    address: 36
    on_touch: 
      - logger.log: Touchscreen:: Touched
    on_release: 
      - logger.log: Touchscreen:: Released
binary_sensor.tt21100: [source <unicode string>:98]
  
  Platform not found: 'binary_sensor.tt21100'.
  platform: tt21100
  name: Home
  index: 0
  on_press: 
    - logger.log: Home:: Touched
  on_release: 
[... 7 lines cut ...]
[06:11:29][V][tt21100:120]: Touch 0: Type=0, Tip=0, EventId=0, TouchId=20, X=184, Y=148, Pressure=24, MajorAxisLen=0, Orientation=0
[06:11:29][D][touchscreen:114]: Touch status: 0/7: raw:( 184,  148) calc:(   0,    0)
[06:11:29][D][main:209]: Touchscreen:: Touched
[06:11:29][V][tt21100:109]: Touch report: Len=17, ID=1, Time=19941, LargeObject=0, RecordNum=0, RecordCounter=0, NoiseEffect=4
[06:11:29][V][tt21100:120]: Touch 0: Type=0, Tip=0, EventId=0, TouchId=24, X=184, Y=148, Pressure=24, MajorAxisLen=0, Orientation=0
[06:11:29][D][touchscreen:114]: Touch status: 0/7: raw:( 184,  148) calc:(   0,    0)
[06:11:29][V][tt21100:109]: Touch report: Len=17, ID=1, Time=19964, LargeObject=0, RecordNum=0, RecordCounter=0, NoiseEffect=6
[06:11:29][V][tt21100:120]: Touch 0: Type=0, Tip=0, EventId=0, TouchId=24, X=184, Y=148, Pressure=24, MajorAxisLen=0, Orientation=0

...

[06:11:29][D][touchscreen:114]: Touch status: 0/7: raw:( 143,  160) calc:(   0,    0)
[06:11:29][V][tt21100:109]: Touch report: Len=17, ID=1, Time=20421, LargeObject=0, RecordNum=0, RecordCounter=0, NoiseEffect=0
[06:11:29][V][tt21100:120]: Touch 0: Type=0, Tip=0, EventId=0, TouchId=12, X=143, Y=160, Pressure=37, MajorAxisLen=0, Orientation=0
[06:11:29][D][touchscreen:114]: Touch status: 0/7: raw:( 143,  160) calc:(   0,    0)
[06:11:29][V][tt21100:109]: Touch report: Len=7, ID=1, Time=20471, LargeObject=0, RecordNum=0, RecordCounter=0, NoiseEffect=2
[06:11:29][D][touchscreen:114]: Touch status: 0/0: raw:( 143,  160) calc:(   0,    0)
[06:11:29][D][main:214]: Touchscreen:: Released
```

The on_touch and on_release events are firing and it continues to work after multiple touches.

> And connect pin gpo3 to the a bineary_sensor: so we can monitor that pin for later use.

I'll try to find some time to disassemble the box so I can solder a wire to the test pad.  I was just using a probe and holding it precariously to test previously.

nielsnl68, 2024-01-30:
I see that i made a type while creating the branch
Instead of making it **nvds-tt21100 i made it **nvds-t2119-fix2**

I updated the YAML example above.

nielsnl68, 2024-01-30:
not sure why yet but it seams that the display width and height are not taken over from the display settings.
could you add:

```yaml
calibration:
   x_min: 0
   y_min: 0
   x: max: "display_width_value"
   y_max:  "display_heigtj_value"
```

You need to replace : "display_width_value", "display_height_value"  that you use with this display.

tommck, 2024-01-31:
@nielsnl68 if you need me to test things while @cptskippy is on holiday, let me know. I have an ESP32-S3-BOX that I can test this with.

tommck, 2024-01-31:
I'm trying to figure out the entire file contents I need to have in my YAML. Can you post all of what you want me to test?

tommck, 2024-01-31:
I don't have these 3 secrets... are they needed?

```
wifi_hotspot_password: !secret wifi_hotspot_password
api_encryption_key: !secret api_encryption_key
ota_password: !secret ota_password
```

tommck, 2024-01-31:
I'm also getting errors reading the packages

```
  device: !include templates/esp32.s3.template.yaml
  base: !include templates/wifi.template.yaml
```

tommck, 2024-01-31:
@nielsnl68 by the way, the screen is 320x240 pixels in case you needed to know that for the display calibration

nielsnl68, 2024-02-01:
The only part you need is the "touchscreen" part the other stuff you can use the default parts you use in other esphome devices,

nielsnl68, 2024-02-01:
i should have said: "The only parts you need are the "**display**" and "touchscreen"  parts ... "

tommck, 2024-02-01:
Right now I have this and it complains about the touchscreen component not being found ("Component not found: touchscreen.
". I'm pretty new at this, so forgive any ignorance:

```yaml
---
substitutions:
  name: esp32-s3-box
  friendly_name: ESP32 S3 Box

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  platformio_options:
    board_build.flash_mode: dio
  min_version: 2023.11.5

esp32:
  board: esp32s3box
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

psram:
  mode: octal
  speed: 80MHz

external_components:
  - source: github://nielsnl68/esphome#nvds-t2119-fix2
    components: [ display, touchscreen, tt21100 ]
    refresh: 0sec

spi:
  clk_pin: 7
  mosi_pin: 6

display:
  - platform: ili9xxx
    id: s3_box_lcd
    model: S3BOX
    data_rate: 40MHz
    cs_pin: 5
    dc_pin: 4
    reset_pin: 48
    update_interval: never

## BEGIN - Touchscreen
i2c:
  sda: GPIO8
  scl: GPIO18
  scan: false

touchscreen:
  - platform: tt21100
    update_interval: 50ms
    address: 0x24
    interrupt_pin: GPIO3
    # reset_pin: GPIO48
    on_touch:
      - logger.log: "Touchscreen:: Touched"
    on_release:
      - logger.log: "Touchscreen:: Released"

## END - Touchscreen
```

tommck, 2024-02-03:
Ahh.. the URL was supposed to have an `@`, not `#` in it... `github://nielsnl68/esphome@nvds-t2119-fix2`

tommck, 2024-02-03:
@nielsnl68 I got it loaded finally.  I touch it and it only responds the first time :(

tommck, 2024-02-03:
This is my current setup:

```yaml
---
substitutions:
  name: esp32-s3-box
  friendly_name: ESP32 S3 Box
  loading_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/loading_320_240.png
  idle_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/idle_320_240.png
  listening_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/listening_320_240.png
  thinking_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/thinking_320_240.png
  replying_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/replying_320_240.png
  error_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/error_320_240.png

  loading_illustration_background_color: '000000'
  idle_illustration_background_color: '000000'
  listening_illustration_background_color: 'FFFFFF'
  thinking_illustration_background_color: 'FFFFFF'
  replying_illustration_background_color: 'FFFFFF'
  error_illustration_background_color: '000000'

  voice_assist_idle_phase_id: '1'
  voice_assist_listening_phase_id: '2'
  voice_assist_thinking_phase_id: '3'
  voice_assist_replying_phase_id: '4'
  voice_assist_not_ready_phase_id: '10'
  voice_assist_error_phase_id: '11'  
  voice_assist_muted_phase_id: '12'

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  project:
    name: esphome.voice-assistant
    version: "1.0"
  platformio_options:
    board_build.flash_mode: dio
  min_version: 2023.11.5
  on_boot:
      priority: 600
      then: 
        - script.execute: draw_display
        - delay: 30s
        - if:
            condition:
              lambda: return id(init_in_progress);
            then:
              - lambda: id(init_in_progress) = false;
              - script.execute: draw_display

esp32:
  board: esp32s3box
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

psram:
  mode: octal
  speed: 80MHz

external_components:
  - source: github://pr#5230
    components: esp_adf
    refresh: 0s
  - source: github://nielsnl68/esphome@nvds-t2119-fix2
[... 333 lines cut ...]
      - id: muted_page
        lambda: |-
          it.fill(Color::BLACK);

## BEGIN - Touchscreen
i2c:
  sda: GPIO8
  scl: GPIO18
  scan: false

touchscreen:
  - platform: tt21100
    update_interval: 50ms
    address: 0x24
    interrupt_pin: GPIO3
    # reset_pin: GPIO48
    display: s3_box_lcd
    on_touch:
      - logger.log: "Touchscreen:: Touched"
    on_release:
      - logger.log: "Touchscreen:: Released"
## END - Touchscreen
```

It only spits out "touched" and no "released" message

krzychb, 2024-02-09:
Hi @nielsnl68, I can do the testing as well. 
Only the home button works but only till the touch screen is touched. 
Here is the log (I have deleted the Wi-Fi part):

```
[17:37:40][I][app:102]: ESPHome version 2023.12.9 compiled on Feb  9 2024, 17:29:54
[17:37:40][I][app:104]: Project esphome.temperature-controller version 0.3
[17:37:40][C][wifi:573]: WiFi:
...
[17:37:40][C][logger:439]: Logger:
[17:37:40][C][logger:440]:   Level: VERBOSE
[17:37:40][C][logger:441]:   Log Baud Rate: 115200
[17:37:40][C][logger:443]:   Hardware UART: USB_SERIAL_JTAG
[17:37:40][C][i2c.idf:061]: I2C Bus:
[17:37:40][C][i2c.idf:062]:   SDA Pin: GPIO8
[17:37:40][C][i2c.idf:063]:   SCL Pin: GPIO18
[17:37:40][C][i2c.idf:064]:   Frequency: 50000 Hz
[17:37:40][C][i2c.idf:067]:   Recovery: bus successfully recovered
[17:37:40][I][i2c.idf:077]: Results from i2c bus scan:
[17:37:40][I][i2c.idf:083]: Found i2c device at address 0x18
[17:37:40][I][i2c.idf:083]: Found i2c device at address 0x24
[17:37:40][I][i2c.idf:083]: Found i2c device at address 0x40
[17:37:40][I][i2c.idf:083]: Found i2c device at address 0x68
[17:37:40][C][spi:067]: SPI bus:
[17:37:40][C][spi:068]:   CLK Pin: GPIO7
[17:37:40][C][spi:069]:   SDI Pin: 
[17:37:40][C][spi:070]:   SDO Pin: GPIO6
[17:37:40][C][spi:072]:   Using HW SPI: SPI2_HOST
[17:37:40][C][ledc.output:164]: LEDC Output:
[17:37:40][C][ledc.output:165]:   Pin GPIO45
[17:37:40][C][ledc.output:166]:   LEDC Channel: 0
[17:37:40][C][ledc.output:167]:   PWM Frequency: 1000.0 Hz
[17:37:40][C][ledc.output:168]:   Bit depth: 14
[17:37:40][V][ledc.output:169]:   Max frequency for bit depth: 4882.812500
[17:37:40][V][ledc.output:171]:   Min frequency for bit depth: 76.298599
[17:37:40][V][ledc.output:172]:   Max frequency for bit depth-1: 9765.625000
[17:37:40][V][ledc.output:174]:   Min frequency for bit depth-1: 152.597198
[17:37:40][V][ledc.output:175]:   Max frequency for bit depth+1: 2441.406250
[17:37:40][V][ledc.output:177]:   Min frequency for bit depth+1: 38.149300
[17:37:40][V][ledc.output:178]:   Max res bits: 14
[17:37:40][V][ledc.output:179]:   Clock frequency: 80000000.000000
[17:37:40][C][ili9xxx:069]: ili9xxx
[17:37:40][C][ili9xxx:069]:   Rotations: 0 °
[17:37:40][C][ili9xxx:069]:   Dimensions: 320px x 240px
[17:37:40][C][ili9xxx:070]:   Width Offset: 0
[17:37:40][C][ili9xxx:071]:   Height Offset: 0
[17:37:40][C][ili9xxx:077]:   Color mode: 16bit
[17:37:40][C][ili9xxx:086]:   Data rate: 40MHz
[17:37:40][C][ili9xxx:088]:   Reset Pin: GPIO48
[17:37:40][C][ili9xxx:089]:   CS Pin: GPIO5
[17:37:40][C][ili9xxx:090]:   DC Pin: GPIO4
[17:37:40][C][ili9xxx:092]:   Swap_xy: NO
[17:37:40][C][ili9xxx:093]:   Mirror_x: YES
[17:37:40][C][ili9xxx:094]:   Mirror_y: YES
[17:37:40][C][ili9xxx:099]:   Update Interval: never
[17:37:40][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Settings'
[17:37:40][C][gpio.binary_sensor:016]:   Pin: GPIO0
[17:37:40][C][tt21100:138]: TT21100 Touchscreen:
[17:37:40][C][tt21100:139]:   Address: 0x24
[17:37:40][C][tt21100:140]:   Interrupt Pin: GPIO3
[17:37:40][C][light:103]: Light 'LCD Backlight'
[17:37:40][C][light:105]:   Default Transition Length: 0.0s
[17:37:40][C][light:106]:   Gamma Correct: 2.80
[17:37:40][C][psram:020]: PSRAM:
[17:37:40][C][psram:021]:   Available: YES
[17:37:40][C][psram:024]:   Size: 8191 KB
[17:37:40][C][tt21100.binary_sensor:015]: TT21100 Button 'Home'
[17:37:40][C][tt21100.binary_sensor:016]:   Index: 0
[17:37:40][C][captive_portal:088]: Captive Portal:
[17:37:40][C][mdns:115]: mDNS:
[... 16 lines cut ...]
[17:37:41][C][api:140]:   Address: esp32-s3-box-tester.local:6053
[17:37:41][C][api:142]:   Using noise encryption: YES
[17:38:12][V][tt21100:095]: Button report: Len=14, ID=3, Time=15603, Value=[1], Signal=[008A][0000][0000][0000]
[17:38:12][D][binary_sensor:036]: 'Home': Sending state ON
[17:38:12][D][main:215]: Home:: Touched
[17:38:13][V][tt21100:095]: Button report: Len=14, ID=3, Time=15822, Value=[0], Signal=[0000][0000][0000][0000]
[17:38:13][D][binary_sensor:036]: 'Home': Sending state OFF
[17:38:13][D][main:221]: Home:: Released
[17:38:13][V][tt21100:095]: Button report: Len=14, ID=3, Time=16014, Value=[1], Signal=[0086][0000][0000][0000]
[17:38:13][D][binary_sensor:036]: 'Home': Sending state ON
[17:38:13][D][main:215]: Home:: Touched
[17:38:13][V][tt21100:095]: Button report: Len=14, ID=3, Time=16164, Value=[0], Signal=[0000][0000][0000][0000]
[17:38:13][D][binary_sensor:036]: 'Home': Sending state OFF
[17:38:13][D][main:221]: Home:: Released
[17:38:13][V][tt21100:095]: Button report: Len=14, ID=3, Time=16387, Value=[1], Signal=[0074][0000][0000][0000]
[17:38:13][D][binary_sensor:036]: 'Home': Sending state ON
[17:38:13][D][main:215]: Home:: Touched
[17:38:13][V][tt21100:095]: Button report: Len=14, ID=3, Time=16564, Value=[0], Signal=[0000][0000][0000][0000]
[17:38:13][D][binary_sensor:036]: 'Home': Sending state OFF
[17:38:13][D][main:221]: Home:: Released
[17:38:15][V][tt21100:109]: Touch report: Len=17, ID=1, Time=17819, LargeObject=0, RecordNum=0, RecordCounter=0, NoiseEffect=2
[17:38:15][V][tt21100:120]: Touch 0: Type=0, Tip=0, EventId=0, TouchId=20, X=148, Y=125, Pressure=34, MajorAxisLen=0, Orientation=0
[17:38:15][D][touchscreen:114]: Touch status: 0/7: raw:( 148,  125) calc:(   0,    0)
[17:38:15][D][main:237]: Touchscreen:: Touched
```

cptskippy, 2024-02-11:
Hey @nielsnl68 , I'm back.

[I was able to solder a wire to GPIO3 and connected it to GPIO41.](https://imgur.com/a/1REJNg9)

I added a binary sensor  to monitor GPIO41 called "Debug":

```
binary_sensor:
  - platform: gpio
    id: settings
    name: "Settings"
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true
    on_press:
      - logger.log: "Settings:: Pressed"
    on_release:
      - logger.log: "Settings:: Released"

  - platform: tt21100
    name: "Home"
    index: 0
    on_press:
      - logger.log: "Home:: Touched"
    on_release:
      - logger.log: "Home:: Released"

  - platform: gpio
    id: debug
    name: "Debug"
    pin:
      number: GPIO41
      mode:
        input: true
```

And this is what the logs look like when I touched the Home button twice then the touchscreen:

```
...

[18:01:11][C][mdns:116]:   Hostname: voice-assistant-1
[18:01:11][V][mdns:117]:   Services:
[18:01:11][V][mdns:119]:   - _esphomelib, _tcp, 6053
[18:01:11][V][mdns:121]:     TXT: version = 2023.12.9
[18:01:11][V][mdns:121]:     TXT: mac = [redacted]
[18:01:11][V][mdns:121]:     TXT: platform = ESP32
[18:01:11][V][mdns:121]:     TXT: board = esp32s3box
[18:01:11][V][mdns:121]:     TXT: network = wifi
[18:01:11][V][mdns:121]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[18:01:11][C][ota:097]: Over-The-Air Updates:
[18:01:11][C][ota:098]:   Address: voice-assistant-1.local:3232
[18:01:11][C][ota:101]:   Using Password.
[18:01:11][C][api:139]: API Server:
[18:01:11][C][api:140]:   Address: voice-assistant-1.local:6053
[18:01:11][C][api:142]:   Using noise encryption: YES
[18:01:14][D][binary_sensor:036]: 'Debug': Sending state OFF
[18:01:14][V][tt21100:095]: Button report: Len=14, ID=3, Time=32347, Value=[1], Signal=[00A5][0000][0000][0000]
[18:01:14][D][binary_sensor:036]: 'Home': Sending state ON
[18:01:14][D][main:192]: Home:: Touched
[18:01:14][D][binary_sensor:036]: 'Debug': Sending state ON
[18:01:14][D][binary_sensor:036]: 'Debug': Sending state OFF
[18:01:14][V][tt21100:095]: Button report: Len=14, ID=3, Time=32947, Value=[0], Signal=[0000][0000][0000][0000]
[18:01:14][D][binary_sensor:036]: 'Home': Sending state OFF
[18:01:14][D][main:198]: Home:: Released
[18:01:14][D][binary_sensor:036]: 'Debug': Sending state ON
[18:01:15][D][binary_sensor:036]: 'Debug': Sending state OFF
[18:01:15][V][tt21100:095]: Button report: Len=14, ID=3, Time=33948, Value=[1], Signal=[0110][0000][0000][0000]
[18:01:15][D][binary_sensor:036]: 'Home': Sending state ON
[18:01:15][D][main:192]: Home:: Touched
[18:01:15][D][binary_sensor:036]: 'Debug': Sending state ON
[18:01:16][D][binary_sensor:036]: 'Debug': Sending state OFF
[18:01:16][V][tt21100:095]: Button report: Len=14, ID=3, Time=34348, Value=[0], Signal=[0000][0000][0000][0000]
[18:01:16][D][binary_sensor:036]: 'Home': Sending state OFF
[18:01:16][D][main:198]: Home:: Released
[18:01:16][D][binary_sensor:036]: 'Debug': Sending state ON
[18:01:19][D][binary_sensor:036]: 'Debug': Sending state OFF
[18:01:19][V][tt21100:109]: Touch report: Len=17, ID=1, Time=37269, LargeObject=0, RecordNum=0, RecordCounter=0, NoiseEffect=2
[18:01:19][V][tt21100:120]: Touch 0: Type=0, Tip=0, EventId=0, TouchId=20, X=141, Y=166, Pressure=23, MajorAxisLen=0, Orientation=0
[18:01:19][D][touchscreen:114]: Touch status: 0/1: raw:( 141, 166,  23) calc:(140, 165)
[18:01:19][D][main:245]: Touchscreen:: Touched
```

nielsnl68, 2024-02-11:
> @cptskippy: I added a binary sensor to monitor GPIO41 called "Debug"
Cool good idea. although you could have used the GIO3 for this as well, no need to add an extra wire.

@tommck and @cptskippy do you mind going go to   [Discoard](https://discord.com/channels/429907082951524364/1196256748407902228) to see what happens exactly.

boozer2, 2025-03-01:
I'm having the same issue. As soon as you touch the screen once, touch stops working.

As soon as I commented out the interrupt pin, this started working. I set a temporary binary_sensor on the interrupt pin (GPIO3 on my ESP32-S3-BOX (version 1), and I found the root cause.

In: https://github.com/esphome/esphome/blob/dev/esphome/components/tt21100/touchscreen/tt21100.cpp, the TT21100Touchscreen::setup() has this line:
    this->attach_interrupt_(this->interrupt_pin_, gpio::INTERRUPT_FALLING_EDGE);

The touchscreen component is set to trigger only on falling edge, but the way this device works, touching the screen is falling edge, but releasing is rising edge. So this should be INTERRUPT_ANY_EDGE. I don't have an easy way to change this on my setup but it's very clear from my binary_sensor on the interrupt pin that this is how it's working.

However - one possible issue is that the extra touch buttons seem to do falling edge but then immediately rise, so for the touch buttons we might get two rapid-fire events after this change.

For now a successful workaround is just to remove any interrupt pin set up on the touchscreen, but changing the above line should fix the issue at least for the screen itself.

boozer2, 2025-03-02:
Ignore my confidence above - the change alone doesn't fix the issue though I think it might still make sense to trigger an interrupt for the release as well as the touch. 

It seems like the touchscreen code just stops working once you touch the screen once with interrupts enabled. 

I'm thinking the touchscreen interrupt code in general is having issues, since I see at least one other touchscreen platform with the same issue: https://github.com/esphome/issues/issues/5708

I'll try and dig deeper on this.

psychowood, 2025-04-12:
Sorry to ask, is this issue still open?
I'm trying to add touch capabilities to the standard HA [voice assistant configuration](https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box/esp32-s3-box.yaml) and I'm not sure if I'm missing something in the configuration or if it is indeed broken.
I've got the esp32-s3-box just recently and I'm fairly new to it.

closed 2025-04-28 by jesserockz