esphome/issues #5275: Serial device error with 2023.12.* and adafruit_qtpy_esp32

asolochek, 2023-12-22:
### The problem

After updating ESPHome to 2023.12.1 (and 2023.12.3) and trying to rebuild the images for all my devices, I had a couple adafruit_qtpy_esp32 boards fail with the following: 

```src/esphome/components/logger/logger.cpp: In member function 'void esphome::logger::Logger::pre_setup()':
src/esphome/components/logger/logger.cpp:287:29: error: 'Serial' was not declared in this scope
         this->hw_serial_ = &Serial;
                             ^~~~~~
src/esphome/components/logger/logger.cpp:287:29: note: suggested alternative: 'Serial2'
         this->hw_serial_ = &Serial;
                             ^~~~~~
                             Serial2
*** [.pioenvs/office-air-quality/src/esphome/components/logger/logger.cpp.o] Error 1
========================== [FAILED] Took 1.75 seconds ==========================```

### Which version of ESPHome has the issue?

2023.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

Adafruit QTPY ESP32

### Component causing the issue

logger

### Example YAML snippet

```yaml
logger:
  level: ERROR
```

clydebarrow, 2023-12-24:
Try using ESP-IDF instead of Arduino. The Arduino support for the USB_CDC uart seems to be broken. Do you use the UART for logging? If not you can turn it off with `baud_rate: 0`.

The default logger uart was UART0 before the latest release. You can restore that with `hardware_uart: UART0`

asolochek, 2023-12-24:
I do use logging, which I assumed was a soft uart since I access those logs wirelessly.

I would use esp-idf except there is no support for the neopixel in that framework and I use that on one of these boards. 

My temporary workaround has been disabling the logging which is fine since these boards are already debugged and doing what they need, but it would be really obnoxious to be trying to develop on one right now.

ssieb, 2023-12-24:
Can you paste your entire config?
Also, you should use the rmt component instead of neopixel.

clydebarrow, 2023-12-24:
Wireless logging works without a uart - setting the baudrate to 0 will disable the uart, but wireless logging will still work.

AzonInc, 2024-03-23:
Is there anything new with this? On 2024.3 it seems to be still kind of an issue.

The improv_serial Component depends on it, so setting baud_rate to 0 is no option for me.
I can't use IDF either because https://github.com/uriyacovy/ESPHome_nuki_lock is depending on the arduino framework.

My Hardware is based on an ESP32-S3 and is only capable of USB CDC.
It doesn't have a serial converter as the USB Port is directly connected to USB_D+ and USB_D-.

ssieb, 2024-03-23:
No config has been provided, so there's no way to answer this and there have been various uart fixes since this was filed.

closed 2024-03-23 by ssieb