esphome/issues #5276: Devices with esp32-c3-devkitm-1 Inaccessible after updating 2023.12.*

lensherm, 2023-12-22:
### The problem

After this update none of the 4 SwitchBot Plug Mini devices are connecting to wifi. The LED on them keeps blinking blue, indicating no network connectivity. The captive portal hotspot wifi is available and showing up, but once I enter the password and authenticate onto the wifi, no DHCP assignment takes place. Manually assigning the network info on my PC doesn't do anything either, as the device doesn't appear at the 192.168.4.1 address. Looks like the wifi/network component of esphome is somehow malfunctioning after this update.

### Which version of ESPHome has the issue?

2023.12.*

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:
  name: compressor-switch
  friendly_name: Compressor Switch

esphome:
  name: ${name}

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"

web_server:
  port: 80

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "SwitchBot Plug Hotspot"
    password: !secret esphome_fallback_ap_password

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  password: !secret esphome_api_password

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Enable Bluetooth Proxying
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

switch:
  - platform: gpio
    name: "Relay"
    id: "relay"
    device_class: outlet
    pin: 6
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      then:
        - light.turn_on: white_led
    on_turn_off:
      then:
        - light.turn_off: white_led

binary_sensor:
  - platform: gpio
    internal: true
    pin:
      number: 2
      mode: INPUT_PULLUP
      inverted: true
    name: "Button"
    filters:
      - delayed_on: 10ms
    on_press:
      then:
        - switch.toggle: relay
      
light:
  - platform: binary
    internal: true
    name: "white led"
    id: "white_led"
    output: white_output
    
output:
  - id: white_output
    platform: gpio
    pin: GPIO7
    inverted: true

# set blue led as status
status_led:
  pin:
    number: GPIO8
    inverted: true

sensor:
  - platform: hlw8012
    sel_pin:
      number: 20
      inverted: true
    cf_pin: 18
    cf1_pin: 19
    model: BL0937
    voltage_divider: 1467
    current:
      name: "Current"
    voltage:
      name: "Voltage"
    power:
      name: "Power"
    energy:
      name: "Energy"
    update_interval: 1s
    change_mode_every: 4
```

lensherm, 2023-12-22:
Was able to get the device into safe mode and flash it with the bluetooth proxy config removed. The offending piece seems to be `esp32_ble_tracker:`

clydebarrow, 2023-12-24:
Have you tried setting the scan parameters to more reasonable values?
```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

```
this basically dedicates all the shared 2.4GHz transceiver time to BLE, the window should be smaller than the interval to allow time for WiFi operations. The defaults are 320ms/30ms.

ssieb, 2023-12-24:
Even with the default settings, it still has some trouble.  There's a workaround at https://github.com/esphome/issues/issues/3668#issuecomment-1287568813 that seems to work.

clydebarrow, 2023-12-24:
The C3 is not a really good candidate for a BLE proxy, being a single core. Seems like it can be made to work, but it's always going to be rather marginal for the job I would think.

clydebarrow, 2023-12-24:
Since the issue seems to be that the WiFi needs exclusive control of the RF section for a fairly long time when first connecting (and presumably reconnecting), you could try something like this:

```
esp32_ble_tracker:
  scan_parameters:
    continuous: false

api:
  on_client_connected:
     - esp32_ble_tracker.start_scan:
        continuous: true
  on_client_disconnected:
     - esp32_ble_tracker.stop_scan:

 ```

Since without the API connection (which requires WiFi connected) there is no point in scanning anyway, and if the WiFi drops the API connection will drop, allowing a reconnect.

lensherm, 2023-12-24:
> Since the issue seems to be that the WiFi needs exclusive control of the RF section for a fairly long time when first connecting (and presumably reconnecting), you could try something like this: [...]

Thanks for the suggestions. I'll play around with it some more.

I'm still wondering what changed in 2023.12 which is causing the issue. Couldn't see anything obvious in the change logs. I've had these devices configured and working as BT proxies for about half a year, iteratively going along through all the updates. Wonder if it's a timing issue.

00RedD00, 2023-12-27:
A friend and I both had the same problem with our 4 switchbot each, also have been steadily updating through the  past 8-10 months without any issues before this update. Thanks the the workaround!

clydebarrow, 2023-12-28:
I tested this out today on a C3 and confirmed the problem, and that the "workaround" to start and stop scanning on api connect/disconnect solves the problem.

lensherm, 2023-12-28:
> I tested this out today on a C3 and confirmed the problem, and that the "workaround" to start and stop scanning on api connect/disconnect solves the problem.

Clyde, for the testing you performed, did you only have the `api:` section, which you mentioned, or did you also have to have the `on_boot:` section, as mentioned in an earlier workaround?

lensherm, 2023-12-28:
> on_client_connected:
>      - esp32_ble_tracker.start_scan:
>         continuous: true
>   on_client_disconnected:
>      - esp32_ble_tracker.stop_scan:

Clyde, never mind. I answered my own question by running a number of test cases. Your workaround seems to be sound.

clydebarrow, 2023-12-28:
> does having the tracker being non continuous not introduce the bug in the first place?

Yes, if you check the docs for the `ble_tracker`, when `continuous` is false, scanning does not start until a `start_scan` action is executed. So that flag implies both continuous and auto-start.

lensherm, 2023-12-28:
@clydebarrow , I saw the documentation fix and I'm wondering if in addition to that, there can be a compile-time warning on single-core chips, or even an error thrown, preventing one from compiling without configuring with the workaround. 

The reason I'm even talking about going to these lengths is cases like mine. The specific device I experienced this issue with, doesn't lend itself to easy serial flashing, without destroying its housing. As such, if I didn't have it previously setup with the correct WiFi credentials already, then putting it into safe mode, would still not have provided me with the opportunity to unbrick it.

Finally, I'm still perplexed, as I did have this exact config working on four of these devices for at least half a year, until the update to 2023.12. Hence, I'm a bit puzzled as to what caused the undesired behavior in the first place.

clydebarrow, 2023-12-29:
I have no idea why it worked before. The only pertinent change seems to be the PR that allowed the BLE to be enabled and disabled and a quick look at the ble_tracker changes didn't raise any red flags.

I'm not inclined to think a compile-time check is a good idea, at least not until the basic cause is well characterised. 

I just did another test, and without the workaround, but setting the scan duration to 10s the problem seemed to go away.

loopy321, 2024-01-07:
> Finally, I'm still perplexed, as I did have this exact config working on four of these devices for at least half a year, until the update to 2023.12. Hence, I'm a bit puzzled as to what caused the undesired behavior in the first place.

I am experiencing this exact same issue with four of these switchbot plugs.  They worked great for almost a year and with the most recent esphome update that all stopped reliably connecting to wifi.  In my case they all worked very well when I was on 2023.7.1, now on esphome 2023.12.5 nothing but flaky behavior and dropping wifi, in addition the fallback hotspot does not accept the correct password.

I [posted in the community forum here](https://community.home-assistant.io/t/support-for-switchbot-plug-mini-w1901400-with-bluetooth-proxying/441893/46).

@lensherm how did you get these into safe mode?

lensherm, 2024-01-07:
> > Finally, I'm still perplexed, as I did have this exact config working on four of these devices for at least half a year, until the update to 2023.12. Hence, I'm a bit puzzled as to what caused the undesired behavior in the first place.
> 
> I am experiencing this exact same issue with four of these switchbot plugs. They worked great for almost a year and with the most recent esphome update they all stopped reliably connecting to wifi. In my case they all worked very well when I was on 2023.7.1, now on esphome 2023.12.5 nothing but flaky behavior and dropping wifi, in addition the fallback hotspot does not accept the correct password.
> 
> I [posted in the community forum here](https://community.home-assistant.io/t/support-for-switchbot-plug-mini-w1901400-with-bluetooth-proxying/441893/46).
> 
> @lensherm how did you get these into safe mode?

Cycle power to the device up to 10 times, until the device boots into safe mode. 
-- Apply power, wait about 5-7 seconds, disconnect from power, repeat.. 
[https://esphome.io/components/ota.html](url)

In my case I had the plugs configured to show status through the LED, so when they went into safe mode, the LED started blinking more frequently than before.

jjcf89, 2024-02-20:
Does anyone get a relay clicking sounds along with the flashing light? My switchbot plug with the above code also stopped working around December and I thought it was fried but perhaps its this instead?  Though I cannot seem to get it to go into safe mode using the 10 power cycle method above...

closed 2024-02-27 by nagyrobi

myhomeiot, 2024-05-11:
@clydebarrow Can you check if [this](https://digiblur.com/wiki/ha/esphome-bluetooth-proxy-esp32c3) solution (CONFIG_ESP_TASK_WDT_TIMEOUT_S) will helps?
Thanks in advance.

clydebarrow, 2024-05-12:
> @clydebarrow Can you check if [this](https://digiblur.com/wiki/ha/esphome-bluetooth-proxy-esp32c3) solution (CONFIG_ESP_TASK_WDT_TIMEOUT_S) will helps? Thanks in advance.

Why don't you check it?

myhomeiot, 2024-05-12:
> Why don't you check it?

Don't have a ESP32-C3 or SwitchBot Plug Mini yet.

AJolly, 2024-05-24:
Did not seem to work great for me last night:
my Mini would not connect to wifi, and while I could see it I couldn't successfully connect to the fall back AP.
 but luckily this morning I mange to reconnect to my switchbot plug mini (and I promptly removed the ble proxy from my config)