esphome/issues #5277: 4 Haier airco's fail to uodate to 2023.12.3

Keeze, 2023-12-22:
### The problem

I try to update my 4 haier airco's to ESPHome 2023.12.3 and they all fail with the same issue.

The yaml:

```
esphome:
  name: airco-zolder

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXX"

ota:
  password: "XXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Airco-Zolder Fallback Hotspot"
    password: "XXXXXX"

substitutions:
  device_name: Airco Zolder

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

uart:
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  id: ac_port  
    
logger:
  level: DEBUG
    
climate:
  - platform: haier
    id: airco_zolder
    protocol: hOn
    name: ${device_name} 
    uart_id: ac_port
    wifi_signal: true           # Optional, default true, enables WiFI signal transmission from ESP to AC
    beeper: true                # Optional, default true, disables beep on commands from ESP
    display: true               # Optional, default true, can be used to turn off LED display
    answer_timeout: 200ms       # Optional, request answer timeout, can be used to increase the timeout
                                # for some ACs that have longer answer delays
    outdoor_temperature:        # Optional, outdoor temperature sensor
      name: ${device_name} outdoor temperature
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
    - 'OFF'
    - COOL
    - HEAT
    - HEAT_COOL
    - DRY
    - FAN_ONLY
    supported_presets:          # Optional, can be used to disable some presets if your AC does not support it
      - ECO
      - BOOST
      - SLEEP
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH
```

### Which version of ESPHome has the issue?

2023.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32-IDF

### Board

ESP32-for-Haier 

### Anything in the logs that might be useful for us?

```txt
Linking .pioenvs/airco-zolder/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/airco-zolder/src/esphome/components/haier/haier_base.o:(.literal._ZN7esphome5haier16HaierClimateBase18set_answer_timeoutEj+0x0): undefined reference to `haier_protocol::ProtocolHandler::set_answer_timeout(long long)'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/airco-zolder/src/esphome/components/haier/haier_base.o: in function `esphome::haier::HaierClimateBase::set_answer_timeout(unsigned int)':
/data/build/airco-zolder/src/esphome/components/haier/haier_base.cpp:122: undefined reference to `haier_protocol::ProtocolHandler::set_answer_timeout(long long)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/airco-zolder/firmware.elf] Error 1
========================= [FAILED] Took 217.96 seconds =========================
```

paveldn, 2023-12-23:
Hi @Keeze!

Ups. Sorry. It looks like I missed a function in a library. Will fix it ASAP. Meanwhile, you can just remove answer_timeout parameter. Anyway, starting from 2023.12.0 default timeout is 200 ms so your AC will work without this parameter.

Keeze, 2023-12-23:
All airco's are updated now, I have one airco which uses a D1 mini (smartair2) and here timeout was set to 400 ms. Lets see if it gives any problems. Problem solved for now, tnx!

closed 2023-12-23 by Keeze