esphome/issues #5278: not support pin validation : deepsleep/esp32_ext1_wakeup/pins

coolbiz, 2023-12-23:
### The problem

The "allow_other_uses" option cannot be used in "deep_sleep / esp32_ext1_wakeup / pins" for pin validation check.
I can't update esphome anymore.

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Supervisor 2023.12.0

### What platform are you using?

ESP32

### Board

esp32-s3

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO16
      mode: INPUT_PULLUP
      inverted: true
    name: "bed-left_closed"
    id: b_l_closed
    device_class: door
    on_press:
      then:
        - stepper.report_position:
            id: b_l_stepper
            position: -4500
        - stepper.set_target:
            id: b_l_stepper
            target: !lambda return id(b_l_stepper).current_position;

  - platform: gpio
    pin:
      number: GPIO17
      mode: INPUT_PULLUP
      inverted: true
    name: "bed-left_opened"
    id: b_l_opened
    device_class: door
    on_press:
      then:
        - stepper.report_position:
            id: b_l_stepper
            position: 0
        - stepper.set_target:
            id: b_l_stepper
            target: !lambda return id(b_l_stepper).current_position;

  - platform: gpio
    pin:
      number: GPIO25
      allow_other_uses: true
    name: "bed-left_open_switch"
    id: b_l_open_switch
    on_release:
      - if:
          condition:
            binary_sensor.is_on: b_l_opened
          then:
            - stepper.report_position:
                id: b_l_stepper
                position: -4500
          else:
            - stepper.report_position:
                id: b_l_stepper
                position: 0
      - stepper.set_target:
          id: b_l_stepper
          target: -4500
      - delay: 90s
      - deep_sleep.enter:
          id: c_opener_sleep

  - platform: gpio
    pin:
      number: GPIO26
      allow_other_uses: true
    name: "bed-left_close_switch"
    id: b_l_close_switch
    on_release:
      - if:
          condition:
            binary_sensor.is_on: b_l_closed # opened
          then:
            - stepper.report_position:
                id: b_l_stepper
                position: 0
          else:
            - stepper.report_position:
                id: b_l_stepper
                position: -4500
      - stepper.set_target:
          id: b_l_stepper
          target: 0
      - delay: 90s
      - deep_sleep.enter:
          id: c_opener_sleep

stepper:
  - platform: a4988
    id: b_l_stepper
    step_pin: GPIO14
    dir_pin: GPIO27
    max_speed: 1000 steps/s
    # Optional:
    sleep_pin: GPIO13
    acceleration: 300 steps/s^2

deep_sleep:
  id: c_opener_sleep
  run_duration: 100s
  esp32_ext1_wakeup:
    pins:
      - 25
      - 26
    mode: ANY_HIGH
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.3
INFO Reading configuration /config/esphome/esp32-curtain.yaml...
Failed config

deep_sleep: [source <unicode string>:148]
  
  Pin 25 is used in multiple places.
  id: c_opener_sleep
  run_duration: 100s
  esp32_ext1_wakeup: 
    pins: 
      - number: 25
        mode: 
          input: True
          output: False
          open_drain: False
          pullup: False
          pulldown: False
        inverted: False
        ignore_strapping_warning: False
        drive_strength: 20.0
      - number: 26
        mode: 
          input: True
          output: False
          open_drain: False
          pullup: False
          pulldown: False
        inverted: False
        ignore_strapping_warning: False
        drive_strength: 20.0
    mode: ANY_HIGH
Pin 26 is used in multiple places
```

### Additional information

I'm giving a signal to pins 25 and 26 with an infrared remote control.

clydebarrow, 2023-12-24:
You need to add `allow_other_uses` to all instances of the reused pins.
```
    pins:
      -
        number: 25
        allow_other_uses: true
      -
        number: 26
        allow_other_uses: true
```

closed 2023-12-24 by clydebarrow

coolbiz, 2023-12-24:
Thank you very much.