esphome/issues #5280: ESPHome 2023.12.3 killed my Sonoff Mini devices

MT31972, 2023-12-23:
### The problem

I have several Sonoff Mini devices which worked quite well until I upgraded from 2023.11.6 to 2023.12.3. 
After update they don't appear in my wifi (and HomeAssistant) anymore and they started recovery AP.
When I connect to recovery AP I can reach them by ping 192.168.4.1, but when try to update them from windows commandline I get an error.

esphome run --device 192.168.4.1 C:\Users\marku\Downloads\wzlichtwandlinks.yaml
...
RAM:   [====      ]  39.0% (used 31932 bytes from 81920 bytes)
Flash: [====      ]  42.6% (used 436403 bytes from 1023984 bytes)
============================================= [SUCCESS] Took 5.64 seconds =============================================
INFO Successfully compiled program.
INFO Connecting to 192.168.4.1
ERROR Connecting to 192.168.4.1:8266 failed: [WinError 10061] Es konnte keine Verbindung hergestellt werden, da der Zielcomputer die Verbindung verweigerte (sorry for german, connection is refused, wrong port?).

First I thought that maybe I have some wifi issues, but after the third device shows the same issue, I suppose it might be related to ESPHome version. Other Sonoff devices or Gosund sockets updated without any issues.

### Which version of ESPHome has the issue?

2023.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12

### What platform are you using?

ESP8266

### Board

sonoff mini

### Example YAML snippet

```yaml
substitutions:
  plug_name: "wzlichtwandlinks"
  friendly_name: "Wand Links"

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp8285

wifi:
  ssid: !secret wifi50_name
  password: !secret wifi50_pwd
  domain: .tondorf
  use_address: 192.168.90.207

 # manual IP
  manual_ip:
    static_ip: 192.168.90.207
    gateway: 192.168.90.1
    subnet: 255.255.255.0
    dns1: 1.1.1.1

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name}"
    password: !secret esp_fallback_hs_pwd

# Enable logging
logger:
#  level: VERBOSE
  
# Enable Home Assistant API
api:
  encryption:
    key: !secret esp_api_key

ota:
  safe_mode: True
  password: !secret esp_ota_pwd

time:
  - platform: homeassistant
    id: homeassistant_time

status_led:
  pin:
    number: GPIO13
    inverted: true

binary_sensor:
  - platform: gpio
    pin: GPIO00
    id: reset
    internal: true
    filters:
      - invert:
      - delayed_off: 10ms
    on_press:
      - light.toggle: 
          id: ${plug_name}_light

  - platform: gpio
    name: "${friendly_name} Switch"
    pin: GPIO04
    id: ${plug_name}_switch
    on_press:
      then:
        - light.turn_on:
            id: ${plug_name}_light
    on_release:
      then:
        - light.turn_off:
            id: ${plug_name}_light

output:
  - platform: gpio
    pin: GPIO12
    id: ${plug_name}_relay

light:
  - platform: binary
    name: "${friendly_name} Licht"
    id: ${plug_name}_light
    output: ${plug_name}_relay
```

### Anything in the logs that might be useful for us?

```txt
No
```

acrespog, 2023-12-23:
Same here with Sonmoff basic (HW: TH_V1.1). After several years running without problems, 2023.12.3 version kills all Sonoff devices in my home.

Conected vía USB serial, the device is in Wifi conection loop:

`[I][wifi:300]: WiFi Connecting to 'MySSID'...
[W][wifi_esp8266:493]: Event: Disconnected ssid='MySSID' reason='Probe Request Unsuccessful'
[W][wifi:618]: Error while connecting to network.
[W][wifi:654]: Restarting WiFi adapter...`

The config:
`esphome:
  name: luz-escritorio
  friendly_name: Luz_escritorio

esp8266:
  board: sonoff_basic

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "edited"

ota:
  password: "edited"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "edited"
    password: "edited"

captive_portal:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode:
        input: true
        pullup: true
      inverted: true
    name: "Luz escritorio Boton"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Luz escritorio"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes`

MT31972, 2023-12-23:
I was able to recover one of the devices. It seems that not latest firmware was uploaded. A few weeks ago I rearranged my network and moved esphome devices from VLAN 50 to VLAN90 and disabled VLAN 50. No problems until ESPHome 12.x.
I now reactivated VLAN50 and device appeared in the VLAN. Then I was able to access it and send a firmware file to it.
I could not test the other devices, because I already replaced them with other Sonoff Mini devices that run Tasmota.
Will check after XMas.

acrespog, 2023-12-23:
> Same here with Sonmoff basic (HW: TH_V1.1). After several years running without problems, 2023.12.3 version kills all Sonoff devices in my home.
> 
> Conected vía USB serial, the device is in Wifi conection loop:
> 
> ```
> [I][wifi:300]: WiFi Connecting to 'MySSID'...
> [W][wifi_esp8266:493]: Event: Disconnected ssid='MySSID' reason='Probe Request Unsuccessful'
> [W][wifi:618]: Error while connecting to network.
> [W][wifi:654]: Restarting WiFi adapter...
> ```
> 
> The config:
> 
> ```
> 
> esphome:
>   name: luz-escritorio
>   friendly_name: Luz_escritorio
> 
> esp8266:
>   board: sonoff_basic
> 
> 
> logger:
> 
> api:
>   encryption:
>     key: "edited"
> 
> ota:
>   password: "edited"
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> 
>   ap:
>     ssid: "edited"
>     password: "edited"
> 
> captive_portal:
> 
> binary_sensor:
>   - platform: gpio
>     pin:
>       number: GPIO0
>       mode:
>         input: true
>         pullup: true
>       inverted: true
>     name: "Luz escritorio Boton"
>     on_press:
>       - switch.toggle: relay
> 
> switch:
>   - platform: gpio
>     name: "Luz escritorio"
>     pin: GPIO12
>     id: relay
> 
> status_led:
>   pin:
>     number: GPIO13
>     inverted: yes
> ```

Reverted back all devices to 2023.12.1 via serial, and all are working again (I have had to unwire all of them, because to upgrade serially they have to be removed from live cables).

There is definitely a problem in this version (2023.12.3) with the sonoff wifi.

Particpant, 2024-01-19:
https://community.home-assistant.io/t/esphome-2023-12-5-bricked-my-esp-s/661901/19?u=participant
This thread is indicating that 2023.12.6 resolved whatever underlying issue existed. Has anyone else verified that claim?

acrespog, 2024-01-20:
> https://community.home-assistant.io/t/esphome-2023-12-5-bricked-my-esp-s/661901/19?u=participant This thread is indicating that 2023.12.6 resolved whatever underlying issue existed. Has anyone else verified that claim?

I was staying on version 2023.12.1 until there was an explicit solution to the problem, but after seeing your post I decided to give it a try. I just upgraded one of the devices to version 2023.12.8 and it died again.

I can confirm that the problem is still unresolved. For now I will downgrade to version 2023.12.1 again, while waiting for an explicit solution in the Changelog, as each failed test takes me more than half an hour between unwire, flash and mount. If no solution is given, I will consider placing some zigbee, but I feel bad for it because it has been working for years and the problem is some change introduced in version 2023.12.2.

Particpant, 2024-01-22:
Thanks for trying. Sorry it didn't work out. All my Sonoff devices are installed in inaccessible areas, and I've never successfully flashed a single one of them via TTL adapter, so they'd have to go in the bin if I had tried it and they got knocked offline. Thanks.

Particpant, 2024-01-23:
FWIW, I had  2 sonoff basics and a handful of S31s upgrade to 2023.12.9 just fine. 🤷

acrespog, 2024-01-25:
> FWIW, I had 2 sonoff basics and a handful of S31s upgrade to 2023.12.9 just fine. 🤷

Thanks for info.

I have tried with my two Sonoff basics, and they died with 2023.12.9. One of them have conected to Wifi after trying several minutes, but after a reboot, It didn't connect anymore.

Reverted back to 2023.12.1, and working again. Looking for Zigbee switches as I only have 2...