esphome/issues #5283: Fallback AP does not work on cb2s module

hackintoshlover, 2023-12-23:
### The problem

I have 2 channel wireless switch with tuyamcu which contains wifi module CB2S.
I successfuly added esphome and it works amazing with home assistant.
The only problem is when my network goes down or when i relocate the device, the device does not start AP mode automatically.
The same setup works perfectly with esp32 or esp8266 whenever it does not find the registered network, it automatically goes into AP mode. 
Kindly help with CB2S, since right now the only way to upload code in the device is recompiling the code with new wifi credentials and uploading it.
Also when i press factory reset nothing it just restarts the device, be it any device like esp32 or esp8266 or CB2S or anything else.

### Which version of ESPHome has the issue?

 LibreTiny Esphome 2023.9.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

BK72XX

### Board

CB2S

### Component causing the issue

captive portal

### Example YAML snippet

```yaml
esphome:
  name: zen-2c5w

bk72xx:
  board: generic-bk7231n-qfn32-tuya
  framework:
    version: dev

logger:

web_server:

captive_portal:

mdns:

api:
  password: ""

ota:
  password: "796dd4ed7d6958b4c1f37c92b05f50c4"

wifi:
  ssid: myssid
  password: mypassword
  ap:

button:
  - platform: restart
    name: Restart
  - platform: factory_reset
    name: Reset
    
debug:
  update_interval: 30s

text_sensor:
  - platform: debug
    reset_reason:
      name: Reset Reason
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: uptime
    name: Uptime

uart:
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 9600

tuya:
  # DPIDs processed from schema model: 000003p01m

switch:
  - platform: tuya
    switch_datapoint: 1
    name: Switch 1
  - platform: tuya
    switch_datapoint: 2
    name: Switch 2

binary_sensor:
  - platform: tuya
    sensor_datapoint: 103
    name: Pair Result 1
  - platform: tuya
    sensor_datapoint: 104
    name: Pair Result 2
  - platform: tuya
    sensor_datapoint: 110
    name: Clear Result 1
  - platform: tuya
    sensor_datapoint: 111
    name: Clear Result 2
```

szupi-ipuzs, 2023-12-31:
> ### Which version of ESPHome has the issue?
> 
> LibreTiny Esphome 2023.9.0-dev

You seem to be using some ancient version. Could you please try the latest one? Esphome now has (since version 2023.11.1) built-in support for libretiny, no need to use the forked version. I have 3 CB2S devices and they seem to work fine (ie. the fallback wifi shows up).
If it still does not work with the latest version, please submit this bug in the libretiny-eu/libretiny issues. Also note that there are some pending pull requests regarding wifi handling in libretiny, you can try them out if you know how (although I don't think they are relevant to this).

hackintoshlover, 2023-12-31:
I am using latest version of libretiny esphome for beken devices.
[image]

> Esphome now has (since version 2023.11.1) built-in support for libretiny.

I am using latest version of esphome
[image]

And i don't think so or see that esphome has built-in support for libretiny devices.

[image]

I assume both are different from each other.

I also figured out that when you reset the device or try fallback AP mode, it tries to connect to the ssid pre-programmed in memory.
Resetting does not clear that. It clears NVS flash but i guess the pre-programmed ssid and that is the reason why fallback AP was not working.
Now i just keep ssid and password fields blank and everything works as expected.

ssieb, 2023-12-31:
If you're going to use the libretiny fork (which is no longer maintained), then this is not the place to file issues.
The new device wizard doesn't have options for these devices yet, so just pick something, skip the install, and set it yourself.
https://esphome.io/components/libretiny

hackintoshlover, 2023-12-31:
Ok, i see.
I'll check and get back.

hackintoshlover, 2023-12-31:
That works.
Thanks.
Any idea on how to integrate bluetooth or zigbee based devices like bt3l or zt3l.

ssieb, 2023-12-31:
esphome doesn't support zigbee on any device at this time.  Bluetooth could be possible, but someone has to add the support to esphome.

closed 2023-12-31 by ssieb