esphome/issues #5285: Pin number can't be used in multiple places

Griffelinos, 2023-12-24:
### The problem

Until release 2023.12.0 it was possible to use the same pin number for a binary sensor and a pulsemeter sensor. But not anymore? 

### Which version of ESPHome has the issue?

2023.12.0 and forward

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP8266

### Board

d1_mini_pro

### Component causing the issue

binary_sensor.gpio: and sensor.pulse_meter:

### Example YAML snippet

```yaml
substitutions:
  friendly_name: ESP
  device_description: "Measure your natural gas consumption with the optic pulse of gas meter."
  pulse_pin: D1
  pulse_rate: "100" # imp/m3

esphome:
  name: "gas-measure-161"

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "a11b6d8a1e1cdf2dc3a79d3820d1f4ad"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Static IP - because mDNS not working in ESPHome
  manual_ip:
    static_ip: 192.168.1.161
    subnet: 255.255.255.0
    gateway: 192.168.1.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gas-Measure Fallback Hotspot"
    password: "GLhdIKNlmET3"

captive_portal:

binary_sensor:
  - platform: gpio
    pin:
      number: D1
      inverted: true
      mode:
        input: true
        pullup: true
    name: "Gas"
    filters:
      - delayed_on: 800ms
      - delayed_off: 800ms
    device_class: gas
    
sensor:

# Wifi signal 
  - platform: wifi_signal
    name: "Gas-Measure WiFI Signal"
    update_interval: 120s
    
# Gas meter
  - platform: pulse_meter
    pin:
      number: ${pulse_pin}
      mode: INPUT_PULLUP
    internal_filter_mode: PULSE
    internal_filter: 800ms
    name: "${friendly_name} - Gas consumption"
    filters:
      - multiply: 0.06
    unit_of_measurement: "m³/ min"
    id: gas
    accuracy_decimals: 2
    icon: 'mdi:fire'
    total:
      name: '${friendly_name} - Gas total'
      state_class: total_increasing
      device_class: gas
      unit_of_measurement: "m³"
      accuracy_decimals: 2
      filters:
        - lambda: return x * (1.0 / ${pulse_rate});
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.3
INFO Reading configuration /config/esphome/gas-measure-161.yaml...
Failed config

binary_sensor.gpio: [source <unicode string>:40]
  
  Pin 5 is used in multiple places.
  platform: gpio
  pin: 
    number: 5
    inverted: True
    mode: 
      input: True
      pullup: True
      output: False
      open_drain: False
      pulldown: False
      analog: False
  name: Gas
  filters: 
    - delayed_on: 800ms
    - delayed_off: 800ms
  device_class: gas
  disabled_by_default: False
sensor.pulse_meter: [source <unicode string>:61]
  
  Pin 5 is used in multiple places.
  platform: pulse_meter
  pin: 
    number: 5
    mode: 
      input: True
      pullup: True
      output: False
      open_drain: False
      pulldown: False
      analog: False
    inverted: False
  internal_filter_mode: PULSE
  internal_filter: 800ms
  name: ESP - Gas consumption
  filters: 
    - multiply: 0.06
  unit_of_measurement: m³/ min
  id: gas
  accuracy_decimals: 2
  icon: mdi:fire
  total: 
    name: ESP - Gas total
    state_class: total_increasing
    device_class: gas
    unit_of_measurement: m³
    accuracy_decimals: 2
    filters: 
      - lambda: !lambda |-
          return x * (1.0 / 100);
    disabled_by_default: False
    force_update: False
    icon: mdi:pulse
  disabled_by_default: False
  force_update: False
  state_class: measurement
  timeout: 5min
```

### Additional information

ESPhome release 2023.11.6 is working, using pin multible places

ssieb, 2023-12-24:
You need to read the changelog before deploying new releases.  You shouldn't be doing this.  What is that binary sensor even doing?  If you need help, come ask on the esphome discord server.

closed 2023-12-24 by ssieb

Griffelinos, 2023-12-24:
Hi ssieb. I know that I should read changelog before deploying new devices, but I have so many esp around the house that I dont remember what and how I programmed them. Lost track of my code long time ago. So normally I update and fix the problems.
The binary sensor only send a output every time my Gas counter have counted 1/1000 m3, i used it when I made the counter just to debug and see that the hardware worked. I guess I just can remove it. The reason that I posted this issue was only to contribute if it actually was a bug. Just tried to help out. Thanks for the feedback, I have joined the discord server. Have a great Christmas and new year, Best regards Lasse

sshalyminov, 2024-01-15:
Just add allow_other_uses: true in pin 
like this:
  - platform: gpio
    pin:
      number: D1
      inverted: true
      allow_other_uses: true
      mode:
        input: true
        pullup: true
    name: "Gas"

And the same to counter.

ssieb, 2024-01-15:
@sshalyminov that will make it compile, but it's generally not a good idea.  That's why this validation was added.