esphome/issues #5289: Tuya WiFi Temperature and Humidity Sensor

smavridis, 2023-12-25:
### The problem

I recently bought a few of 'Tuya WiFi Temperature and Humidity Sensor', just to try and make them work with ESPHome.

Opening the device revealed that it was a CBU based device.
Probing with the multimeter, i traced various connections and created the attached 'diagram' in the picture.

[image]

Using the findings i created a yaml and successfully flashed it (initially with serial and afterwards using the webui of ESPHome), the status LED and the button on the  back work.

The issue i am facing right now is related to the thermometer/hydrometer.
The marking are S-8310 i didn't find a data-sheet but , since the pin-outs seem to match, i  assumed it would be similar to its predecessor S8305 which is documented just to get started.

I attempted adding an I2C bus definition to my yaml (commented out) but it refuses to build is as
'Pin Mode is not supported'

I assume the hardware I2C is not yet supported(maybe that's a libretuya issue?), which is probably fine because the board seems to use a different pin for SDA (can't be sure because the HW SDA pin is beneath the CBU pcb and i don't want to de-solder the board for now).

I was hopping there was a software I2C implementation in ESPHome i could use, but i could not find anything existing ...

Finally i also tried using the tyua component, but it always reports an initialization failure so i have it commented out for now.

How should i proceed with this?

### Which version of ESPHome has the issue?

 2023.12.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

 2023.12.5

### What platform are you using?

ESP32-IDF

### Board

cbu

### Component causing the issue

api/i2c

### Example YAML snippet

```yaml
esphome:
  name: upk2esphome-bk7231n

bk72xx:
  board: cbu

logger:
  baud_rate: 0

# tuya:

web_server:

captive_portal:

mdns:

api:
  password: ""
  reboot_timeout: 0s

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.1.32
  ap:

button:
  - platform: restart
    name: Restart

debug:
  update_interval: 30s

text_sensor:
  - platform: debug
    reset_reason:
      name: Reset Reason
  - platform: libretiny
    version:
      name: LibreTiny Version
sensor:
  - platform: uptime
    name: Uptime
  - platform: adc
    pin: P23
    name: "Battery Level"

uart:
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 115200

light:
  - platform: binary
    name: "Status"
    output: status_output

output:
  - id: status_output
    platform: gpio
    pin: P16

binary_sensor:
  - platform: gpio
    pin: 
      number: P14
      inverted: true
    name: "Button"

# i2c:
#  sda: P21
#  scl: P22
#  scan: true
```

TheNextLVL, 2023-12-26:
Hi @smavridis ,

I have the same issue.
Searching I found that i2c isn't implemented, it's in the TODO list:

https://docs.libretiny.eu/docs/TODO/?h=i2c#rtl8710b
[image]

kifeo, 2024-01-12:
I have also the same issue with the white PCB one.
The sda pin is 8 and scl pin is 7, but there is the error message "this pin mode is not supported"

`i2c:
  sda: 8
  scl: 7`

and because the SHT3xd requires i2c, this is currently not possible to get the Temp and Humidity values :dagger: 
```
  - platform: sht3xd
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    address: 0x44
    update_interval: 60s
```

AzSaSiN, 2024-01-19:
I'm also been working on similar sensor:
Interested in i2c working so as deep sleep.

Otherwise this type of sensor would be useless.

shakin89, 2024-01-26:
For me it's the same, i'm trying to implement an i2c sensor in an BK7231N chipset, but it's still not implemented.

ssieb, 2024-01-26:
github issues are not for discussions like this.  Use the discord server or one of the forums.
i2c and deep sleep are not implemented in libretiny yet, so esphome can't use them, although someone could implement software i2c if they want.

closed 2024-01-26 by ssieb