esphome/issues #5292: Servo Restore Not Working

DaAwesomeP, 2023-12-26:
### The problem

Setting the `restore` value on a `servo` does not work. When running `servo.write` after boot, the servo does not move back to its remembered position.

The documentation says:
> **restore** (_Optional_, boolean): Whether to restore the state of the servo motor at startup. This is useful if you have an absolute servo motor and it goes back to its 0 position at startup. Defaults to `false`.
> https://esphome.io/components/servo.html

To reproduce:

1. Use YAML below
2. Run `servo.write` to set a position
3. Power off the ESP32
4. Move the servo
5. Power on the ESP32
6. Servo does not move back to remembered position at boot

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

m5stack-atom

### Component causing the issue

servo

### Example YAML snippet

```yaml
output:
  - platform: ledc
    id: blind_servo_pwm_output
    pin: GPIO25
    frequency: 50 Hz

servo:
  - id: blind_servo
    output: blind_servo_pwm_output
    transition_length: 2s
    auto_detach_time: 0.1s
    restore: True
```

### Anything in the logs that might be useful for us?

```txt
Not really

[22:03:50][C][servo:014]: Servo:
[22:03:50][C][servo:015]:   Idle Level: 7.5%
[22:03:50][C][servo:016]:   Min Level: 3.0%
[22:03:50][C][servo:017]:   Max Level: 12.0%
[22:03:50][C][servo:018]:   auto detach time: 100 ms
[22:03:50][C][servo:019]:   run duration: 2000 ms
```
```

ssieb, 2023-12-30:
Did you wait a minute after setting the position before powering it off?

DaAwesomeP, 2024-01-17:
@ssieb Yes, it jumps no matter how long it has been left (seconds, minutes, days).

closed 2024-03-14 by ssieb