esphome/issues #5294: sht3xd not connecting after latest update

deanfourie1, 2023-12-26:
### The problem

The sht3xd I2C sensor no longer works on my weather station. This was as a direct result of the latest esphome update.

The sht3xd still shows in the I2C scan.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

ESP32MiniKit

### Component causing the issue

sht3xd

### Example YAML snippet

```yaml
esphome:
  name: weather-station
  friendly_name: weather-station
  

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key:

ota:
  password:

i2c:
  sda: 21
  scl: 22
  scan: True

sun:
  latitude:
  longitude:

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    bssid: 
  fast_connect: true

[... 500 lines cut ...]
[22:02:37][D][sensor:094]: '${friendly_name} rain gauge': Sending state 0.00000 mm with 4 decimals of accuracy
[22:02:37][D][sensor:094]: '${friendly_name} rainfall per min': Sending state 0.00000 mm with 6 decimals of accuracy
[22:02:37][D][sensor:094]: '${friendly_name} total daily rain': Sending state 17.20008 mm with 6 decimals of accuracy
[22:02:37][D][sensor:094]: 'Wind Direction': Sending state 264.01965  with 1 decimals of accuracy
[22:02:38][D][sensor:094]: '${friendly_name} wind speed': Sending state 0.28102 m/s with 2 decimals of accuracy
[22:02:38][D][sensor:094]: '${friendly_name} wind speed (km/h)': Sending state 1.01166 km/h with 2 decimals of accuracy
[22:02:39][D][mlx90393:070]: received -3569.159912 219.029999 -13625.919922
[22:02:39][D][sensor:094]: 'mlx_x': Sending state -3569.15991 µT with 0 decimals of accuracy
[22:02:39][D][sensor:094]: 'mlx_y': Sending state 219.03000 µT with 0 decimals of accuracy
[22:02:39][D][sensor:094]: 'mlx_z': Sending state -13625.91992 µT with 0 decimals of accuracy
[22:02:39][W][component:214]: Component mlx90393.sensor took a long time for an operation (0.17 s).
[22:02:39][W][component:215]: Components should block for at most 20-30ms.
[22:02:39][D][sensor:094]: '${friendly_name} wind speed average': Sending state 0.28102 m/s with 2 decimals of accuracy
[22:02:39][D][sensor:094]: '${friendly_name} wind speed average (km/h)': Sending state 1.01166 km/h with 2 decimals of accuracy
[22:02:39][D][text_sensor:064]: '${friendly_name} Beaufort wind scale code': Sending state '1'
[22:02:39][D][text_sensor:064]: '${friendly_name} Beaufort wind scale': Sending state 'Light Air'
[22:02:39][D][sensor:094]: 'Wind Direction': Sending state 263.98831  with 1 decimals of accuracy
[22:02:39][D][sensor:094]: '${friendly_name} wind speed': Sending state 2.45861 m/s with 2 decimals of accuracy
[22:02:39][D][sensor:094]: '${friendly_name} wind speed (km/h)': Sending state 8.85101 km/h with 2 decimals of accuracy
[22:02:40][D][sensor:094]: '${friendly_name} wind speed': Sending state 1.66766 m/s with 2 decimals of accuracy
[22:02:40][D][sensor:094]: '${friendly_name} wind speed (km/h)': Sending state 6.00359 km/h with 2 decimals of accuracy
[22:02:41][E][mlx90393:085]: failed to read data
[22:02:41][W][component:214]: Component mlx90393.sensor took a long time for an operation (0.14 s).
[22:02:41][W][component:215]: Components should block for at most 20-30ms.
```

alexruffell, 2023-12-27:
It appears I have the same issue (communication failure with sensor) but only if frequency of the i2c bus is anything faster than the default 50kHz. Before this update (or one of the previous ones these last couple months) it was running at 400kHz without issues.

deanfourie1, 2023-12-27:
I fixed this issue by removing the sensor from my config, saving and uploading. 

Then re adding the config for the sht3xd and saving and uploading again.

This solved the issue for me. Very strange.

alexruffell, 2023-12-27:
@deanfourie1 I opened another issue as in my case whether an i2c frequency of 400khz works or not depends on the framework. With `arduino` I can set the frequency to 400khz (did not try anything higher) while with `esp-idf` I cannot exceed 50 to 150khz depending on node I test it on.

https://github.com/esphome/issues/issues/5303

j-stahlhut, 2024-04-20:
As @deanfourie1 wrote:
> I fixed this issue by removing the sensor from my config, saving and uploading.
> Then re adding the config for the sht3xd and saving and uploading again.
I just have the same issue after upgrade to ESPHome 2024.4.0 - 17th April 2024
Thanx for the solutions

mrtoy-me, 2024-05-03:
External link I suggested in issue#5710 is now part of release 2024.4.2

tymscar, 2024-05-03:
It is an I can confirm it's working