esphome/issues #5296: ESP32-S3-BOX-3 Voice assistant stability issue when i2c is activated

moonilightshaker, 2023-12-26:
### The problem

Hello,
Voice Assistant is installed on the ESP32-S3-BOX-3.
After activating i2c. The Voice Assistant pipeline no longer activates after a few hours. A factory reset does not help. Only deactivating the i2c components and reflashing helps.

The Red Button under the Display does not work either. 

Best regards,
M.

### Which version of ESPHome has the issue?

2023.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-BOX-3

### Component causing the issue

Voice assistant

### Example YAML snippet

```yaml
substitutions:
  name: esp32-s3-box-3-5ace4c
  friendly_name: ESP32 S3 Box 3
  loading_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/loading_320_240.png
  idle_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/idle_320_240.png
  listening_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/listening_320_240.png
  thinking_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/thinking_320_240.png
  replying_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/replying_320_240.png
  error_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/error_320_240.png

  loading_illustration_background_color: '000000'
  idle_illustration_background_color: '000000'
  listening_illustration_background_color: 'FFFFFF'
  thinking_illustration_background_color: 'FFFFFF'
  replying_illustration_background_color: 'FFFFFF'
  error_illustration_background_color: '000000'

  voice_assist_idle_phase_id: '1'
  voice_assist_listening_phase_id: '2'
  voice_assist_thinking_phase_id: '3'
  voice_assist_replying_phase_id: '4'
  voice_assist_not_ready_phase_id: '10'
  voice_assist_error_phase_id: '11'  
  voice_assist_muted_phase_id: '12'
  #wifi_update_interval: 3s

#packages:
#  esphome.voice-assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
[... 695 lines cut ...]
[12:14:58][D][esp-idf:000]: I (14872) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1

[12:14:58][D][esp-idf:000]: I (14876) RSP_FILTER: sample rate of source data : 16000, channel of source data : 2, sample rate of destination data : 16000, channel of destination data : 1

[12:14:58][D][esp-idf:000]: I (14879) AUDIO_PIPELINE: Pipeline started

[12:14:58][D][esp_adf.microphone:294]: Microphone started
[12:14:58][D][voice_assistant:422]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[12:14:58][D][voice_assistant:176]: Waiting for speech...
[12:14:58][D][voice_assistant:422]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[12:15:05][D][sensor:094]: 'Battery voltage': Sending state 3.56853 V with 3 decimals of accuracy
[12:15:05][D][sensor:094]: 'Battery level': Sending state 62.85499 % with 1 decimals of accuracy
[12:15:27][D][aht10:097]: AHT10 is busy, waiting...
[12:15:27][D][sensor:094]: 'Temperature': Sending state 38.69686 °C with 2 decimals of accuracy
[12:15:27][D][sensor:094]: 'Humidity': Sending state 22.14909 % with 2 decimals of accuracy
[12:15:27][W][component:214]: Component aht10.sensor took a long time for an operation (0.07 s).
[12:15:27][W][component:215]: Components should block for at most 20-30ms.
[12:15:31][D][esp32.preferences:114]: Saving 1 preferences to flash...
[12:15:31][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[12:15:35][D][sensor:094]: 'Battery voltage': Sending state 3.57047 V with 3 decimals of accuracy
[12:15:35][D][sensor:094]: 'Battery level': Sending state 62.96910 % with 1 decimals of accuracy
[12:15:37][D][esp-idf:000]: I (54079) tsens: Config temperature range [-10°C ~ 80°C], error < 1°C

[12:15:37][D][sensor:094]: 'Internal Temperature': Sending state 40.05940 °C with 1 decimals of accuracy
```

BigBobbas, 2023-12-27:
can form that having 12c: component in logs causing pipeline to become unresponsive. Touch and button do not function. Serial logs show following errors
[I][esp_adf:019]: Start codec chip
[D][esp-idf:000]: E (172) i2c: i2c driver install error

[D][esp-idf:000]: E (172) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (173) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (174) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (176) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (177) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (178) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (179) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (180) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (181) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (182) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (183) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (184) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: I (185) DRV8311: ES8311 in Slave mode

[D][esp-idf:000]: E (185) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (187) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (188) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (189) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (190) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (191) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (192) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (193) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (194) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (195) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (196) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (197) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (198) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (199) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (200) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (201) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (202) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (203) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (204) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (205) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[... 179 lines cut ...]

[D][esp-idf:000]: E (299) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (300) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (301) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (302) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (303) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:130 (i2c_bus_read_bytes):Handle error

[D][esp-idf:000]: E (304) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (305) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (306) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (307) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (308) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][esp-idf:000]: E (309) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:89 (i2c_bus_write_bytes):Handle error

[D][ili9xxx:021]: Setting up ILI9xxx
[D][esp-idf:000]: I (311) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

anekinloewe, 2023-12-27:
Hi all.
Same here. I would say it is due to the GT911 device. I played around with the AHT10 for several hours. Only when I added the GT911 did VAD no longer work. Simply commenting out GT911 is not enough. You have to comment out the entire i2c. Then VAD will work again.

clydebarrow, 2023-12-29:
I think the ADF is using the i2c and the ESPHome i2c component is not playing nicely with it. Without ADF the gt911 works just fine.

anekinloewe, 2023-12-29:
> I think the ADF is using the i2c and the ESPHome i2c component is not playing nicely with it. Without ADF the gt911 works just fine.

Not exactly, without i2c component ADF works fine. As soon as the ic2 component is added, the adf is broken. However, you only notice this after a power reset because ES8311 and ES7210 obviously save the settings until the power reset.

ChristophCaina, 2023-12-29:
ok, that's interesting...

I have two S3 Boxes... one is the S3-Box, and the other one is the S3-Box3 ...
for the Box3 I was using the i2c to get the AHT Sensor working, but "suddenly" VA stopped working.

Also, it appeard, that this was somehow affecting the other Box as well (configured with a different wake word)... I don't know, how it could be, that the older S3-Box did not work... but now, after removing the i2c from the Box3 config, both S3-Boxes are working again with their configured Wake-Word.

But yes, at least, I can confirm an issue when using the i2c on the Box3

jaymunro, 2024-01-05:
The conflict became apparent to me last month with the microphone not responding after a few minutes so I commented out the I2C code. However an idea occurred to me while enabling the radar that the mic and such are on bus_a and we shouldn't be doing anything on this bus, so I added `scan: false` and it has been working for 10+ minutes so far.

full code is:
```
i2c:
  - id: bus_a
    sda: GPIO08
    scl: GPIO18
    scan: false

  - sda: GPIO41
    scl: GPIO40
    id: bus_b

sensor:
  - platform: aht10
    i2c_id: bus_b
    variant: AHT20
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s
```
Can anyone else confirm this works for them too?

ChristophCaina, 2024-01-05:
> Can anyone else confirm this works for them too?

Can you try to powercycle the device?
As mentioned by @anekinloewe it seems to be related with a PWR Reset.

I cannot do any tests until today in the afternoon, but will do then.

moonilightshaker, 2024-01-05:
> The conflict became apparent to me last month with the microphone not responding after a few minutes so I commented out the I2C code. However an idea occurred to me while enabling the radar that the [...]

Will test it. I report back later.

moonilightshaker, 2024-01-06:
> > Can you try to powercycle the device?
> 
> Confirmed. Powercycling the device results in mic not working when i2c is enabled. `scan: false` does not work.

Same as mine. A power cycle after 16 hours bricked the device.

jaymunro, 2024-01-06:
> Same as mine. A power cycle after 16 hours bricked the device.

That's bad. Mine didn't brick. Just mic and maybe other VA stuff stopped working. Connection was still working and logs still live.

What are you gonna do with your new paperweight? 😟

moonilightshaker, 2024-01-06:
> What are you gonna do with your new paperweight? 😟

A reflesh did the trick. VA stuff is not working. The Sensors and Connection is working.

BigBobbas, 2024-01-16:
has anyone been able to get anywhere with this ? having the the touchscreen and voice working would be great!

X-Ryl669, 2024-02-02:
From [there](https://github.com/espressif/esp-adf/blob/48dea7711391e0abf00c65aca18557572fc5189a/components/audio_hal/driver/es8311/es8311.c) it seems that the expected I2C bus configuration for ADF is to run at 100KHz with both pullup enabled. 
By default, ESPHome runs its I2C bus at 50KHz (see [here](https://github.com/esphome/esphome/blob/dev/esphome/components/i2c/__init__.py) around line 59). So it's possible that the ADF is breaking the I2C configuration for the bus (or ESPHome) by both configuring conflicting frequency for the bus. 

I believe the driver underneath in esp-idf will ignore this conflict if the parameters are the same.

Can you try to use this in your YAML?
```yaml
i2c:
  - id: bus_a
    sda: GPIO08
    scl: GPIO18
    scan: false
    sda_pullup_enabled: true
    scl_pullup_enabled: true
    frequency: 100kHz

  - sda: GPIO41
    scl: GPIO40
    scan: false
    sda_pullup_enabled: true
    scl_pullup_enabled: true
    frequency: 100kHz
    id: bus_b

```

BigBobbas, 2024-02-02:
This appears to work for the screen at least , not tried temp, hum and button but the screen work! genius !!!!

X-Ryl669, 2024-02-02:
Indeed, the code in esp-adf (that's imported from PR#5230 from @jesserockz) doesn't care about sharing its I2C handle (see [here](https://github.com/espressif/esp-adf/blob/48dea7711391e0abf00c65aca18557572fc5189a/components/audio_hal/driver/es7210/es7210.c), the handle is static to this file, thus can't be shared with any other. 
It's using `i2c_bus_create` on the I2C0. 

Hopefully, this function (see [here](https://github.com/espressif/esp-iot-solution/blob/6359293fbe0727767bd3431c26e33044c3144057/components/bus/i2c_bus.c#L79) checks if it's being called with the same parameters and in that case, returns the same handle. Hence, if we had the same configuration, we should have received the same handle.

However, in that case, we should have received a warning message like this: `i2c0 has been inited, return handle directly, ref_counter=1` in the logs (which isn't here), so somehow the configuration is different or not seen as the same.

Indeed, ESPHome doesn't use the `i2c_bus_create` function, but is more low level, using the `i2c_driver_install` directly. So we can't actually share the I2C bus with ESP-ADF. This is going to be problematic for having a functional ESP32BOX (3) on ESPHome.

ESP-ADF use [this](https://github.com/espressif/esp-adf/blob/48dea7711391e0abf00c65aca18557572fc5189a/components/esp_peripherals/driver/i2c_bus/i2c_bus.c#L78C1-L79C1) installation mode (notice the `0, 0, 0)` for the last parameters) while ESPHome use [this](https://github.com/esphome/esphome/blob/0fa0904bc53aa72e2ab291ba0840e73738ebef18/esphome/components/i2c/i2c_bus_esp_idf.cpp#L48) (notice the `0, 0, ESP_INTR_FLAG_IRAM)` for the last parameters).

Maybe it's time to update the ESPHome I2C implementation to follow more the espressif directions so we can share components from espressif's ADF format.

anekinloewe, 2024-03-04:
@X-Ryl669 : I have tried your pr with the touch screen component but I can't get it work. May be I use it wrong. 

```
....
 external_components:
  - source: github://pr#5230
    components: esp_adf
    refresh: 0s
  - source: github://pr#6313
    components: i2c
    refresh: 0s

i2c:
  - id: bus_0
    sda: GPIO08
    scl: GPIO18
    scan: false
    sda_pullup_enabled: true
    scl_pullup_enabled: true
    frequency: 100kHz

  - sda: GPIO41
    scl: GPIO40
    scan: false
    sda_pullup_enabled: true
    scl_pullup_enabled: true
    frequency: 100kHz
    id: bus_b

touchscreen:
  platform: gt911
  id: my_touchscreen
  interrupt_pin: GPIO3
  i2c_id: bus_0
  on_touch:
    - if:
        condition:
          display.is_displaying_page: idle_page
        then:
          - display.page.show: sensor_page
          - component.update: s3_box_lcd
        else:
          - display.page.show: idle_page
          - component.update: s3_box_lcd
```
```
src/esphome/components/i2c/i2c_bus_esp_adf.cpp: In destructor 'esphome::i2c::ScopeLock::~ScopeLock()':
src/esphome/components/i2c/i2c_bus_esp_adf.cpp:30:31: error: 'bus' was not declared in this scope
   ~ScopeLock() { mutex_unlock(bus->bus_lock); }
                               ^~~
src/esphome/components/i2c/i2c_bus_esp_adf.cpp:30:31: note: suggested alternative: 'bus_'
   ~ScopeLock() { mutex_unlock(bus->bus_lock); }
                               ^~~
                               bus_

```

X-Ryl669, 2024-03-23:
Ok, can you try with this YAML (please adapt the substitutions to fit your needs) and report:
```yaml
substitutions:
  name: esp32-s3-box-3
  friendly_name: ESP32 S3 Box 3 Voice Assist
  loading_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/loading_320_240.png
  idle_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/idle_320_240.png
  listening_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/listening_320_240.png
  thinking_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/thinking_320_240.png
  replying_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/replying_320_240.png
  error_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/error_320_240.png

  loading_illustration_background_color: "000000"
  idle_illustration_background_color: "000000"
  listening_illustration_background_color: "FFFFFF"
  thinking_illustration_background_color: "FFFFFF"
  replying_illustration_background_color: "FFFFFF"
  error_illustration_background_color: "000000"

  voice_assist_idle_phase_id: "1"
  voice_assist_listening_phase_id: "2"
  voice_assist_thinking_phase_id: "3"
  voice_assist_replying_phase_id: "4"
  voice_assist_not_ready_phase_id: "10"
  voice_assist_error_phase_id: "11"
  voice_assist_muted_phase_id: "12"

  wifi_ssid: !secret wifi_ssid # add in your esphome secrets file.
  wifi_password: !secret wifi_password # add in your esphome secrets file. -> per default this is also used for ota_password and web_password
  ota_password: !secret ota_password # add in your esphome secrets file. - manual change in code required to activate
  web_password: !secret wifi_password # add in your esphome secrets file. - manual change in code required to activate
  api_password: !secret api_key # add in your esphome secrets file. - manual change in code required to activate  
 
  # ## static ip config ##
  ip: "192.168.1.170"
  gw: "192.168.1.254"
  subnet: "255.255.255.0"
  dns: "192.168.1.254"

external_components:
  - source: github://pr#5894 #5198
    components: [ahtxx]
    refresh: 60s
  - source: github://X-Ryl669/esphome@esp-adf-take2 #pr#5230
    components: [esp_adf]
    refresh: 0s
#  - source: github://X-Ryl669/esphome-ms58
  - source: github://X-Ryl669/esphome@at581x
    components: at581x
    refresh: 0s

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
    version: "1.0"
  min_version: 2023.11.5
  on_boot:
    - priority: 200
      then:
        - at581x.settings:
            id: "Waveradar"
            hw_frontend_reset: false
            frequency: 5800Mhz  
            sensing_distance: 200 # 0-1023
            poweron_selfcheck_time: 2s
[... 410 lines cut ...]
    type: aht30
    i2c_id: bus_a
    id: aht_20
    address: 0x38
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 20s
  

logger:
  # CDC logging is currently not working on the S3
  # See https://github.com/espressif/esp-idf/commit/9924d9f27192a5fab5f66230c72249ecc6cad34c
  #hardware_uart: USB_CDC
  #level: VERY_VERBOSE
  #level: INFO

# Enable Home Assistant API
web_server:
  port: 80
  auth:
    username: admin
    password: ${web_password}
```

anekinloewe, 2024-03-24:
In the first try the mic, radar and aht30 work fine. Touch and speaker have a problem. I need a few days to test more.

anekinloewe, 2024-03-27:
ok, I have finished the tests. The speaker still does not work and as soon as the 2nd ic2 bus is used, e.g. for the AHT30, the touchscreen no longer works. So either voice assistant or i2c devices.

X-Ryl669, 2024-03-27:
I don't think the touchscreen is on the same bus as the other sensors. So there's no reason it would interfere with the other sensors. For the speaker, I've identified the issue and I'm working on fixing it (should be there soon).

anekinloewe, 2024-03-28:
> I don't think the touchscreen is on the same bus as the other sensors. So there's no reason it would interfere with the other sensors. For the speaker, I've identified the issue and I'm working on fixing it (should be there soon).

Cool! I will test it again. May be I have done something wrong with the I2C configuration.

X-Ryl669, 2024-04-05:
Ok, it works for me for all features with this YAML:
```yaml
substitutions:
  name: esp32-s3-box-3
  friendly_name: ESP32 S3 Box 3 Voice Assist
  loading_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/loading_320_240.png
  idle_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/idle_320_240.png
  listening_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/listening_320_240.png
  thinking_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/thinking_320_240.png
  replying_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/replying_320_240.png
  error_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/error_320_240.png

  loading_illustration_background_color: "000000"
  idle_illustration_background_color: "000000"
  listening_illustration_background_color: "FFFFFF"
  thinking_illustration_background_color: "FFFFFF"
  replying_illustration_background_color: "FFFFFF"
  error_illustration_background_color: "000000"

  voice_assist_idle_phase_id: "1"
  voice_assist_listening_phase_id: "2"
  voice_assist_thinking_phase_id: "3"
  voice_assist_replying_phase_id: "4"
  voice_assist_not_ready_phase_id: "10"
  voice_assist_error_phase_id: "11"
  voice_assist_muted_phase_id: "12"

  wifi_ssid: !secret wifi_ssid # add in your esphome secrets file.
  wifi_password: !secret wifi_password # add in your esphome secrets file. -> per default this is also used for ota_password and web_password
  ota_password: !secret ota_password # add in your esphome secrets file. - manual change in code required to activate
  web_password: !secret wifi_password # add in your esphome secrets file. - manual change in code required to activate
  api_password: !secret api_key # add in your esphome secrets file. - manual change in code required to activate  
 
  # ## static ip config ##
  ip: "192.168.1.170"
  gw: "192.168.1.254"
  subnet: "255.255.255.0"
  dns: "192.168.1.254"

external_components:
  - source: github://pr#5894 #5198
    components: [ahtxx]
    refresh: 60s
  - source: github://X-Ryl669/esphome@esp-adf-take2 #pr#5230
    components: [esp_adf]
    refresh: 0s
  - source: github://X-Ryl669/esphome@at581x
    components: at581x
    refresh: 0s

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
    version: "1.0"
  min_version: 2023.11.5
  on_boot:
    - priority: 200
      then:
        - at581x.settings:
            id: "Waveradar"
            hw_frontend_reset: false
            frequency: 5800Mhz  
            sensing_distance: 200 # 0-1023
            poweron_selfcheck_time: 2s
            protect_time: 1s 
[... 438 lines cut ...]
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 20s
  

logger:
  # CDC logging is currently not working on the S3
  # See https://github.com/espressif/esp-idf/commit/9924d9f27192a5fab5f66230c72249ecc6cad34c
  #hardware_uart: USB_CDC
  #level: VERY_VERBOSE
  #level: INFO

# Enable Home Assistant API
web_server:
  port: 80
  auth:
    username: admin
    password: ${web_password}
```

Tested and working: Touchscreen, Radar, Speaker, Mic, Temperature/Humidity sensor.
Please confirm it works for you too.

In the current YAML, I've scripted the LCD backlight to switch on only upon radar detection, or touch or voice activity detection.

anekinloewe, 2024-04-07:
@X-Ryl669 : I tried it but it does not compile.
```
...
Compiling .pioenvs/s3-box-3-01/src/esphome/components/esp_adf/speaker/esp_adf_speaker.cpp.o
Compiling .pioenvs/s3-box-3-01/src/esphome/components/factory_reset/button/factory_reset_button.cpp.o
Compiling .pioenvs/s3-box-3-01/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling .pioenvs/s3-box-3-01/src/esphome/components/gt911/touchscreen/gt911_touchscreen.cpp.o
src/esphome/components/esp_adf/i2c_bus_esp_adf.cpp:19:13: warning: 'void esphome::i2c::recover_i2c_hard(i2c_port_t, void*)' defined but not used [-Wunused-function]
 static void recover_i2c_hard(i2c_port_t, void *bus) { ((ADFI2CBus *) bus)->recover_(); }
             ^~~~~~~~~~~~~~~~
Compiling .pioenvs/s3-box-3-01/src/esphome/components/i2c/i2c.cpp.o
src/esphome/components/esp_adf/speaker/esp_adf_speaker.cpp: In static member function 'static void esphome::esp_adf::ESPADFSpeaker::player_task(void*)':
src/esphome/components/esp_adf/speaker/esp_adf_speaker.cpp:177:39: error: 'filter' was not declared in this scope
   audio_pipeline_unregister(pipeline, filter);
                                       ^~~~~~
src/esphome/components/esp_adf/speaker/esp_adf_speaker.cpp:177:39: note: suggested alternative: 'fileno'
   audio_pipeline_unregister(pipeline, filter);
                                       ^~~~~~
                                       fileno
Compiling .pioenvs/s3-box-3-01/src/esphome/components/i2c/i2c_bus_arduino.cpp.o
Compiling .pioenvs/s3-box-3-01/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o
Compiling .pioenvs/s3-box-3-01/src/esphome/components/ili9xxx/ili9xxx_display.cpp.o
Compiling .pioenvs/s3-box-3-01/src/esphome/components/image/image.cpp.o
*** [.pioenvs/s3-box-3-01/src/esphome/components/esp_adf/speaker/esp_adf_speaker.cpp.o] Error 1

```

BigBobbas, 2024-04-07:
have you tried cleaning build files

anekinloewe, 2024-04-07:
> have you tried cleaning build files
Yes. Just tried it again to be really sure.

X-Ryl669, 2024-04-08:
Sorry, please try again, I forgot to push my changes to the github repository.

X-Ryl669, 2024-04-09:
Please try again. I've fixed some CI stuff. It builds for me, starting from scratch.

briodan, 2024-04-09:
I was able to build successfully and it seems to be working, will be testing more.

anekinloewe, 2024-04-10:
I was able to build it successfully. For the first test everything seems to work fine. I will test more in the next days and will report if I found something.
@X-Ryl669 Thank you very much!

DrShivang, 2024-04-11:
@X-Ryl669  Hi,
Tested & It works perfectly before reboot

After reboot the options greyed out
[image]

the display works when motion is detected (great feature) but I was expecting if it can also remain on after wake word detection and processing (that way at least it appears its working)

will report as I test further..

X-Ryl669, 2024-04-11:
Thanks Dr Shivang for reporting.

The options are greyed out I think until the box connects to the VA. It shouldn't stay grayed.

I've added the option to turn the backlight off after inactivity. It's turned on upon motion, as you said, but also after wake word is detected (not after VAD). You can change the time before it turns off.

Look at:
```
voice_assistant:
 [...]
  on_stt_vad_start:
    then:
      - light.turn_on: led
      - delay: 5s
      - light.turn_off: led
```

You could move the `light.turn_off` part to other hooks (like `on_stt_vad_end` or `on_tts_stream_end`).

DrShivang, 2024-04-12:
@jaymunro , Thanks for the update I'll test and revert.

Also see this great achievement by @X-Ryl669 at https://github.com/esphome/issues/issues/5296
All the sensors are functioning well and tested.

I'll try if we can integrate both of these together.

Drewsky208, 2024-05-09:
I want to thank you for this thread!  I got an ESP32-S3-BOX-3B and the audio input was completely broken using the standard methods.  Switching to this YAML in this thread made it work!  Even though the 3B is supposed to be electronically the same I know many have troubles making it work, so if you arrived here from a search, you are on the right track.  Use this YAML and you are on your way.

ghedo, 2024-05-28:
@X-Ryl669 thank you very much for this work!

I originally tried your config with esphome `2024.5.3` but it didn't seem to work at all as the device didn't show any sign of life (wouldn't connect to WiFi, couldn't get logs via USB, ...) and only after downgrading to esphome `2024.3.2` and flashing again it started working properly.

I imagine there might have been some incompatible change in between those releases, any chance your config/changes could be made to work with latest esphome versions? Also considering your `at581x` PR has been merged in the meantime it could simplify things a bit.

X-Ryl669, 2024-05-28:
I'm currently adding support for capturing and saving the utterance to the SD card upon button press so we can have real voice samples for the wake word to train.
Right now, I'm not trying to keep up with ESPHome last release (I have numerous issue with the latest version, the webserver doesn't work on all my ESP8266 and Libretiny devices, which break my setup). I'm waiting until it stabilize from ESPHome before trying to upgrade.

maxi1134, 2024-05-31:
> Ok, it works for me for all features with this YAML:
>
> ```yaml
> substitutions:
>   name: esp32-s3-box-3
>   friendly_name: ESP32 S3 Box 3 Voice Assist
>   loading_illustration_file: [...]

Hi there!

I am trying this config on a ESP32-S3-Box-3 but get a black screen and no connection to the logs through USB. this is my code:

```
substitutions:  
  micro_wake_word_model: alexa
  name: esp32-s3-box-3
  friendly_name: ESP32 S3 Box 3 Voice Assist
  loading_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/loading_320_240.png
  idle_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/idle_320_240.png
  listening_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/listening_320_240.png
  thinking_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/thinking_320_240.png
  replying_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/replying_320_240.png
  error_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/error_320_240.png

  loading_illustration_background_color: "000000"
  idle_illustration_background_color: "000000"
  listening_illustration_background_color: "FFFFFF"
  thinking_illustration_background_color: "FFFFFF"
  replying_illustration_background_color: "FFFFFF"
  error_illustration_background_color: "000000"

  voice_assist_idle_phase_id: "1"
  voice_assist_listening_phase_id: "2"
  voice_assist_thinking_phase_id: "3"
  voice_assist_replying_phase_id: "4"
  voice_assist_not_ready_phase_id: "10"
  voice_assist_error_phase_id: "11"
  voice_assist_muted_phase_id: "12"

  wifi_ssid: !secret wifi_ssid # add in your esphome secrets file.
  wifi_password: !secret wifi_password # add in your esphome secrets file. -> per default this is also used for ota_password and web_password
  ota_password: !secret ota_password # add in your esphome secrets file. - manual change in code required to activate
  web_password: !secret wifi_password # add in your esphome secrets file. - manual change in code required to activate
  api_password: !secret api_key_voice_assistant # add in your esphome secrets file. - manual change in code required to activate  
 
  # ## static ip config ##
  ip: "192.168.0.230"
  gw: "192.168.0.1"
  subnet: "255.255.255.0"
  dns: "192.168.0.10"

external_components:
  - source: github://pr#5894 #5198
    components: [ahtxx]
    refresh: 60s
  - source: github://X-Ryl669/esphome@esp-adf-take2 #pr#5230
    components: [esp_adf]
    refresh: 0s
  - source: github://X-Ryl669/esphome@at581x
    components: at581x
    refresh: 0s

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
[... 445 lines cut ...]
    type: aht30
    i2c_id: bus_a
    id: aht_20
    address: 0x38
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 20s
  

logger:
  # CDC logging is currently not working on the S3
  # See https://github.com/espressif/esp-idf/commit/9924d9f27192a5fab5f66230c72249ecc6cad34c
  #hardware_uart: USB_CDC
  #level: VERY_VERBOSE
  #level: INFO

# Enable Home Assistant API
web_server:
  port: 80
  auth:
    username: admin
    password: ${web_password}
    ```

MHeuvel, 2024-07-19:
I tried using the temperature sensor with the newest esphome 2024.07 and it still has problems that the microphone doesn't work after a reset. I was under the impression this was fixed?