esphome/issues #5297: ESPHome container keeps hammering device with pings

ThinkPadNL, 2023-12-26:
### The problem

I am running ESPHome in a Docker container (`ghcr.io/esphome/esphome:latest`). As my IOT-devices are in a separate VLAN i have disabled mdns in the config of the devices) and have set `ESPHOME_DASHBOARD_USE_PING` to `true` in the container env. 

What i notice in my firewall logging is that the container keeps hammering the devices with ping. Much more often than my likings:
[image]

In the firewall ping is allowed. I made a Wireshark and also see request and after that a reply. So nothing is blocked.

I notice that when i stop the container, start it, no pings are happening yet. When i open the dashboard the pings start.
But after that they keep running.

What drew my attention is that in the container logs i see requests lines
2023-12-26 14:27:56,404 INFO 304 GET /devices (192.168.4.40) 0.89ms
2023-12-26 14:28:01,413 INFO 304 GET /devices (192.168.4.40) 0.94ms
2023-12-26 14:28:07,411 INFO 304 GET /devices (192.168.4.40) 0.91ms
**2023-12-26 14:28:10,237 INFO 200 GET /version (127.0.0.1) 0.50ms**
2023-12-26 14:28:13,429 INFO 304 GET /devices (192.168.4.40) 0.91ms
2023-12-26 14:28:19,409 INFO 304 GET /devices (192.168.4.40) 0.95ms
2023-12-26 14:28:25,416 INFO 304 GET /devices (192.168.4.40) 1.10ms
2023-12-26 14:28:31,417 INFO 304 GET /devices (192.168.4.40) 1.21ms
2023-12-26 14:28:36,409 INFO 304 GET /devices (192.168.4.40) 0.82ms
**2023-12-26 14:28:40,314 INFO 200 GET /version (127.0.0.1) 0.49ms**
2023-12-26 14:28:42,404 INFO 304 GET /devices (192.168.4.40) 0.80ms
2023-12-26 14:28:48,410 INFO 304 GET /devices (192.168.4.40) 0.92ms
2023-12-26 14:28:53,411 INFO 304 GET /devices (192.168.4.40) 0.99ms
2023-12-26 14:28:59,424 INFO 304 GET /devices (192.168.4.40) 0.81ms

4.40 is my laptop. But why is the container requesting it self? Could this be causing the ping test not coming to a pause?

### Which version of ESPHome has the issue?

2023.12.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

multiple

### Anything in the logs that might be useful for us?

```txt
2023-12-26 14:25:39,856 INFO 200 GET /version (127.0.0.1) 0.56ms
2023-12-26 14:26:40,014 INFO 200 GET /version (127.0.0.1) 0.51ms
2023-12-26 14:26:09,935 INFO 200 GET /version (127.0.0.1) 0.51ms
```

akail, 2023-12-26:
I just noticed the same thing after updating esphome a few days ago from 2023.11 to 2023.12.  Nearly brought down my firewall today filling up /var/log.  

Over the course of 24 hours Esphome has sent out 29 Million pings to only 6 devices.

CarlosGS, 2023-12-26:
Should have been fixed by https://github.com/esphome/esphome/pull/6001
Can you check again with version 2023.12.5?

ThinkPadNL, 2023-12-27:
Yes i can confirm that after updating the container to 2023.12.5 the issue has been fixed. No more pings after i have closed the browser tab with the dashboard.
Also the pings are sent in a much lower rate, every 2s instead of multiple per second continuously.

closed 2023-12-27 by ThinkPadNL