esphome/issues #5298: Template Number error

lordzid, 2023-12-26:
### The problem

component template number in the latest update compilation error

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

 Nodemcu

### Component causing the issue

Template Number

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "VBDZNTcoIKAx"

captive_portal:

number:                              #выбор мелодии
  - platform: template
    name: "MP3 select"
    id: select
    optimistic: true
    min_value: 0
    max_value: 100
    step: 1
```

### Anything in the logs that might be useful for us?

[... 330 lines cut ...]
           ^~~~~~
src/main.cpp:176:11: error: request for member 'traits' in '* select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
   select->traits.set_mode(number::NUMBER_MODE_AUTO);
           ^~~~~~
src/main.cpp:177:11: error: request for member 'set_optimistic' in '* select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
   select->set_optimistic(true);
           ^~~~~~~~~~~~~~
src/main.cpp:178:11: error: request for member 'set_initial_value' in '* select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
   select->set_initial_value(0.0f);
           ^~~~~~~~~~~~~~~~~
Compiling .pioenvs/test/lib64d/WiFi/WiFiClient.cpp.o
In file included from src/esphome/components/api/api_connection.h:7,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/application.h: In instantiation of 'C* esphome::Application::register_component(C*) [with C = int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)]':
src/main.cpp:168:32:   required from here
src/esphome/core/application.h:148:19: error: static assertion failed: Only Component subclasses can be registered
     static_assert(std::is_base_of<Component, C>::value, "Only Component subclasses can be registered");
                   ^~~
Archiving .pioenvs/test/lib293/libAsyncTCP-esphome.a
Compiling .pioenvs/test/lib64d/WiFi/WiFiGeneric.cpp.o
Compiling .pioenvs/test/lib64d/WiFi/WiFiMulti.cpp.o
*** [.pioenvs/test/src/main.cpp.o] Error 1
========================= [FAILED] Took 87.87 seconds =========================
`

ssieb, 2023-12-26:
Use a different `id:` than `select`.

lordzid, 2023-12-27:
thank you, it helped

> Используйте другой, `id:`чем `select`.

closed 2023-12-27 by lordzid