esphome/issues #5300: time() is ambiguous in esphome 2023.12.5

baudneo, 2023-12-26:
### The problem

I have an emporia vue utility connect device that connects to a zigbee smart energy protocol power meter and communicates usage data to the esp32. I am using a custom .h file and using time().

After upgrading today to 2023.12.5 I can not build and install or update this device, it gives me this error:
```
In file included from src/main.cpp:74:
src/emporia_vue_utility.h: In member function 'virtual void EmporiaVueUtility::loop()':
src/emporia_vue_utility.h:568:19: error: reference to 'time' is ambiguous
             now = time(&now);
                   ^~~~
In file included from src/esphome/components/homeassistant/time/homeassistant_time.h:4,
                 from src/esphome.h:24,
                 from src/main.cpp:3:
src/esphome/components/time/real_time_clock.h:11:11: note: candidates are: 'namespace esphome::time { }'
 namespace time {
           ^~~~
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/time.h:22,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/time.h:414,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/time.h:16,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/pthread.h:18,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:48,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr.h:151,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/ext/atomicity.h:35,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:39,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/string:52,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/stdexcept:39,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/array:39,
                 from src/esphome/components/api/api_noise_context.h:3,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/time.h:59:11: note:                 'time_t time(time_t*)'
 time_t    time (time_t *_timer);
           ^~~~
```
So it seems xtensa `time.h` is clashing with esphomes `time.h` or `real_time_clock.h`. I tried adding `#include "time.h"` but, I get the same error. I am unsure how to resolve this as I dont use C language.

Here is the gist of the custom .h file -> https://gist.github.com/baudneo/4f6ca6713697399b3cd4cd05aa0e96e5#file-gistfile1-txt-L568

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.2

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

custom include of an .h file

### Example YAML snippet

```yaml
esphome:
  name: vue-utility
  friendly_name: vue-utility-connect
  platform: ESP32
  board: esp-wrover-kit
  includes:
      - emporia_vue_utility.h
```

### Additional information

Can I somehow explicitly use whichever time() was being used before? IDK if thats xtensa time.h or esphome time.h or?

ssieb, 2023-12-26:
Github issues aren't for custom or external components.  If you have more questions, come ask on the esphome discord server.
You should be using the configured time component instead of `time()` directly.  But otherwise try using `::time()`.

closed 2023-12-26 by ssieb

baudneo, 2023-12-26:
Its not my code, I modified it a bit for my meter but the base code is written by someone else.  I dont know C so I wasnt sure how to fix.

Using `::time(&now);` fixes it for me though.

Thank you!