esphome/issues #5302: RGBW bulbs state OFF but still are DIM.

ThrashinVictim, 2023-12-27:
### The problem

I just updated to ESPHOME 2023.12.5. And for some reason my Wallfire E27 bulbs state they are OFF but still glow very dim.

This didn't happen until I updated.
I searched and found a post from March 7th that stated the same issue but was unresolved.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
esphome:
  name: smart-bulb-1-right

esp8266:
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:

logger:

ota:

# WebServer
web_server:
  port: 80

my9231:
  data_pin: GPIO13
  clock_pin: GPIO15
  num_channels: 4
  num_chips: 1
  bit_depth: 8

output:
  - platform: my9231
    id: output_red
    channel: 3
  - platform: my9231
    id: output_green
    channel: 2
  - platform: my9231
    id: output_blue
    channel: 1
  - platform: my9231
    id: output_cold_white
    channel: 0

light:
  - platform: rgbw
    name: smart-bulb-1-right
    red: output_red
    green: output_green
    blue: output_blue
    white: output_cold_white
```

ssieb, 2023-12-27:
The only relevant change appears to be https://github.com/esphome/esphome/pull/5765.

ssieb, 2023-12-27:
What version did you have installed on the bulb before this?  That change was made in 2023.11.1

ThrashinVictim, 2023-12-27:
I believe it was the version before this one.

ssieb, 2023-12-27:
Which should have been the version with that change, unless you installed the .0 version.

ssieb, 2023-12-27:
@Mat931 any ideas about this?

Mat931, 2023-12-27:
Try increasing the bit_depth as a workaround. The problem will still happen but the light will be so dim that you can barely see it.

ThrashinVictim, 2023-12-27:
> Try increasing the bit_depth as a workaround. The problem will still happen but the light will be so dim that you can barely see it.

Increased bit_depth to 12 and that fixed the issue. Thanks.

closed 2023-12-27 by ThrashinVictim