esphome/issues #5306: Waveshare 7.5in B V3 inverted color Issue

beyeja, 2023-12-28:
### The problem

Since esphome 2023.12.0 and up the epaper display graphics of are inverted. I downgraded to 2023.11.6 and it works perfectly.
In a rare instance the display was even red with white text but I was not able to reproduce this. This might be unrelated.
I tested model: "7.50in-bV3" as instructed by the documentation. I also tested "7.50in-bV2" which does not seem to cause this issue. 
I suspect it might be related to this change from the pure description: https://github.com/esphome/esphome/pull/5239

### Which version of ESPHome has the issue?

2023.12.0 - 2023.12.5

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

FireBeetle ESP32

### Component causing the issue

waveshare_epaper

### Example YAML snippet

```yaml
esphome:
  name: e-ink-pictureframe
  friendly_name: E-Ink Pictureframe

esp32:
  board: firebeetle32

spi:
  - id: spiBus0
    clk_pin: GPIO18
    mosi_pin: GPIO19

font:
  - file: "gfonts://Roboto@700"
    id: robotoFontMed
    size: 27

display:
  - platform: waveshare_epaper
    id: displayComponent
    model: 7.50in-bV3
    #pins
    busy_pin: 
      number: GPIO25
      inverted: false
    reset_pin: GPIO1
    cs_pin: GPIO23
    dc_pin: GPIO3
    spi_id: spiBus0
      
    update_interval: 30s
    reset_duration: 2ms
    rotation: 270
    lambda: |-
      it.print(100, 100, id(robotoFontMed), "Hello World");

captive_portal:
```

### Additional information

esphome 2023.11.6
[image]

esphome 2023.12.0 + 2023.12.5
[image]

sybux, 2024-01-01:
Exact same problem for me too.

sybux, 2024-01-01:
Exact same problem also for me since last major update

JeroenNL85, 2024-01-02:
Same for me, V3 with custom esp32 board

tvdhorst, 2024-01-07:
Same here, with V3 board.

@lucasprim could you take a look?

Jorggs, 2024-01-10:
I'm also seeing the same behaviour with V3 and e-paper ESP32 driver board.

Pofilo, 2024-01-11:
I created a PR to fix this inversion issue.
Can you also try it ?

Jorggs, 2024-01-12:
Forgive my ignorance, but I don't know how to test the new fix. Am happy to do so with guidance.

Pofilo, 2024-01-12:
You can add something like that:
```
external_components:
  - source:
      type: git
      url: https://github.com/Pofilo/esphome
      ref: fix_waveshare_7P5InBV2and3
    components: [waveshare_epaper]
```

Doc is [here](https://esphome.io/components/external_components.html).

Don't forget to remove this change in your conf once you tested, otherwise it won't work anymore once the PR is merged and the branch deleted.
If you are comfortable with git, you can clone the `esphome` repo and checkout my branch to use it until a new version of `esphome` is release with the fix.

Jorggs, 2024-01-12:
Thank you for the guidance. I can confirm that that is now working for me. Many thanks.

tvdhorst, 2024-01-12:
> You can add something like that:
>
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/Pofilo/esphome
>       ref: fix_waveshare_7P5InBV2and3
>     components: [...]

Thx @Pofilo, I will test this weekend

sybux, 2024-01-12:
Confirm here also that problem is solved.

beyeja, 2024-01-14:
Tested it, issue is fixed. Tested on ESPHome 2023.12.5

Awesome work @Pofilo!

closed 2024-01-18 by jesserockz