esphome/issues #5308: undefined reference to `_ZN14haier_protocol15ProtocolHandler18set_answer_timeoutEx' after ESPHome Update

matthewgrima, 2023-12-29:
### The problem

`Updating /config/esphome/aidan-ac-controller.yaml
------------------------------------------------------------

INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/aidan-ac-controller.yaml...
INFO Detected log level for Haier protocol: DEBUG
INFO Generating C++ source...
INFO Compiling app...
Processing aidan-ac-controller (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- HaierProtocol @ 0.9.24
Linking .pioenvs/aidan-ac-controller/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aidan-ac-controller/src/esphome/components/haier/haier_base.cpp.o:(.text._ZN7esphome5haier16HaierClimateBase18set_answer_timeoutEj+0x0): undefined reference to `_ZN14haier_protocol15ProtocolHandler18set_answer_timeoutEx'
collect2: error: ld returned 1 exit status
*** [.pioenvs/aidan-ac-controller/firmware.elf] Error 1
========================== [FAILED] Took 3.78 seconds ==========================
===== [ERROR] /config/esphome/aidan-ac-controller.yaml =====`

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
esphome:
  name: aidan-ac-controller
  friendly_name: Aidan AC Controller

esp8266:
  board: d1_mini

# Enable logging
logger:
  hardware_uart: UART1

# Enable Home Assistant API
api:
  encryption:
    key: "a128B9b4xWBkhx/E2eC6JvjmzE7fHMzKPXOnN9CipMc="

ota:
  password: "ab6e5adf7e71caf0b952ef3d31d96aad"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Aidan-Ac-Controller"
    password: "Bb85ZzHIzgbk"

captive_portal:

uart:
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3
  id: ac_port  

climate:
  - platform: haier
    id: haier_ac
    protocol: hon
    name: Aidan AC 
    uart_id: ac_port
    wifi_signal: true           # Optional, default true, enables WiFI signal transmission from ESP to AC
    display: true               # Optional, default true, can be used to turn off LED display
    answer_timeout: 1000ms       # Optional, request answer timeout, can be used to increase the timeout
                                # for some ACs that have longer answer delays
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
    - 'OFF'
    - COOL
    - HEAT
    - DRY
    #- FAN_ONLY
    supported_presets:          # Optional, can be used to disable some presets if your AC does not support it
      #- BOOST
      #- COMFORT
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
    - 'OFF'
    #- VERTICAL
    #- HORIZONTAL
    - BOTH
```

ssieb, 2023-12-29:
If you comment out the `answer_timeout:` line, it will compile.
@paveldn this is caused by https://github.com/esphome/esphome/pull/5713.  You need to add this config line to test3 when you fix this.  Apparently, the compiler is really smart and won't include that method if it's not used.

paveldn, 2023-12-29:
@ssieb,
Yes already done. My fix (including the test changes you suggested)  in the pull request https://github.com/esphome/esphome/pull/6015 needs to be approved and released 

@matthewgrima I see you are using answer_timeout: 1000ms it is huge actually. First, try to remove the line at all. It will use default 200ms which should be enough.  If it will cause communication 
 problems with AC after this you will need to roll back to 2023.11.6 and use it until my fix is released.

matthewgrima, 2023-12-29:
Thanks @ssieb & @paveldn, commenting it out worked for me

closed 2023-12-29 by matthewgrima