esphome/issues #5310: Incomplete error message with wrong indentation

micz, 2023-12-29:
### The problem

I'm including a file with the code

```
packages:
  sensori_vari: !include sensor/sensori_vari.yaml
```

If I have an error in the indentation in the included file `sensori_vari.yaml`, with version 2023.11.6 I get this error message:

```
INFO ESPHome 2023.11.6
INFO Reading configuration caldaia.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block collection
  in "sensor/sensori_vari.yaml", line 2, column 3:
      - platform: internal_temperature
      ^
expected <block end>, but found '<block sequence start>'
  in "sensor/sensori_vari.yaml", line 7, column 4:
       - platform: adc
```

With version 2023.12.5 I get this error message:

```
INFO ESPHome 2023.12.5
INFO Reading configuration caldaia.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block collection
  in "<unicode string>", line 2, column 3
did not find expected '-' indicator
  in "<unicode string>", line 7, column 4
```

That makes impossible to understand the source of the error.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

Unfocused, 2024-01-11:
Yea, this makes the error messages largely useless if you're using `!include` 😢 It's not just invalid YAML syntax, but invalid schema errors too.

#5354 seems to show the same issue with single file configs, but at least there the impact is more cosmetic (missing filename of a known file) - although it still outputs one less line of less context.

Seems to be due to using the YAML C loader when it's available, intended to speed things up (added in ESPHome 2023.12.0b1): 
- https://github.com/esphome/esphome/pull/5721 (CC @bdraco)

It was ported over from Home Assistant, but seems to have missed HA's fix for the error messages:
https://github.com/home-assistant/core/blob/e71304580df6e945fc0a3e594b6294a4976bd9da/homeassistant/util/yaml/loader.py#L254-L260

Given the impact for anyone using `!include`, it'd be good to get this either fixed or temporarily reverted in a quick point release.

bdraco, 2024-01-11:
I think what we did in HA for a similar case was fallback to the slow loader on error so you only have to pay the high overhead of the pure python version if you make a mistake in your yaml

closed 2024-01-14 by jesserockz