esphome/issues #5312: Cannot add ESPHome to Home Assistant 2023.12.3 or .4

bkvargyas, 2023-12-30:
### The problem

ESPHome does not show up in the repository or store add-on list.   If you try and add the ESPHome/Addons repository, it says it already exists.

This may be a bug with Home Assistant 2023.12 vs ESPHome...but wanted to make others aware.    Rebooting router, Home Assistant Yellow And refreshing browser, trying other systems did not succeed.   Even tried to downgrade to 2023.12.3 and it didn't work.    I ended up having to CLONE the esphome addon repo and then install it from the new github location, and that worked.

### Which version of ESPHome has the issue?

2023.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
No errors, other than a message that states the repository already exists, yet it did not.   Restarts did not help.
```

ssieb, 2023-12-30:
Many people are doing this, so it's something specific to your system.

bkvargyas, 2023-12-30:
You would think so, however, I've got 4-5 installations running out in the wild, and unless I specifically add ESP Home manually to the repository, it does not show up in the add on repository.   Now I do have 1 install where I have duplicate ESP Home's showing up, probably because on that install, the built in repository actually worked.

All installs are behind Ubiquiti Unifi Systems, so maybe something in those is triggering the firewall system, but I'm using outside DNS, and have no issues with Github for other repositories except ESPHome (built-in version only).   If it wasn't for the fact the latest versions won't let you add the repository in because it considers it a duplicate, I wouldn't have to have cloned the repo to still make it work.