esphome/issues #5314: Touchscreen y-coordinate half of expected with FT63x6 and ILI9xxx

patfelst, 2023-12-30:
### The problem

I'm using an M5Stack Core2 ESP32 module. The touchscreen is working ok, but the `current` Y-Coordinate is half of what it should be, however the `y_raw` is correct. This is mostly evident when using the binary sensor virtual touch buttons, as I have to halve the y_min and y_max values. This is pretty easy to work around, but probably worth fixing. 

Note that the Core2 has 3 red touch buttons below the active display area which add on an extra 40 pixels. So the display resolution is 320x240, but the touch resolution is 320x280. This isn't part of the issue as they touch component is reporting the extra 40 pixels ok.
[image]

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

M5Stack Core2

### Component causing the issue

FT63x6

### Example YAML snippet

```yaml
touchscreen:
  - platform: ft63x6
    i2c_id: bus_a
    interrupt_pin: 39
    id: touch_screen
    on_touch:
      # M5Stack Core2 is 320 x 240 + 40 pixels below display active area for 3x red circular touch buttons
      - lambda: |-
          ESP_LOGE("touch", "Current (x,y) = %d, %d", touch.x, touch.y);        // y-coord is half of expected
          ESP_LOGE("touch", "Raw (x,y) = %d, %d\n", touch.x_raw, touch.y_raw);  // y-coord is correct
          id(touch_x) = touch.x_raw; 
          id(touch_y) = touch.y_raw;

display:
  - platform: ili9xxx
    model: M5STACK
    id: lcd_display
    # dimensions: "320X240"
    update_interval: 500ms
    cs_pin: 5
    dc_pin: 15
    pages:
      - id: page1
        lambda: |-

The log output is as follows:

The work around I've used for the virtual touch buttons, e.g. button A
```yaml
binary_sensor:
  - platform: touchscreen
    name: Button A
    id: btn_a
    internal: True
    x_min: 0
    x_max: 106
    y_min: 120 # should be 240, but is half due to bug in ESPHome touch y-coordinate
    y_max: 140 # should be 280
    filters:
      - delayed_on: ${touch_delay}
    on_press:
      - display.page.show: page1
      - component.update: lcd_display
```

Roughly centre of display
```logs
[21:14:27][E][touch:113]: Current (x,y) = 127, 73
[21:14:27][E][touch:114]: Raw (x,y) = 128, 147
```

Virtual Button A
```logs
[21:14:46][E][touch:113]: Current (x,y) = 46, 139
[21:14:46][E][touch:114]: Raw (x,y) = 47, 279
```
```

guillempages, 2024-01-13:
@nielsnl68 you might want to take a look; this is probably related to the latest touchscreen changes

nielsnl68, 2024-01-14:
Hi, could you try to use my PR#6055 to see of it will fix your issue?

patfelst, 2024-01-14:
yes, give me a day or so, I will report back. Thanks for looking at this.

patfelst, 2024-01-14:
> Hi, could you try to use my PR#6055 to see of it will fix your issue?

Just tested, exactly the same result.

nielsnl68, 2024-01-14:
Okay, did you already play with the `transform:` variables? Maybe you need to set the "swap_xy: true".

patfelst, 2024-01-15:
do you mean the transform option in the `ili9xxx` configuration? OK I just tried and it completely messed up the display, it was rotated 90 deg, but also corrupted (didn't draw properly).

In any case as my example yaml shows above, the .raw values are correct (without transform).

patfelst, 2024-01-15:
by the way, using PR#6055 didn't show a validation error when I made a mistake with the transform option `transform: swap_xy`, it just paused at the end of the display lambda

patfelst, 2024-01-16:
I didn't realise that. ok just tried, but it just rotates the numbers as you'd expect, but y is still half of what it should be. Raw touch coords are still correct though.

nielsnl68, 2024-01-16:
> I just saw on the latest ESPHome page, there's a whole bunch of calibration settings, I'll try these out in the next few days. Not sure if these are new, or I just missed them in the doco before.

for that to work you need to add PR#5997 as external_components

patfelst, 2024-01-16:
I get validation errors if I try PR#5997 `component not found: touchscreen` and `Platform not found: 'binary_sensor.touchscreen'.`.

Config is:
```yaml
external_components:
  - source: github://Pionizer/pionizer-axp192
    components: [axp192]
    refresh: 1days
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: graph
    components: [ graph ]
    refresh: 1min
  # - source: github://pr#6055
  #   components: [ ft63x6 ]
  #   refresh: 0s
  - source: github://pr#5997
    components: [ display, touchscreen, ft63x6 ]
    refresh: 0s
```

nielsnl68, 2024-01-16:
I cant reproduce this issue. Could you show the whole compile log.
Also try to do a "clean build files" to see of this will fix it.
And to be sure you are running this on "DEV" branch?

patfelst, 2024-01-16:
That may be the problem, no I don't believe I'm on DEV branch. How do I do that? Is this where I need to install the dev add-on?

Compile log:
```logs
INFO ESPHome 2023.12.6
INFO Reading configuration /config/esphome/power-monitor-kitchen.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5997/head
ERROR Unable to import component touchscreen:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/ea27d8aa/esphome/components/touchscreen/__init__.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
ERROR Unable to import component touchscreen.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1128, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/ea27d8aa/esphome/components/touchscreen/__init__.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
ERROR Unable to import component touchscreen.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1128, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/ea27d8aa/esphome/components/touchscreen/__init__.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
ERROR Unable to import component touchscreen.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
[... 68 lines cut ...]
  x_max: 213
  y_min: 120
  y_max: 140
  filters: 
    - delayed_on: 10ms
  on_press: 
    - display.page.show: page2
    - component.update: lcd_display
binary_sensor.touchscreen: [source <unicode string>:292]
  
  Platform not found: 'binary_sensor.touchscreen'.
  platform: touchscreen
  name: Button C
  id: btn_c
  internal: True
  x_min: 214
  x_max: 319
  y_min: 120
  y_max: 140
  filters: 
    - delayed_on: 10ms
  on_press: 
    - display.page.show: page3
    - component.update: lcd_display
```

nielsnl68, 2024-01-16:
> That may be the problem, no I don't believe I'm on DEV branch. How do I do that? Is this where I need to install the dev ESPHome add-on?

Yes, the current and dev can be used together. But be aware after install you most likely need to reboot HA.

nielsnl68, 2024-01-16:
can you give me your YAML file to test out? so i can compare with my version.

patfelst, 2024-01-16:
ok thanks, sorry for all the messing around. I've got DEV working with PR#5997 now.

Playing with the y-axis calibration fixes the issue except that the `current` y-value maxes out at the display's max y-resolution (i.e. won't show the virtual touch buttons which are from y=240 to y=280) which makes sense. 

```yaml
touchscreen:
  - platform: ft63x6
    i2c_id: bus_a
    interrupt_pin: 39
    id: touch_screen
    calibration:
      y_min: 0
      y_max: 240
    on_touch:
      # M5Stack Core2 is 320 x 240 + 40 pixels below display active area for 3x red circular touch buttons
      - lambda: |-
          ESP_LOGI("touch", "Current (x,y) = %d, %d", touch.x, touch.y);        // y-coord is half of expected
          ESP_LOGI("touch", "Raw (x,y) = %d, %d\n", touch.x_raw, touch.y_raw);  // y-coord is correct
```
So in other words, now the `current` y-values exactly match the `raw` (with 1-pixel offset), except when raw > 239, current remains at 239.

[image]

patfelst, 2024-01-16:
do you still want my whole yaml file? what's the best way to share it?

nielsnl68, 2024-01-16:
when it works then it is fine :D

patfelst, 2024-01-16:
ok so I can solve the touch y-axis issue completely if I set the display resolution to 320x280
```yaml
display:
  - platform: ili9xxx
    model: M5STACK
    id: lcd_display
    dimensions: "320X280"
    update_interval: 500ms
    cs_pin: 5
    dc_pin: 15
    pages:
```
and touchscreen calibration:
```yaml
touchscreen:
  - platform: ft63x6
    i2c_id: bus_a
    interrupt_pin: 39
    id: touch_screen
    calibration:
      y_min: 0
      y_max: 280
    on_touch:
      # M5Stack Core2 is 320 x 240 + 40 pixels below display active area for 3x red circular touch buttons
      - lambda: |-
 ```
but it has the side effect of a flashing display every other frame/update of black 40 pixels at the top of the display. So this is not really a solution. In other words, you get one complete dispaly that looks ok, and immediately after, the top 40 pixels gets filled with black, so the top 40 pixels are blinking at the display update interval :(

Any thoughts @nielsnl68 ?

zendes, 2025-02-21:
I also just ran into this problem. I took a look at the code, and the calibration segment force maps the coordinates to the display dimensions. I created a PR that'll allow you to optionally specify the touchscreen dimensions independently of the display ones: https://github.com/esphome/esphome/pull/8296

Feel free to give that a test and see if it works for you

patfelst, 2025-02-22:
@zendes thanks, it's been over a year and I don't remember all the details now, can you tell me exactly what you've fixed?

patfelst, 2025-02-22:
ah sorry, I've read your PR now, I will test it out soon!

closed 2025-03-07 by clydebarrow