esphome/issues #5316: Modbus components slows down over time (and finally makes esphome go away)

henrykuijpers, 2023-12-30:
### The problem

I'm running a Modbus readout on an ESP32 (wroom32 dev board), in combination with a MAX485 board (the one that has DI, RE, DE, RO).

Often see messages in the logs about operations taking a lot of time.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

ESP32 

### Component causing the issue

wroom32 (nodemcu-32s)

### Example YAML snippet

```yaml
substitutions:
  name: modbus-esp
  friendly_name: Modbus

esphome:
  name: ${name}
  comment: ${friendly_name}
  platform: esp32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain

  ap:
    ssid: $friendly_name Wifi
    password: !secret wifi_password
    
captive_portal:

logger:
  level: VERY_VERBOSE
  baud_rate: 0

api:

ota:

switch:
  - platform: restart
    name: ${friendly_name} Restart

web_server:
  port: 80

#serial port
[... 363 lines cut ...]
    state_class: "measurement"
    unit_of_measurement: ""
    accuracy_decimals: 3
    
  - platform: modbus_controller
    modbus_controller_id: SDM120M
    name: ${friendly_name} Phase 3 Power Factor
    register_type: read
    address: 0x0022
    value_type: FP32
    device_class: "power_factor"
    state_class: "measurement"
    unit_of_measurement: ""
    accuracy_decimals: 3
```

### Anything in the logs that might be useful for us?

```txt
16:02:11 | [W] | [component:214] | Component modbus took a long time for an operation (0.05 s).

(The message is apparently a feature, and it's fine that it takes a bit of time to process everything.)

After a while it starts increasing to 0.10 s, 0.20 s, ... And then suddenly it goes offline. ESP32 is still turned on (LEDs still burning), but the web interface is gone and there is no way to interact with it.
```

henrykuijpers, 2023-12-30:
I closed the web interface and everything seems to be fine now. Tomorrow I can test again with the web interface open. 

Somehow, that seems to be related to the problem. It could be that it is too heavy to create all the json packages? (I'm not very much into the technical details of esphome to know for sure it has anything to do with it or not.)

ssieb, 2023-12-31:
Why did you disable the serial logs?  You could watch those to see what's happening when it's not responding.

henrykuijpers, 2024-01-02:
> Why did you disable the serial logs? You could watch those to see what's happening when it's not responding.

Where do you see the serial logs disabled? I had them disabled at first, to not interfere too much with other things (it was written in an article somewhere), but I believe serial logs are not disabled anymore?

ssieb, 2024-01-02:
You modified the pasted config after I saw it.
So you can watch the serial logs to see what's happening.

henrykuijpers, 2024-01-02:
Ah! I didn't realise that, then you were pretty quick! 

Yes, I know, I can indeed monitor the serial logs. They should be the same as the OTA logs, right? Maybe the logs continue through the serial port, while the networking goes offline? That would be good test. I'll try to test this soon.

The only thing I observed was that it just stops. From that moment on, the esp is offline. 

A few pieces of logging that I could capture (through web): 

```

Time
level
Tag
Message
06:13:45
[D]
[sensor:094]
'Modbus Neutral Current': Sending state 0.00000 A with 3 decimals of accuracy
06:13:45
[V]
[json:038]
Attempting to allocate 512 bytes for JSON serialization
06:13:45
[V]
[json:058]
Size after shrink 88 bytes
06:13:45
[VV
][api.service:140]
send_sensor_state_response: SensorStateResponse {
  key: 2993591581
  state: 0
  missing_state: NO
}
06:13:45
[W]
[component:214]
Component modbus_controller took a long time for an operation (0.07 s).
06:13:45
[W]
[component:214]
Component modbus_controller took a long time for an operation (0.07 s).
06:13:45
[W]
[component:215]
Components should block for at most 20-30ms.
06:13:45
[V]
[modbus_controller:047]
Sending next modbus command to device 1 register 0x156 count 4
06:13:45
[VV
][uart.arduino_esp32:180]
   Flushing...
06:13:46
[V]
[modbus:199]
Modbus write: 01.04.01.56.00.04.10.25 (8)
06:13:46
[V]
[modbus_controller:509]
Command sent 4 0x156 4
06:13:46
[V]
[modbus:042]
Modbus received Byte  1 (0X1)
06:13:46
[V]
[modbus:042]
Modbus received Byte  4 (0X4)
06:13:46
[V]
[modbus:042]
Modbus received Byte  8 (0X8)
06:13:46
[V]
[modbus:042]
Modbus received Byte  67 (0X43)
06:13:46
[V]
[modbus:042]
Modbus received Byte  100 (0X64)
06:13:46
[V]
[modbus:042]
Modbus received Byte  215 (0Xd7)
06:13:46
[V]
[modbus:042]
Modbus received Byte  76 (0X4c)
06:13:46
[V]
[modbus:042]
Modbus received Byte  67 (0X43)
06:13:46
[V]
[modbus:042]
Modbus received Byte  28 (0X1c)
06:13:46
[V]
[modbus:042]
Modbus received Byte  24 (0X18)
06:13:46
[V]
[modbus:042]
Modbus received Byte  147 (0X93)
06:13:46
[V]
[modbus:042]
Modbus received Byte  216 (0Xd8)
06:13:46
[V]
[modbus:042]
Modbus received Byte  40 (0X28)
06:13:46
[V]
[modbus_controller:078]
Modbus response queued
06:13:46
[W]
[component:214]
Component modbus took a long time for an operation (0.07 s).
06:13:46
[W]
[component:215]
Components should block for at most 20-30ms.
06:13:46
[V]
[modbus_controller:086]
Process modbus response for address 0x156 size: 8
06:13:46
[V]
[modbus_controller:121]
data for register address : 0x156 :
06:13:46
[D]
[modbus_controller.sensor:025]
Sensor new state: 228.84
06:13:46
[V]
[sensor:043]
'Modbus Total Active Energy': Received new state 228.841003
06:13:46
[V]
[sensor:043]
'Modbus Total Active Energy': Received new state 228.841003
06:13:46
[D]
[sensor:094]
'Modbus Total Active Energy': Sending state 228.84100 kWh with 3 decimals of accuracy
06:13:46
[V]
[json:038]
Attempting to allocate 512 bytes for JSON serialization
06:13:46
[V]
[json:058]
Size after shrink 96 bytes
06:13:46
[VV
][api.service:140]
send_sensor_state_response: SensorStateResponse {
  key: 759117560
  state: 228.841
  missing_state: NO
}
06:13:46
[D]
[modbus_controller.sensor:025]
Sensor new state: 156.10
06:13:46
[D]
[modbus_controller.sensor:025]
Sensor new state: 156.10
06:13:46
[V]
[sensor:043]
'Modbus Total Reactive Energy': Received new state 156.095993
06:13:46
[V]
[sensor:043]
'Modbus Total Reactive Energy': Received new state 156.095993
06:13:46
[D]
[sensor:094]
'Modbus Total Reactive Energy': Sending state 156.09599 kVArh with 3 decimals of accuracy
06:13:46
[V]
[json:038]
Attempting to allocate 512 bytes for JSON serialization
06:13:46
[V]
[json:058]
Size after shrink 100 bytes
06:13:46
[V]
[json:058]
Size after shrink 100 bytes
06:13:46
[VV
][api.service:140]
send_sensor_state_response: SensorStateResponse {
  key: 3439512477
  state: 156.096
  missing_state: NO
}
06:13:46
[W]
[component:214]
Component modbus_controller took a long time for an operation (0.13 s).
06:13:46
[W]
[component:215]
Components should block for at most 20-30ms.
06:13:48
[VV
][scheduler:226]
Running interval '' with interval=10000 last_execution=29024 (now=39032)
06:13:58
[VV
][scheduler:226]
Running interval '' with interval=60000 last_execution=4294955911 (now=48616)
06:13:58
[VV
][scheduler:226]
Running interval '' with interval=10000 last_execution=39024 (now=49025)

```

ssieb, 2024-01-02:
The serial logs work even if wifi is not connected.  What I'm looking for is what is happening when it stops working.

henrykuijpers, 2024-05-02:
@ssieb Do you have a suggestion of what kind of configuration to use, to be able to analyze this further?

ssieb, 2024-05-02:
What I already suggested.  You need to watch the serial logs at the point (or after) it goes offline.