esphome/issues #5319: Requesting data from DHT failed!

alfwro13, 2023-12-31:
### The problem

After upgrading to the latest version of ESPHOME my temperature sensor started reporting this error:

```
[08:35:05][W][dht:169]: Requesting data from DHT failed!
[08:35:05][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option.
```

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

lolin_s2_mini

### Component causing the issue

DHT11

### Example YAML snippet

```yaml
- platform: dht
    pin: 5
    temperature:
      name: ${friendly_name} Temperature
      filters:
         - offset: -1.0
    humidity:
      name: ${friendly_name} Humidity
    update_interval: 15s
```

### Anything in the logs that might be useful for us?

```txt
error message with model: auto_detect

[08:35:05][W][dht:169]: Requesting data from DHT failed!
[08:35:05][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option.

error message with model: dht11

[08:43:14][W][dht:169]: Requesting data from DHT failed!
[08:43:14][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

### Additional information

it was working fine prior to the upgrade (upgraded from esphome 2023.11.x version to the 2023.12.5).

kjudom, 2024-02-03:
try a hard reboot. My DHT22 has similar issue after it wakes up from deep sleep.

alfwro13, 2024-02-08:
yep, that did the trick. Thanks

closed 2024-02-08 by alfwro13