esphome/issues #5320: Nextion display won´t sleep

Hellfish78, 2023-12-31:
### The problem

Auto sleep and wake on touch is enabled but the nextion display (nspanel) won´t go into sleep mode.
I´ve set thsp=30 thup=1 as preinitialze event in nextion editor.
Within the nextion debugger it works as intended and the display sleeps after 30 seconds.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

sonoff nspanel

### Example YAML snippet

```yaml
esphome:
  name: nspanel
  comment: $device_name
esp32:
  board: esp32dev

substitutions:
 device_name: nspanel
 wifi_ssid: !secret wifi_ssid
 wifi_pass: !secret wifi_password
 tft_url: !secret tft_upload_url

#web_server:
#  port: 80
#  css_url: ""
#  js_url: https://esphome.io/_static/v2/www.js
#  version: 2

# Configure the screen itself
display:
- platform: nextion
  id: disp1
  uart_id: tf_uart
  tft_url: $tft_url
  touch_sleep_timeout: 30
  auto_wake_on_touch: True

# Uart for the Nextion display
uart:
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200 # has to be set at program.s on the nextion editor 
  id: tf_uart

# i2c bus configuration
i2c:
  sda: 19
  scl: 22
  scan: true
  id: bus_a

[... 264 lines cut ...]
[09:08:58][C][nextion:134]:   Exit reparse:     NO
[09:08:58][C][nextion:137]:   Touch Timeout:    30
[09:08:58][C][captive_portal:088]: Captive Portal:
[09:08:58][C][mdns:115]: mDNS:
[09:08:58][C][mdns:116]:   Hostname: nspanel
[09:08:58][C][ota:097]: Over-The-Air Updates:
[09:08:58][C][ota:098]:   Address: nspanel.local:3232
[09:08:58][C][ota:101]:   Using Password.
[09:08:58][C][api:139]: API Server:
[09:08:58][C][api:140]:   Address: nspanel.local:6053
[09:08:58][C][api:144]:   Using noise encryption: NO
[09:09:53][D][binary_sensor:036]: 'nspanel Left Button': Sending state ON
[09:09:53][D][binary_sensor:036]: 'nspanel Left Button': Sending state OFF
[09:11:39][D][ads1115:186]: 'ADS1115 Channel A0-GND': Got Voltage=1.095875V
[09:11:39][D][sensor:094]: 'ADS1115 Channel A0-GND': Sending state 49.81250  with 3 decimals of accuracy
[09:12:02][D][sensor:094]: 'ntc_source': Sending state 1.70800 V with 2 decimals of accuracy
[09:12:02][D][resistance:039]: 'resistance_sensor' - Resistance 12016.1Ω
[09:12:02][D][sensor:094]: 'resistance_sensor': Sending state 12016.08008 Ω with 1 decimals of accuracy
[09:12:02][D][ntc:026]: 'nspanel Temperature' - Temperature: 20.9°C
[09:12:02][D][sensor:094]: 'nspanel Temperature': Sending state 20.92329 °C with 1 decimals of accuracy
```

### Additional information

[image]

edwardtfn, 2024-05-07:
Looks like this issue is on the Nextion side. I've tried with a model NX4832F035_011C (the one used in NSPanel) and it failed, however when trying with a NX3224F024_011R it works just fine. Both are from Discovery series, and both running with fw v72.

I've used debug mode for uart, so I'm sure the thsp command is sent, but looks like the Nextion display just ignores that.

robopeter, 2025-04-07:
Strangely, my NX3224K028B_011 was working fine until a couple of updates of ESP Home ago, and now it never sleeps. :-(

edwardtfn, 2025-04-07:
I don't remember any change in the sleep command in the latest changes to this component, but I will double check this.

Do you know which firmware version you have on your Nextion display?

robopeter, 2025-04-07:
I do not, nor do I know how to tell.

edwardtfn, 2025-04-09:
When ESPHome establishes connection to your display it will report the firmware version on the logs.
Something like this:
```log
[09:08:58][C][nextion:128]: Nextion:
[09:08:58][C][nextion:129]:   Device Model:     NX4832F035_011C
[09:08:58][C][nextion:130]:   Firmware Version: 72
[09:08:58][C][nextion:131]:   Serial Number:    B8B034015C9DFF53
[09:08:58][C][nextion:132]:   Flash Size:       16777216
[09:08:58][C][nextion:133]:   Wake On Touch:    YES
[09:08:58][C][nextion:134]:   Exit reparse:     NO
[09:08:58][C][nextion:137]:   Touch Timeout:    30
```

robopeter, 2025-04-16:
Thanks @edwardtfn ! I have the following in my logs upon initial boot:

```
[21:07:37][I][nextion:097]: Nextion is connected
[21:07:37][C][nextion:138]: Nextion:
[21:07:37][C][nextion:142]:   Device Model:     NX3224K028_011R
[21:07:37][C][nextion:143]:   Firmware Version: 163
[21:07:37][C][nextion:144]:   Serial Number:    E469CCC8832D2021
[21:07:37][C][nextion:145]:   Flash Size:       16777216
[21:07:37][C][nextion:147]:   Wake On Touch:    YES
[21:07:37][C][nextion:148]:   Exit reparse:     NO
[21:07:37][C][nextion:151]:   Touch Timeout:    120
[21:07:37][C][nextion:155]:   Wake Up Page:     0
[21:07:37][W][component:237]: Component display took a long time for an operation (179 ms).
[21:07:37][W][component:238]: Components should block for at most 30 ms.
[21:07:37][C][nextion:138]: Nextion:
[21:07:37][C][nextion:142]:   Device Model:     NX3224K028_011R
[21:07:37][C][nextion:143]:   Firmware Version: 163
[21:07:37][C][nextion:144]:   Serial Number:    E469CCC8832D2021
[21:07:37][C][nextion:145]:   Flash Size:       16777216
[21:07:37][C][nextion:147]:   Wake On Touch:    YES
[21:07:37][C][nextion:148]:   Exit reparse:     NO
[21:07:37][C][nextion:151]:   Touch Timeout:    120
[21:07:37][C][nextion:155]:   Wake Up Page:     0
```
(yes, it seems to have been duplicated in the actual logs for some reason).