esphome/issues #5326: M5Stack Atom Echo Voice Assistant: Wake work only works once after reboot, and no audio from speakers

michelcve, 2024-01-02:
### The problem

I bought two new M5Stack Atom Echo devices, and flashed these per the instructions here: https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/

Made no changes to the yaml or anything, just flashed it from there.

atom-echo-voice-assistant
by m5stack
Firmware: 2023.12.3 (Dec 25 2023, 13:55:19)
Hardware: 1.0

Next, I'm using Homeassistant 2023.12.4, set up voice control using the following docker images:
- wyoming-piper:1.4.0
- wyoming-whisper:1.0.0
- wyoming-wakewords:1.8.0

I used the wakeword "Hey Jarvis".

The device boots properly, and after a while, it accepts my wakeword (LED starts white, after wake word starts flashing blue, and after a short while starts flashing faster, after which is returns to begin white).

So far so good, but after that two problems occur:
- I hear no audio response, whether I use a proper sentence, or just say something that is completely random
- The device won't respond to the wake word again, unless I reboot the device. Then it works again for 1 time only.

I tried looking at the logs, but either I don't understand the console, or it just doesn't keep logging anything after boot? If someone can give me some pointers, I can look into it a bit further.

### Which version of ESPHome has the issue?

Not sure, I'm using docker homeassistant code with esphome integration, but can't seem to find the version number

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

M5Stack Atom Echo

GaWr26, 2024-01-05:
Exactly the same for me. Any ideas?

illigtr, 2024-01-05:
I haven't been able to test as a recent update to ESPHome broke all my ILI9488 displays... been spending several days trying to figure out why.... Once I have that sorted out, I'll be back to my private voice integration tests, which so far, I've only just begun. If I do hear/see anything related, I'll post.

michelcve, 2024-01-12:
I rechecked with:

Firmware: 2023.12.5 (Jan 10 2024, 02:19:27)
Hardware: 1.0

Unfortunately the problem still persists. Anyone have a clue so far?

ilndboy, 2024-01-13:
I am having the same issue as well

arunes11, 2024-01-14:
same thing for me with two Atom Echos, same problem

skyw33, 2024-01-17:
Same issue

srwalter, 2024-01-22:
I have a similar-but-slightly different issue (running HA 2023.12.3).  My M5Stack Atom Echo's (I have 2) both work flawlessly at first start-up.  They recognize the wake word and I get audio from the speakers, and it works multiple times in a row.  However, after some amount of time, they stop responding to the wake word.

To fix them, I can either toggle "Use Wake Word" twice for the device, or I can press the switch for 250ms.  From looking at the script, pressing the button should just stop the voice assistant and then go back to wait-for-wake-word (start_continuous), so that implies that the device was not already running in continuous mode for some reason.  The LED remained on, so I think I can rule out on_client_disconnected, at least.  I also have an Onju Voice connected to the same HA server, and it does not have this issue.

michelcve, 2024-01-25:
I rechecked with:

Firmware: 2023.12.8 (Jan 19 2024, 23:03:36)
Hardware: 1.0

Unfortunately the problem still persists. Would there be any way I could help?

joshfedo, 2024-02-02:
Same issue over here

Ciaccona, 2024-02-10:
> I have a similar-but-slightly different issue (running HA 2023.12.3). My M5Stack Atom Echo's (I have 2) both work flawlessly at first start-up. They recognize the wake word and I get audio from the [...]

Exactly the same issue here.

michelcve, 2024-02-16:
I rechecked with:

Firmware: 2023.12.9
Hardware: 1.0

Unfortunately the problem still persists.

I did find this troubleshooting guide, that may be of some help: https://www.home-assistant.io/voice_control/troubleshooting/

In my case, it hangs on tts.piper. The request is apparently sent there (??), but it keeps spinning. Not sure what is causing this, but at least I have a bit more information. This also explains why I don't hear any response.

michelcve, 2024-02-16:
@srwalter and @Ciaccona just curious as you do seem to get audio: how are you running homeassistant? With HASS or in a docker container?

michelcve, 2024-02-16:
Okay! Progress!!

This was a needle in a haystack, but basically, I had to add the following to my configuration.yaml (or if you don't use that, set it up correctly in settings --> system --> network).

```
homeassistant:
  external_url: "https://www.example.com"
  internal_url: "http://192.168.x.x:8123"
```

With that in place, piper started working, and the atom talked back to me!!

I have tried it 4 times in a row without restarting it, and so far I've been given a response each and every time.

In Settings --> System --> Network, the following text gave me the hint:

"Configure what website addresses Home Assistant should share with other devices when they need to fetch data from Home Assistant (eg. to play text-to-speech or other hosted media)."

So apparently there are some components that need to contact homeassistent by themselves, and piper is one of them.

Hope this helps others as well...

closed 2024-02-16 by michelcve

skyw33, 2024-02-18:
For what it's worth, this did not help my situation.

michelcve, 2024-02-18:
@skyw33 sad to hear, did you try troubleshooting like in https://www.home-assistant.io/voice_control/troubleshooting/ ?

skyw33, 2024-02-18:
I'm trying the troubleshooting steps now and will report the results shortly.  I just rebooted and in my case, the voice assistant works repeatedly after reboot.  It's only after some period of idle time that I come back and it won't work until I reboot again.  Right now, I'm just waiting until it stops working and I'll report back the error.

Ciaccona, 2024-02-18:
> @srwalter and @Ciaccona just curious as you do seem to get audio: how are you running homeassistant? With hass.io or in a docker container?

HAS OS for me. I'll try the troubleshooting steps now, cheers.

skyw33, 2024-02-20:
@michelcve Contrary to what I said earlier, this appears to have fixed my issue.  Initially, it stopped responding after a few hours, but it has been up for a couple days now -- longer than ever before.  I'll let you know if I run into any more trouble.  Thank you for finding a solution for this!

michelcve, 2024-02-21:
That's good to hear!

skyw33, 2024-02-24:
Well it happened again.  The device is on but not responding.  When I go to debug there is nothing there.
[image]

skyw33, 2024-02-24:
ESPHome log doesn't show anything is wrong either.
[image]

michelcve, 2024-02-25:
Maybe you can find more information in Settings --> Voice Assistant --> Your voice configuration --> .. menu, and then debug

skyw33, 2024-02-29:
Anybody coming here with a similar issue can also follow this open issue:  https://github.com/esphome/issues/issues/5038

michelcve, 2024-03-01:
My M5 Stack stopped responding too. I checked the logs of the device, this is what I found:

```
ice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
```

So in my case, apparently when playing back the response, some buffers eventually got overloaded (or not freed up properly).

I'll also add a comment to the issue @skyw33 mentioned. As this one is closed, and is another problem.

andiukas, 2024-05-23:
the problem still persists.

zox-gh, 2024-05-24:
I confirm I still have the problem. After first wakeword, it stops responding. If I switch "use wakeword" off, and use only button, it mostly works.
Have tried the trick with network settings but no luck.

zox-gh, 2024-05-24:
UPDATE:
New version of ESPHome, 2024.5.3 released half an hour ago, seems to have the issue resolved!
2024.5.3
[voice_assistant] Don't allocate buffers until starting the microphone for the first time [esphome#6800](https://github.com/esphome/esphome/pull/6800) by [@jesserockz](https://github.com/jesserockz)

Now wakeword works well. At least in last few minutes, even with repeated wakewords after each other.

Izaic, 2024-05-25:
All of my M5 stacks are also doing the same. They seem to intermittently just stop responding to wake words. The logs show UDP and CRC errors. Thing is I have iirc 10 total. I tried two that were old ones I had already used a few months back and which worked fine with this exact HAOS setup I've had, and I also pulled out a brand new one and flashed it as well. They're all doing the same thing, so the new voice firmware or something else must have an issue.

stevelavergne1, 2024-05-25:
Following. Same issue on multiple M5's.  

Suspect line in log "Packet write failed OUT_OF_MEMORY errno=11"

stevelavergne1, 2024-06-04:
> Following. Same issue on multiple M5's.
> 
> Suspect line in log "Packet write failed OUT_OF_MEMORY errno=11"

Adding the below to devices config.yaml and installing from the ESPHome dash corrected it.

external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s

  - source: github://pr#6718
    components:
      - voice_assistant
      - i2s_audio
      - speaker

esp_adf:

Source: https://github.com/esphome/issues/issues/5849

ThijmenDam, 2024-06-16:
@skyw33 for the life of me I cannot figure out how to access the ESP logs like you show in your response. Would you be so kind to elaborate?

skyw33, 2024-06-17:
@ThijmenDam That screenshot is from ESPHome dashboard.  I run ESPHome as a separate docker container, so if you run it as an addon to home assistant, it may be different.  Anyway, from the dashboard, there is a "Logs" option.

Cadseubi, 2024-06-22:
I just installed the device and I have the same problem, it recognizes the wakeword only once.

smithimage, 2024-06-26:
I have same exact issue. Why is this issue closed ?

melqudsi, 2024-07-02:
I did a thing that helped improve the situation, issue is not completely gone but it made it much more stable and usable.  
I "outsourced" the playback/response part to another Echo Atom (media player) by using information from the two URL's below:
https://community.home-assistant.io/t/link-m5-atom-echo-to-external-speaker/666773/13
https://www.youtube.com/watch?v=o3yZWD_sFIE

Seems like an Echo performance issue maybe?