esphome/issues #5327: The hlw8012 doesn't work due to an issue in the pulse_counter component

wjtje, 2024-01-02:
### The problem

When I try to use pulse_counter sensor (used in the hlw8012 sensor that I actually want to use) the counter stays on 0, but when I manualy create a pulse_meter sensor it works.

After diggin in the code some I figured out that the issue is inside the [gpio_intr](https://github.com/esphome/esphome/blob/dev/esphome/components/pulse_counter/pulse_counter_sensor.cpp#L20) function. When the digital pin in read, it will (almost) always return 0.

Maybe it's possible to change the hlw8012 component so it uses the pulse_meter sensor internally?

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

BK72XX

### Board

cb2s

### Component causing the issue

hlw8012

### Example YAML snippet

```yaml
sensor:
# Both pins have pulses, but only the pulse_meter sees any
  - platform: pulse_counter
    pin: 24
    name: "Pulse Counter"
  - platform: pulse_meter
    pin: 16
    name: "Pulse Meter"
```

Destroyer061090, 2024-01-02:
I have same issues here… I’m using yf-b6 sensor and I get always zero value…
`sensor:
- platform: pulse_counter
pin: GPI033
name: "Consumo acqua attuale"
id: consumo_acqua_attuale
update_interval : 1s
unit_of_measurement: "1/min"
icon: "mdi:water"
accuracy_decimals: 2
filters:
- lambda: |-
id(pulses). publish_state(x);
x= (x / (6.6 * 60)); `

Brokly, 2024-04-06:
Я имею аналогичную проблему  с компонентом  hlw8012 и процессором BK72XX.
```yaml
 - platform: hlw8012
    sel_pin: P9
    cf_pin: P20
    cf1_pin: P8
    model: BL0937
```

Brokly, 2024-04-07:
I've adjusted the component a bit. Since pulse_counter often breaks down, I thought it would be better to abandon it altogether. In the process, I discovered that the SEL output was not working correctly (perhaps only for BL0937). If anyone is interested, the modified component is here:  https://github.com/Brokly/hlw8012  .