esphome/issues #5328: ESPHome dashboard does not start after update to ESPHome 2023.12.5

Buggykiter, 2024-01-02:
### The problem

ESPHOME dashboard does not start. Protokoll:
[11:57:01] INFO: Service ESPHome dashboard exited with code 1 (by signal 0)

### Which version of ESPHome has the issue?

ESPHome 2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 System: Home Assistant OS 11.2  (amd64 / qemux86-64)  Home Assistant Core: 2023.12.4  Home Assistant Supervisor: 2023.12.0

### What platform are you using?

ESP8266

### Board

All ESP8266 and ESP32

### Component causing the issue

ESPHome

### Anything in the logs that might be useful for us?

```txt
Protokoll:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2023.12.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / qemux86-64)
 Home Assistant Core: 2023.12.4
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[11:56:59] INFO: Configuring timezone (Europe/Berlin)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service esphome: starting
s6-rc: info: service esphome successfully started
s6-rc: info: service init-nginx successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service discovery: starting
[11:57:01] INFO: Waiting for ESPHome dashboard to come up...
[11:57:01] INFO: Starting ESPHome dashboard...
./run: line 57: /usr/local/bin/esphome: Success
[11:57:01] INFO: Service ESPHome dashboard exited with code 1 (by signal 0)
[11:57:03] INFO: Starting ESPHome dashboard...
./run: line 57: /usr/local/bin/esphome: Success
[11:57:03] INFO: Service ESPHome dashboard exited with code 1 (by signal 0)
[11:57:05] INFO: Starting ESPHome dashboard...
./run: line 57: /usr/local/bin/esphome: Success
[11:57:05] INFO: Service ESPHome dashboard exited with code 1 (by signal 0)
```

### Additional information

Restart of ESPHome does not help, allways get the message: 
Das Add-on scheint noch nicht bereit zu sein, möglicherweise startet es noch. Willst du es nochmal versuchen?
NEIN   ERNEUT VERSUCHEN

nielsnl68, 2024-01-02:
I have the same issue.
the work-a-round is to reboot HA.
After that Esphome is back online.

Buggykiter, 2024-01-02:
Reboot does not help in my case. The first time I go to the ESPHome UI after a reboot of the VM and restart of HA, all nodes are visible but EDIT or LOGs does not work. Second attempt gives: Add-on not ready

vactomas, 2024-01-03:
Similar issue here as well.

I am running HAOS VM on x86. Tried reinstalling, restoring VM etc.

When I try to open dashboard, I only see a black screen. There is nothing in the logs though.

```
s6-rc: info: service esphome: starting
s6-rc: info: service esphome successfully started
s6-rc: info: service init-nginx successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service discovery: starting
[09:53:36] INFO: Waiting for ESPHome dashboard to come up...
[09:53:36] INFO: Starting ESPHome dashboard...
2024-01-03 09:53:37,464 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[09:53:37] INFO: Starting NGINX...
2024-01-03 09:53:37,762 INFO 200 GET /devices (0.0.0.0) 3.08ms
[09:53:37] INFO: Successfully send discovery information to Home Assistant.
s6-rc: info: service discovery successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2024-01-03 09:58:37,831 INFO 200 GET /devices (0.0.0.0) 2.85ms
2024-01-03 10:03:37,831 INFO 200 GET /devices (0.0.0.0) 2.99ms
```

dickmann, 2024-01-03:
same here...
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2023.12.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2023.12.4
 Home Assistant Supervisor: 2023.12.0

Fresh install, but GUI shows only black screen...
Kind regards, Michael

illigtr, 2024-01-05:
I have a related issue with OS Core 2024.1, but it only happens on ESP Home (dev). I have both ESP Home and dev installed in HA. production version works, but every time I click "rebuild" in the addon for ESP Home (dev) it breaks the GUI link. An OS reboot (RPi/4) fixes  the issue temporarily.

vactomas, 2024-01-05:
> Similar issue here as well.
> 
> I am running HAOS VM on x86. Tried reinstalling, restoring VM etc.
> 
> When I try to open dashboard, I only see a black screen. There is nothing in the logs though.
> 
> ```
> s6-rc: info: service esphome: starting
> s6-rc: info: service esphome successfully started
> s6-rc: info: service init-nginx successfully started
> s6-rc: info: service nginx: starting
> s6-rc: info: service nginx successfully started
> s6-rc: info: service discovery: starting
> [09:53:36] INFO: Waiting for ESPHome dashboard to come up...
> [09:53:36] INFO: Starting ESPHome dashboard...
> 2024-01-03 09:53:37,464 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
> [09:53:37] INFO: Starting NGINX...
> 2024-01-03 09:53:37,762 INFO 200 GET /devices (0.0.0.0) 3.08ms
> [09:53:37] INFO: Successfully send discovery information to Home Assistant.
> s6-rc: info: service discovery successfully started
> s6-rc: info: service legacy-services: starting
> s6-rc: info: service legacy-services successfully started
> 2024-01-03 09:58:37,831 INFO 200 GET /devices (0.0.0.0) 2.85ms
> 2024-01-03 10:03:37,831 INFO 200 GET /devices (0.0.0.0) 2.99ms
> ```
> 
> EDIT: Further testing leads me to believe that my problems are related to HA and not ESPHome itself. I installed Adguard to test and I also couldn't get to WebUI

Solved by updating to new Core version.

Buggykiter, 2024-01-06:
I have restored ESPHome backups one after the other. The latest working version I found is ESPHome 2023.11.4. All later versions are faulty. ESPHome 2023.11.4 works fine with the versions of core 2024.1.1, Supervisor 2023.12.0, Operating System 11.3 and Frontend 2024.01.04.0 in an VM on x86.
I think, the bug in ESPHome was introduced with version 2023.12.0.

Solved by restoring ESPHome 2023.11.4

illigtr, 2024-01-06:
There were numerous, poor quality, updates to ESPHome in early December, nearly one a day. Lots of errors with compiling.. It broke nearly half of my ESP32 devices, they stopped responding, even with power cycle. I could no longer use OTA, this forced me to open a number of my project boxes and extract the esp32 dev board and re-load old BIN files via the UART. A real pain and lesson learned to ensure backup bin files for everything before applying an ESPHome upgrade!!! I wish there was better QA for ESPHome updates. There needs to be a wider community of testers that can signoff on stuff that get moved to production branch.

oyvindwe, 2024-01-10:
I experience this starting with version 2023.11.0. Version 2023.10.6 works fine.

Command: `docker run --rm --name esphome -v /home/omw/esphome/:/config/ -v /etc/localtime:/etc/localtime:ro --net=host ghcr.io/esphome/esphome:2023.11.0 dashboard --username omw --password '<redacted>' --address 127.0.0.1 /config`

Log:
```
2024-01-10 18:50:54,718 INFO Starting dashboard web server on http://127.0.0.1:6052 and configuration dir /config...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1003, in run_esphome
    return PRE_CONFIG_ACTIONS[args.command](args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 517, in command_dashboard
    return dashboard.start_web_server(args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/dashboard/dashboard.py", line 1509, in start_web_server
    status_thread.start()
  File "/usr/lib/python3.11/threading.py", line 957, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
```

ssieb, 2024-01-10:
@oyvindwe don't post on unrelated issues.  That is https://github.com/esphome/issues/issues/5142.

oyvindwe, 2024-01-10:
> @oyvindwe don't post on unrelated issues. That is #5142.

Sorry about that, but that was not in any way clear from the issue title!

ssieb, 2024-01-10:
If you can't find an existing issue, then create a new one, but don't hijack an unrelated one.

oyvindwe, 2024-01-10:
> If you can't find an existing issue, then create a new one, but don't hijack an unrelated one.

The title exactly matched my issue though, so I only wanted to help by contributing my log.

ssieb, 2024-01-10:
Titles are often ambiguous.  You always need to read the summary to see if it actually matches your issue.