esphome/issues #5332: USB_CDC not working on ESP32-C3 with Arduino

bearpawmaxim, 2024-01-03:
### The problem

Hello!
Cannot use hardware USB_CDC uart on my ESP32-C3 board.

The symptoms:
1. USB_CDC works (outputs some early esphome messages) for some time after the boot and then stops output.
2. Cannot choose USB_CDC as hardware uart for logger: looks like an error in esphome's logger config validation and code generation because it is actually supported by the mcu.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

logger

bearpawmaxim, 2024-01-09:
@kbx81 could you take a look at this?

kbx81, 2024-01-09:
Yeah, I'll have a look shortly. 😇

kbx81, 2024-01-10:
Hi -- please test https://github.com/esphome/esphome/pull/6069 and comment back if this fixes the issue for you.

```yaml
external_components:
  - source: github://pr#6069
    components: logger
    refresh: 0s
```

bearpawmaxim, 2024-01-10:
> Hi -- please test

It is now working even without setting the `hardware_uart: USB_CDC` (with also works), thanks!

closed 2024-01-10 by jesserockz