esphome/issues #5335: CAN bus doesn't work on ESP32S3 when GPIO19 & GPIO20 are used as RX, TX

strange-v, 2024-01-04:
### The problem

When the CAN bus is configured to use pins GPIO19 & GPIO20 as RX & TX - CAN communication doesn't work. The issue is that pins 19 and 20 are, by default, connected to the internal USB PHY and must be routed back to the GPIO matrix.

After adding
> gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[19], PIN_FUNC_GPIO);
gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[20], PIN_FUNC_GPIO);

everything works fine.

Probably, it is worth modifying the esp32_can component, so when the target is ESP32S3 and pins are 19 & 20, it applies the mentioned re-mappings. Alternatively, it might be done on a pin level. Sorry, I am not yet very familiar with the architecture of ESPHome, and it's hard to say precisely where the fix should be done.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.12.0

### What platform are you using?

ESP32

### Board

esp32-s3-devkitc-1

### Component causing the issue

esp32_can

### Example YAML snippet

```yaml
button:
  - platform: template
    name: Test
    id: my_button
    on_press:
      then:
        - canbus.send:
            data: [ 0x01, 0x02, 0x03 ]

canbus:
  - platform: esp32_can
    tx_pin: GPIO20
    rx_pin: GPIO19
    can_id: 2
    bit_rate: 125kbps
    on_frame:
    - can_id: 4
      use_extended_id: false
      then:
      - lambda: |-
          std::string b(x.begin(), x.end());
          ESP_LOGD("can id 4", "%s", &b[0] );
```

### Additional information

SN65HVD234 CAN bus transceiver is used in those tests

ssieb, 2024-01-05:
Why would you want to use those pins?  Are you also disabling the serial logs?

strange-v, 2024-01-05:
@ssieb, thank you for the replay.

To add more context. I use a premade device like this: https://carlightbulb.com/products/esp32-s3-86-4-inch-intelligent-control-panel-with-temperature-humidity-sensors, it exposes not so many pins, as most of them are used to drive the display.

I do not disable the serial log, as it uses TX0 & RX0, not GPIO19 & GPIO2.

ssieb, 2024-01-05:
I see, those are the USB data pins.  There isn't really any way to automatically switch them like that.  There would need to be some sort of config to indicate that you want to disable USB.

strange-v, 2024-01-05:
> There would need to be some sort of config to indicate that you want to disable USB.

It makes perfect sense.
Overall, ESPHome is quite user-friendly, so the complete solution may look like:
1. Provide a way to disable USB via the config.
1. Add a warning when pins 19 & 20 are used as general-purpose IO and USB is not disabled.

careyer, 2025-02-07:
@strange-v : Where do I need to add these lines? Does not look like it belongs in the YAML, does it?
```
gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[X], PIN_FUNC_GPIO);
gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[Y], PIN_FUNC_GPIO);
```

strange-v, 2025-02-16:
> Where do I need to add these lines? Does not look like it belongs in the YAML, does it?
@careyer, AFAIR, I was using an external component, and those lines were added to it.