esphome/issues #5337: Cannot update display components

00107, 2024-01-05:
### The problem

After one of the last ESPHome updates I can no longer update or change my Liligo displays. Previously everything worked fine, now I get error messages when trying to update the boards.

INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/tdisplay-2.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 458, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/739e7676/components/tdisplays3/display.py", line 107, in to_code
    await display.register_display(var, config)
  File "/esphome/esphome/components/display/__init__.py", line 119, in register_display
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID disp was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12 / 2024.1

### What platform are you using?

ESP32

### Board

Lilygo T-Display-S3 non-touch version

### Example YAML snippet

```yaml
# T-Display 320x170 Stromverbrauch
esphome: 
  name: "tdisplay-2"
  friendly_name: "tdisplay-2"

# für Batteriebetrieb
#  on_boot:
#    - priority: 90
#      then:
#      - output.turn_on: pin15on

external_components:
  - source: 
      type: git
      url: https://github.com/landonr/lilygo-tdisplays3-esphome/
      ref: sept2023
    components: [tdisplays3]

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino

deep_sleep:
time:
  - platform: homeassistant
    id: ha_time
    on_time:
        hours: 00
        minutes: 10
        then:
          - deep_sleep.enter:         
              until: "05:20:01"
              time_id: ha_time

logger:

api:
  encryption:
    key: "******"

ota:
  password: "******"

wifi:
  ssid: MyWifi
  password: *****

  ap:
    ssid: "S4 Fallback Hotspot"
    password: "*****"

captive_portal:

color:
- id: color_yellow
  red: 90%
  green: 90%
  blue: 20%
  white: 0%

- id: color_green
  red: 10%
  green: 90%
  blue: 25%
  white: 0%

- id: color_blue
  red_int: 0
  green_int: 191
  blue_int: 255
  white_int: 0

- id: color_white
  hex: EEEEEE

sensor: 

- platform: homeassistant
  id: growattpowerprozent
  entity_id: sensor.growatt_power_prozent
  accuracy_decimals: 0
  #%
  internal: true 

- platform: homeassistant
  id: growattpower
  entity_id: sensor.growatt_output_power
  accuracy_decimals: 0
  #WATT
  internal: true  

- platform: homeassistant
  id: stromzaehler
  entity_id: sensor.shelly_em_channel_1_power
  accuracy_decimals: 0
  #WATT 
  internal: true 

output:
# für Batteriebetrieb  8 2
#  - platform: gpio
#    pin: 15
#    id: pin15on
  - platform: ledc
    pin: GPIO38
    id: gpio38
    frequency: 2000

light:
- platform: monochromatic
  output: gpio38
  name: "Backlight"
  restore_mode: RESTORE_DEFAULT_ON  

image:
  - file: "watt.png"
    id: wattvertikal
    resize: 100x100

font:
  - file:
      type: gfonts
      family: Roboto
#      weight: 500
    id: roboto1
    size: 28

  - file:
      type: gfonts
      family: Roboto
      weight: 500
    id: roboto2
    size: 125

display:
  - platform: tdisplays3
    id: disp
    update_interval: 5s
    rotation: 270
    lambda: |-
      it.image(292, 15, id(wattvertikal));
      it.printf(276, 145, id(roboto2), color_white, TextAlign::BOTTOM_RIGHT, "%.0f", id(stromzaehler).state);
      it.line(0, 130, 320, 130);
      it.printf(8, 135, id(roboto1), color_yellow, "Solar:");
      it.printf(140, 135, id(roboto1), color_yellow, "%%");
      it.printf(135, 135, id(roboto1), color_yellow, TextAlign::TOP_RIGHT, "%.0f", id(growattpowerprozent).state);
      it.printf(180, 135, id(roboto1), color_yellow, "(%.0f W.)", id(growattpower).state);
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/tdisplay-2.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 458, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/739e7676/components/tdisplays3/display.py", line 107, in to_code
    await display.register_display(var, config)
  File "/esphome/esphome/components/display/__init__.py", line 119, in register_display
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID disp was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.
```

randybb, 2024-01-05:
The problem is in the external component, not in esphome - there have been done many changes related to display components, so old external components, especially something from sept 2023, will not work.

Checking that component - switch to main (remove `ref: sept2023`) branch and you should be good.

00107, 2024-01-05:
Thanks a lot!

closed 2024-01-05 by ssieb