esphome/issues #5339: text_sensor polling state ignoring the publish_state() order inside on_value automation

rpanachi, 2024-01-05:
### The problem

I have used a text_sensor to implement a state machine using the on_value automation to get the current state, execute an action and update to the next state (see .yaml file to have better context).

The logic and the sequence of state machines works as expected (that could be confirmed on logs). But the only problem is that the text_sensor wasn't updating correctly on frontend (HA interface). When my script change the state many times, looks like the sensor only show the first one (check Logbook screenshot), or wasn't not respecting the order that the states was changed.

To be more clear: sensor should be displaying "state_4" (the last state published), but is displaying "state_1" (the last state received on sensor/frontend):
[image]

If you look the logs, the state was published in the correct sequence:
[image]

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

esp32dev (ESP32-WROOM-32U)

### Component causing the issue

text_sensor

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-ceb020
  friendly_name: ESPHome Web ceb020

  on_boot:
    priority: "-100"
    then:
      - delay: 30s
      # start state machine
      - lambda: |-
          id(txt_current_state).publish_state("state_1");

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "[redacted]"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-Ceb020"
    password: "[redacted]"

captive_portal:
    
text_sensor:
  - platform: template
    id: txt_current_state
    name: "Current State"
    on_value:
      - logger.log:
          format: "Current State is %s"
          args: ["x.c_str()"]
      - if:
          condition:
            - text_sensor.state:
                id: txt_current_state
                state: "state_1"
          then:
            - lambda: |-
                id(txt_current_state).publish_state("state_2");
      - if:
          condition:
            - text_sensor.state:
                id: txt_current_state
                state: "state_2"
          then:
            - text_sensor.template.publish:
                id: txt_current_state
                state: "state_3"
      - if:
          condition:
            - text_sensor.state:
                id: txt_current_state
                state: "state_3"
          then:
            - lambda: |-
                id(txt_current_state).publish_state("state_4");
      - if:
          condition:
            - text_sensor.state:
                id: txt_current_state
                state: "state_4"
          then:
            - delay: 30s
            # restart state machine
            - text_sensor.template.publish:
                id: txt_current_state
                state: "state_1"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/esphome-web-ceb020.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-ceb020 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/esphome-web-ceb020/src/main.cpp.o
Linking .pioenvs/esphome-web-ceb020/firmware.elf
RAM:   [=         ]  12.4% (used 40728 bytes from 327680 bytes)
Flash: [=====     ]  49.7% (used 911165 bytes from 1835008 bytes)
Building .pioenvs/esphome-web-ceb020/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esphome-web-ceb020/firmware.bin"], [".pioenvs/esphome-web-ceb020/firmware.elf"])
Wrote 0xefdd0 bytes to file /data/build/esphome-web-ceb020/.pioenvs/esphome-web-ceb020/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 17.36 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.178
INFO Uploading /data/build/esphome-web-ceb020/.pioenvs/esphome-web-ceb020/firmware.bin (916944 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 4.35 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.178 using esphome API
INFO Successfully connected to esphome-web-ceb020 @ 192.168.1.178 in 7.266s
INFO Successful handshake with esphome-web-ceb020 @ 192.168.1.178 in 0.132s
[18:42:16][I][app:102]: ESPHome version 2023.12.5 compiled on Jan  5 2024, 18:41:47
[18:42:16][C][wifi:573]: WiFi:
[18:42:16][C][wifi:405]:   Local MAC: [redacted]
[18:42:16][C][wifi:410]:   SSID: [redacted]
[18:42:16][C][wifi:411]:   IP Address: 192.168.1.178
[18:42:16][C][wifi:413]:   BSSID: [redacted]
[18:42:16][C][wifi:414]:   Hostname: 'esphome-web-ceb020'
[18:42:16][C][wifi:416]:   Signal strength: -54 dB ▂▄▆█
[18:42:16][C][wifi:420]:   Channel: 1
[18:42:16][C][wifi:421]:   Subnet: 255.255.255.0
[18:42:16][C][wifi:422]:   Gateway: 192.168.1.1
[18:42:16][C][wifi:423]:   DNS1: 192.168.1.1
[18:42:16][C][wifi:424]:   DNS2: 0.0.0.0
[18:42:16][C][logger:443]: Logger:
[18:42:16][C][logger:444]:   Level: DEBUG
[18:42:16][C][logger:445]:   Log Baud Rate: 115200
[18:42:16][C][logger:447]:   Hardware UART: UART0
[18:42:16][C][template.text_sensor:020]: Template Sensor 'Current State'
[18:42:16][C][captive_portal:088]: Captive Portal:
[18:42:16][C][mdns:115]: mDNS:
[18:42:16][C][mdns:116]:   Hostname: esphome-web-ceb020
[18:42:16][C][ota:097]: Over-The-Air Updates:
[18:42:16][C][ota:098]:   Address: esphome-web-ceb020.local:3232
[18:42:16][C][api:139]: API Server:
[18:42:16][C][api:140]:   Address: esphome-web-ceb020.local:6053
[18:42:16][C][api:142]:   Using noise encryption: YES
[18:42:45][D][text_sensor:064]: 'Current State': Sending state 'state_1'
[18:42:45][D][main:280]: Current State is state_1
[18:42:45][D][text_sensor:064]: 'Current State': Sending state 'state_2'
[18:42:45][D][main:280]: Current State is state_2
[18:42:45][D][text_sensor:064]: 'Current State': Sending state 'state_3'
[18:42:45][D][main:280]: Current State is state_3
[18:42:45][D][text_sensor:064]: 'Current State': Sending state 'state_4'
[18:42:45][D][main:280]: Current State is state_4
[18:42:45][W][component:214]: Component esphome.coroutine took a long time for an operation (0.07 s).
[18:42:45][W][component:215]: Components should block for at most 20-30ms.
[18:43:15][D][text_sensor:064]: 'Current State': Sending state 'state_1'
[18:43:15][D][main:280]: Current State is state_1
[18:43:15][D][text_sensor:064]: 'Current State': Sending state 'state_2'
[18:43:15][D][main:280]: Current State is state_2
[18:43:15][D][text_sensor:064]: 'Current State': Sending state 'state_3'
[18:43:15][D][main:280]: Current State is state_3
[18:43:15][D][text_sensor:064]: 'Current State': Sending state 'state_4'
[18:43:15][D][main:280]: Current State is state_4
[18:43:15][W][component:214]: Component text_sensor took a long time for an operation (0.08 s).
[18:43:15][W][component:215]: Components should block for at most 20-30ms.
```

ssieb, 2024-01-05:
That's a very bad idea.  You're not letting it actually send to HA until the end.  I'm guessing the messages are getting queued in reverse order or something.  Notice the warning about taking a long time.  If you could reproduce this by publishing values one at a time from outside the component, then there would be an issue.

closed 2024-01-05 by ssieb

rpanachi, 2024-01-10:
> You're not letting it actually send to HA until the end.

@ssieb but shouldn't this be the default behavior of this `on_value` automation (triggered after the value was sent to HA)? If it handles the internal value change, maybe the automation should be renamed to `on_value_internal` and, following the same logic here, another `on_value_complete` should be added to this component, triggered only AFTER the value was sent to HA. There is nothing on the official documentation about this behavior.