esphome/issues #5346: include/ssl.h missing when using MQTT with ssl_fingerprints

Hubertilus, 2024-01-07:
### The problem

Hello,

I'm trying to build a ESPHOME for my ESP8266 using MQTT with SSL fingerprints. But it's not compiling because of missing ssl.h file.
If I remove the ssl_fingerprint line, compilation works fine.

With that line, I always run into the following compilation issue:
`In file included from .piolibdeps/test2/AsyncMqttClient-esphome/src/AsyncMqttClient.hpp:23,`
`                 from .piolibdeps/test2/AsyncMqttClient-esphome/src/AsyncMqttClient.h:4,`
`                 from src/esphome/components/mqtt/mqtt_backend_esp8266.h:6,`
`                 from src/esphome/components/mqtt/mqtt_client.h:15,`
`                 from src/esphome/components/mqtt/custom_mqtt_device.h:7,`
`                 from src/esphome/components/mqtt/custom_mqtt_device.cpp:1:`
`.piolibdeps/test2/ESPAsyncTCP-esphome/src/tcp_axtls.h:44:10: fatal error: include/ssl.h: No such file or directory`

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

MQTT

### Example YAML snippet

```yaml
esphome:
  name: test

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: 192.168.0.14
  port: 8883
  username: !secret MQTT_USER
  password: !secret MQTT_PASSWORD
  ssl_fingerprints: !secret MQTT_SSL_FINGERPRINT
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.5
INFO Reading configuration /config/test2.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test2 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif8266 @ 3.2.0
INFO Installing platformio/espressif8266 @ 3.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif8266@3.2.0 has been installed!
INFO espressif8266@3.2.0 has been installed!
Tool Manager: Installing platformio/toolchain-xtensa @ ~2.100300.0
INFO Installing platformio/toolchain-xtensa @ ~2.100300.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa@2.100300.220621 has been installed!
INFO toolchain-xtensa@2.100300.220621 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif8266 @ ~3.30002.0
INFO Installing platformio/framework-arduinoespressif8266 @ ~3.30002.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif8266@3.30002.0 has been installed!
INFO framework-arduinoespressif8266@3.30002.0 has been installed!
Tool Manager: Installing platformio/tool-esptool @ <2
INFO Installing platformio/tool-esptool @ <2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptool@1.413.0 has been installed!
INFO tool-esptool@1.413.0 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.30000.0
INFO Installing platformio/tool-esptoolpy @ ~1.30000.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.30000.201119 has been installed!
INFO tool-esptoolpy@1.30000.201119 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40502.0
INFO Installing platformio/tool-scons @ ~4.40502.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40502.0 has been installed!
INFO tool-scons@4.40502.0 has been installed!
Library Manager: Installing heman/AsyncMqttClient-esphome @ 2.0.0
INFO Installing heman/AsyncMqttClient-esphome @ 2.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncMqttClient-esphome@2.0.0 has been installed!
INFO AsyncMqttClient-esphome@2.0.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/ESPAsyncTCP-esphome @ ^2.0.0
INFO Installing esphome/ESPAsyncTCP-esphome @ ^2.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@2.0.0 has been installed!
INFO ESPAsyncTCP-esphome@2.0.0 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/test2/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/test2/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/test2/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/test2/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/test2/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/test2/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/test2/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/test2/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/test2/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/test2/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/test2/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
In file included from .piolibdeps/test2/AsyncMqttClient-esphome/src/AsyncMqttClient.hpp:23,
                 from .piolibdeps/test2/AsyncMqttClient-esphome/src/AsyncMqttClient.h:4,
                 from src/esphome/components/mqtt/mqtt_backend_esp8266.h:6,
                 from src/esphome/components/mqtt/mqtt_client.h:15,
                 from src/esphome/components/mqtt/custom_mqtt_device.h:7,
                 from src/esphome/components/mqtt/custom_mqtt_device.cpp:1:
.piolibdeps/test2/ESPAsyncTCP-esphome/src/tcp_axtls.h:44:10: fatal error: include/ssl.h: No such file or directory
   44 | #include "include/ssl.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test2/src/esphome/components/mqtt/custom_mqtt_device.cpp.o] Error 1
```

### Additional information

I'm running ESPHOME in a docker environment. But I also did install ESPHOME on my Linux command line, like it is described [here](https://esphome.io/guides/installing_esphome.html).
But this leads to the same result. So it doesn't look like it's related to the installation itself.

Further I did try
`esp8266:`
`  board: d1_mini`
`  framework:`
`    version: latest`
and
`esp8266:`
`  board: d1_mini`
`  framework:`
`    version: latest`
`    platform_version: 4.2.1`

Maybe the issue is related to a missing dependency? Here is how the auto created platform.ini looks like:
`; Auto generated code by esphome`
``
`[common]`
`lib_deps =`
`build_flags =`
`upload_flags =`
``
`; ========== AUTO GENERATED CODE BEGIN ===========`
`[platformio]`
`description = ESPHome 2023.12.5`
`[env:test2]`
`board = d1_mini`
`board_build.flash_mode = dout`
`board_build.ldscript = eagle.flash.4m.ld`
`build_flags =`
`    -DASYNC_TCP_SSL_ENABLED=1`
`    -DNEW_OOM_ABORT`
`    -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH`
`    -DUSE_ARDUINO`
`    -DUSE_ESP8266`
`    -DUSE_ESP8266_FRAMEWORK_ARDUINO`
`    -Wno-nonnull-compare`
`    -Wno-sign-compare`
`    -Wno-unused-but-set-variable`
`    -Wno-unused-variable`
`    -fno-exceptions`
`extra_scripts =`
`    post:post_build.py`
`framework = arduino`
`lib_deps =`
`    ESP8266WiFi`
`    ESP8266mDNS`
`    heman/AsyncMqttClient-esphome@2.0.0`
`    bblanchon/ArduinoJson@6.18.5`
`    ${common.lib_deps}`
`lib_ldf_mode = off`
`platform = platformio/espressif8266@3.2.0`
`platform_packages =`
`    platformio/framework-arduinoespressif8266@~3.30002.0`
`; =========== AUTO GENERATED CODE END ============`

bialeta, 2024-01-09:
Got the same issue on 2023.12.5no solution so far.

romanlinhard, 2024-04-07:
The same here with actual 2024.3.1 version. Bug stil exists.

ssieb, 2024-04-10:
Duplicate of https://github.com/esphome/issues/issues/2660.

closed 2024-04-10 by ssieb