esphome/issues #5347: Latest esphome update make pillow problems

Heiner1967, 2024-01-08:
### The problem

As requested by latest esp update I upgrade pillow to 10.1.0.
However esphome still complain that I shall upgrade pillow.
Checking the pillow version it is 10.1.0.
Reboot does not help either.

I use  raspberry Pi with Debian Bullseye

Can you help me?

### Which version of ESPHome has the issue?

Esp 2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

Ili9xxx

ssieb, 2024-01-08:
How are you actually running esphome because your descriptions don't match?

Heiner1967, 2024-01-08:
I do not see the conflict neither I understand your question.

I did run a older Version (think it was from October or november so far) which required so far pillow 10.0.1 and it run fine.
Now few days back I upgared to the latest December Version and ESPhome afterwards require to update pillow to 10.1.0.

So far all good.

But the update which went successfully, ESPHOME still tells me to update pillow to 10.1.0 even its there already.

Heiner1967, 2024-01-08:
....and i mention the component ili9xxx as the related lines of Code trigger the pillow demand. without discplay component in my code it seems no pillow is needed and the yaml can we compiled and loaded to my ESP.

ssieb, 2024-01-08:
You say you're running esphome as an HA addon, but you also say you're running Debian on the pi and you don't give the HA version.  So how are you actually running esphome?

Heiner1967, 2024-01-09:
Hi, ah, sorry for that.

I could not choose anything else than HA.

In effect I use a Rasberry PI with debian on it. Then I use fhem and fhempy (https://github.com/fhempy/fhempy) which provide me the ESPhome Application.
You are right HA is not Debian. Sorry for that confusion.

ssieb, 2024-01-09:
That is the right place to file an issue.  This isn't a supported installation method and not something we can help you with.

closed 2024-01-09 by ssieb