esphome/issues #5352: ESPHome Thermostat should have a single setpoint  for when auto is not used.

towerclimber87, 2024-01-09:
### The problem

ESPHome Thermostat should have a single setpoint  for when auto is not used.   example. I want the temp to be set to 70 and i have mode set to Cool. then cool room to 70, when Heat is selected, heat room to 70.  with the curent setup. Amazon Alexa is not able to change the set temp.  Home Assistant has a generic thermostat that works well and has a single setpoint.  for those that dont want manual. 

### Which version of ESPHome has the issue?

Current version: 2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.2

### What platform are you using?

ESP32

### Board

Wemos D1 Mini ESP32

### Component causing the issue

Cliamte

### Example YAML snippet

```yaml
climate:
  - platform: thermostat
    name: "Livingroom Thermostat"
    id: climate_thermostat
    sensor: livingroom_temp_onboard_1
    min_heating_off_time: 30s
    min_heating_run_time: 30s
    min_cooling_off_time: 30s
    min_cooling_run_time: 30s    
    set_point_minimum_differential: 0
    min_idle_time: 30s
    cool_action:
      - switch.turn_on: ac_relay
      - switch.turn_off: heater_relay
      - homeassistant.service:
          service: input_boolean.turn_off
          data_template:
             entity_id: input_boolean.livingroom_heat_trigger
      - homeassistant.service:
          service: input_boolean.turn_on
          data_template:
             entity_id: input_boolean.livingroom_ac_trgger
    heat_action:
      - switch.turn_on: heater_relay
      - switch.turn_off: ac_relay
      - homeassistant.service:
          service: input_boolean.turn_on
          data_template:
             entity_id: input_boolean.livingroom_heat_trigger
      - homeassistant.service:
          service: input_boolean.turn_off
          data_template:
             entity_id: input_boolean.livingroom_ac_trgger
    idle_action:
      - switch.turn_off: ac_relay
      - switch.turn_off: heater_relay
      - homeassistant.service:
          service: input_boolean.turn_off
          data_template:
             entity_id: input_boolean.livingroom_heat_trigger
      - homeassistant.service:
          service: input_boolean.turn_off
          data_template:
             entity_id: input_boolean.livingroom_ac_trgger
    default_preset: Home
    preset:
      - name: Home
        default_target_temperature_low: 70 °F
        default_target_temperature_high: 70 °F        
    visual:
      min_temperature: 18.89
      max_temperature: 25
      temperature_step: 1
```

### Anything in the logs that might be useful for us?

```txt
This is the HA that works well. 
  - platform: dualmode_generic
    name: Bedroom Thermostat
    heater: input_boolean.bedroom_heat_trgger
    cooler: input_boolean.bedroom_ac_trgger
    target_sensor: sensor.h5075_23cd_temperature
    min_temp: 63
    max_temp: 78
    target_temp: 68
    cold_tolerance: 0.5
    hot_tolerance: 0.5
    min_cycle_duration:
      minutes: 1
    keep_alive:
      minutes: 3
    initial_hvac_mode: "off"
    away_temp: 63
    precision: 1.0
```

towerclimber87, 2024-04-04:
Just wanted to see if anyone has looked into this. At the moment, Alexa is unable to change the temperature.

jonoberheide, 2025-02-09:
@towerclimber87 Were you ever able to resolve this?

To clarify, are you referring to the fact that the thermostat devices shows both a heating/cooling setpoint even when in only-heating or only-cooling mode? I'm experiencing that too and trying to figure out how to make the esphome climate thermostat match the behavior of the HA thermostat more closely.

[image]

towerclimber87, 2025-02-09:
I was not.   I keep hoping someone will.

jonoberheide, 2025-02-09:
I'll take a look at it!

jonoberheide, 2025-02-11:
This is somehow related to the ClimateEntityFeature TARGET_TEMPERATURE vs TARGET_TEMPERATURE_RANGE and how set_temperature() uses the target_temperature vs target_temperature_high/low. If the esphome home device advertises dual setpoints (eg. by having a cooling and heating action), it will set supports_two_point_target_temperature=True and therefore set the TARGET_TEMPERATURE_RANGE feature.

This behavior is easy to override in the code, but I'm not actually sure the proper way to fix it. Aka, should the HA-side switch ClimateEntityFeatures based on the HVAC mode? Or should this be driven by the esphome side somehow instead? /cc @kbx81 ?

towerclimber87, 2025-03-06:
Has there been any update on this? this is driving me crazy.

kbx81, 2025-08-27:
Is there an example of a component that does this? HA's "generic thermostat" helper only supports a single climate mode & action, so it only ever shows a single handle in the UI.

kbx81, 2025-09-09:
Well, I think https://github.com/home-assistant/core/pull/151897 and https://github.com/esphome/esphome/pull/10561 satisfy this nicely. 🙂

towerclimber87, 2025-10-01:
> The fix for this issue caused another regression:
> home-assistant/core#153331

I’m curious at this point if we’re all talking about the same issue.  I
haven’t noticed any changes in the way my thermostats worked when I updated
them.  If anything, my nextion screens stopped working

jonoberheide, 2025-10-02:
@towerclimber87 Did you update HA to 2025.10.0 AND update/reflash esphome to 2025.9.0? Both sides of the interface need to be updated.

I just updated to 2025.10 and the changes are working as expected with the single setpoint!🎉