esphome/issues #5355: ENS160 requires one hour warmup each boot (should be 3min)

jxmesw1, 2024-01-09:
### The problem

[According to the docs](https://esphome.io/components/sensor/ens160.html), the ENS160 eCO2/tVOC sensor requires a one-hour warmup the _very first time_ it's powered up. For subsequent boots or restarts, it's three minutes before readings are available. 
However, I'm experiencing an issue whereby I have to wait a full hour for readings every time I reboot my ESP. This should be three minutes rather than an hour.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.2

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

ENS160

### Example YAML snippet

```yaml
- platform: ens160
    eco2:
      name: "Lounge eCO2"
    tvoc:
      name: "Lounge TotalVOC"
    aqi:
      name: "Lounge AQI"
    update_interval: 30s
    address: 0x53
    compensation:
      temperature: "lounge_temp"
      humidity: "lounge_humi"
```

### Anything in the logs that might be useful for us?

```txt
[14:26:08][VV][i2c.arduino:141]: 0x53 TX 20
[14:26:08][VV][i2c.arduino:117]: 0x53 RX 8B
[14:26:08][V][ens160:156]: Status: ENS160 STATAS bit    0x1
[14:26:08][V][ens160:158]: Status: ENS160 STATER bit    0x0
[14:26:08][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x02
[14:26:08][V][ens160:161]: Status: ENS160 NEWDAT bit    0x1
[14:26:08][V][ens160:163]: Status: ENS160 NEWGPR bit    0x1
```

### Additional information

I sometimes see a message in the log saying something along the lines of "One hour warmup required..." although this is not always the case. I think this depends on when I look at the log, e.g. as soon as the ESP has booted

jxmesw1, 2024-01-10:
Update: this is the message I see when looking at the logs during the boot:
[I][ens160:179]: ENS160 readings unavailable - 1 hour startup required after first power on

stlnick2, 2024-02-06:
Same here. Any updates?

jxmesw1, 2024-02-07:
I’m glad I’m not the only one - I was starting to think it was a hardware-related issue.

When I get chance I’ll look at the source code and see if there’s anything obvious, although I’m not an expert in this area.  I’ll let you know how I get on.

Komzpa, 2024-02-12:
[Rumor] While trying to resolve the issue with 0 readings, I read somewhere that the sensor writes to permanent memory only once 24h. So if you started it, went through the warmup, you need to keep it running for 23 more hours to store that calibration. [/Rumor]

Remboooo, 2024-07-28:
It's not just a rumour, it's in the datasheet: 
[image]

BenoitURRUTY, 2025-01-31:
Hi, 
I ran it during 24 hours twice and its still start in status 2. Does someone have the same issues?

Gfvm, 2025-05-27:
@BenoitURRUTY I have the same issue. The only thing I noticed is that I had some periods where the connections to my server wasn't stable, so it didn't read any data. But i think it should not impact the start up sequence.

14790897, 2025-08-26:
same

[09:52:17][I][ens160:179]: ENS160 readings unavailable - 1 hour startup required after first power on