esphome/issues #5356: CAN Bus using mcp2515 not receiving extended id messages

SpenZerX, 2024-01-09:
### The problem

Hello,

esphome on 1st gen dual core ESP 32 (board: esp32dev)

component mcp2515 is not receiving any can messages with extended id.

Transmitted can messages with extended id are received by the destination, Transmitting seems to work.

Enabled mode: LOOPBACK for testing
There is no echo on can messages sent with extended id!

Echo is OK on messages sent without extended ID -> SPI communication should be ok.

May be old issue #2642 is not fixed?
Similar to issue #4885
https://community.home-assistant.io/t/problems-with-extended-id-esphome-mcp2515-can-bus/614512

 

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

tried 2 different types

### Component causing the issue

integration: mcp2515

### Example YAML snippet

```yaml
TESTS ARE DONE WITH mode: LOOPBACK - No echo with extended IDs

spi:
  id: McpSpi
  clk_pin:  GPIO16
  mosi_pin: GPIO5
  miso_pin: GPIO4

canbus:
  - platform: mcp2515
    id: my_mcp2515
    spi_id: McpSpi
    cs_pin: GPIO14
    can_id: 0x04
    clock: 8MHZ
    bit_rate: 125kbps
    mode: LOOPBACK
    on_frame:

...

interval:
  - interval: 10sec
    then:
    - canbus.send:
        data: [ 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0 ]
        use_extended_id: true
 #      use_extended_id: false
        can_id: 0x4
 #      can_id: 0x108040FE
```

### Anything in the logs that might be useful for us?

```txt
TESTS ARE DONE WITH mode: LOOPBACK - No echo with extended IDs

----------------

Log with setting "use_extended_id: true"

[20:53:52][D][canbus:032]: send extended id=0x00000004 rtr=FALSE size=8
[20:53:52][VV][canbus:046]:   data[0]=00
[20:53:52][VV][canbus:046]:   data[1]=00
[20:53:52][VV][canbus:046]:   data[2]=00
[20:53:52][VV][canbus:046]:   data[3]=00
[20:53:52][VV][canbus:046]:   data[4]=00
[20:53:52][VV][canbus:046]:   data[5]=00
[20:53:52][VV][canbus:046]:   data[6]=00
[20:53:52][VV][canbus:046]:   data[7]=00
------------------

Log with setting "use_extended_id: false"

[20:49:39][D][canbus:035]: send standard id=0x004 rtr=FALSE size=8
[20:49:39][VV][canbus:046]:   data[0]=00
[20:49:39][VV][canbus:046]:   data[1]=00
[20:49:39][VV][canbus:046]:   data[2]=00
[20:49:39][VV][canbus:046]:   data[3]=00
[20:49:39][VV][canbus:046]:   data[4]=00
[20:49:39][VV][canbus:046]:   data[5]=00
[20:49:39][VV][canbus:046]:   data[6]=00
[20:49:39][VV][canbus:046]:   data[7]=00
[20:49:39][D][canbus:072]: received can message (#1) std can_id=0x4 size=8
[20:49:39][V][canbus:079]:   can_message.data[0]=00
[20:49:39][V][canbus:079]:   can_message.data[1]=00
[20:49:39][V][canbus:079]:   can_message.data[2]=00
[20:49:39][V][canbus:079]:   can_message.data[3]=00
[20:49:39][V][canbus:079]:   can_message.data[4]=00
[20:49:39][V][canbus:079]:   can_message.data[5]=00
[20:49:39][V][canbus:079]:   can_message.data[6]=00
[20:49:39][V][canbus:079]:   can_message.data[7]=00
```

DaanVervacke, 2024-02-01:
I'm having the same exact problem.

Sleeper85, 2024-08-21:
Same problem, `MCP2515` does not receive the `extended can_id` but it can send them and I receive them well with a `SN65HVD230` board.

p1ngb4ck, 2025-08-26:
same problem still persists, looking into it right now

closed 2025-10-06 by swoboda1337