esphome/issues #5357: The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h

riverscn, 2024-01-10:
### The problem

driver/rmt.h is deprecated.
Should use driver/rmt_tx.h and/or driver/rmt_rx.h instead.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

m5stack-atoms3

### Component causing the issue

esp32_rmt_led_strip

### Example YAML snippet

```yaml
light:
  - platform: esp32_rmt_led_strip
    name: "Indicator"
    id: "indicator"
    rgb_order: GRB
    pin: GPIO35
    num_leds: 1
    rmt_channel: 0
    chipset: ws2812
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/esp32_rmt_led_strip/led_strip.h:12,
                 from src/esphome.h:22,
                 from src/main.cpp:3:
/config/.esphome/platformio/packages/framework-espidf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
```

randybb, 2024-01-10:
We are still using ESP-IDF 4.4, this deprecation warning was introduced in ESP-IDF 5.3. If you are overriding version, you need to deal with all issues. Anyway, this is just a warning and the component will be adjusted once we move to that version.

asucrews, 2024-06-20:
witch the conversation to esp-idf 5. i thinks this should be stale.

rvalitov, 2024-10-14:
For those who would like to make a PR - check the changes introduces in ESP IDF 5 https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s2/migration-guides/release-5.x/peripherals.html#rmt-driver

swoboda1337, 2024-11-25:
> For those who would like to make a PR - check the changes introduces in ESP IDF 5 https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s2/migration-guides/release-5.x/peripherals.html#rmt-driver

I opened a PR here https://github.com/esphome/esphome/pull/7770 works well but won't get merged without a lot more work.

closed 2024-12-19 by kbx81