esphome/issues #5360: !extend scope question

Masterz69, 2024-01-10:
### The problem

Main ```device.yaml```:
```yaml
package:
  offset: !include common/package/filter_temperature_offset.yaml

sensor:
  - platform: aht10
    temperature:
      name: "Temperature"
      id: sensor_temperature
```

File ```common/package/filter_temperature_offset.yaml```:
```yaml
sensor:  
  - id: !extend sensor_temperature
    filters:
      - lambda: return x + id(number_temperature_offset).state;
```

Getting error ```Source for extension of ID 'sensor_temperature' was not found.```.

Looks !extend in package can't extend component from main config, only from another package !?
Would be nice to do it possible, it can be usefull IMHO.

### Which version of ESPHome has the issue?

2023.12.5.

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

Unfocused, 2024-01-12:
From my reading of the review comments of the PR that added this, this is the intended behaviour (and to change it would be very complicated).

- The main file can extend something defined in a package
- A package can extend something defined in an earlier package
- A package can't extend something defined in a later package
- A package can't extend something defined in the main file

See the discussion on this here: https://github.com/esphome/esphome/pull/3555/files#r901821977

To summarize that discussion, it's preserving existing behaviour in terms of the main file always being the overriding source of truth. Also, package order matters - later packages override earlier packages (in a way, the contents of the main file can be thought of as a package that's always included last).

So for example:
```yaml
packages:
  rev2:
    foo: 2
foo: 1
```

The value of `foo` will be 1, not 2. The same logic/order applies to `!extend`.

Personally, I can see some use cases for wanting a package to extend something defined in the main file. But at the same time, I also see the current behaviour as a safety/security feature: If you're using remote packages provided by someone else, then you want your main file to be the final source of truth.

But if you want to get around this limitation, then put the thing you want extended in its own package (just make sure it's included _before_ the package with the `!extend`).

closed 2024-01-12 by ssieb

Masterz69, 2024-04-17:
> From my reading of the review comments of the PR that added this, this is the intended behaviour (and to change it would be very complicated).
>
> * The main file can extend something defined in a [...]

Just noticed - package can't `!extend` included packages.

Main file:
```yaml
packages:
  xxx:
```
Package `xxx`:
```yaml
packages:
  yyy:
    foo: 1
foo: 2
``` 
it throw error.

IMHO package `yyy` not to be treated as later package !