esphome/issues #5361: ili93xxx just does not work

Grey-Lancaster, 2024-01-11:
### The problem

It gives the error 
[19:23:34][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:23:34][W][component:215]: Components should block for at most 20-30ms.

some say login will fix that

But the real problem is sensors do not work
https://community.home-assistant.io/t/sensor-display-stopped-working-esphome-and-ili9xxx/602881

### Which version of ESPHome has the issue?

OS 11.2 core 2023.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

esp32 devc

### Component causing the issue

ili9341

### Example YAML snippet

```yaml
esphome:
  name: esp32tft3
  friendly_name: ESP32TFT3

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "MQjSIhgSVFfwUZ7F/BxWB7/Hy04eb8bR82NFW1JgwgY="

ota:
  password: "9386be9ffac869b6a3bbed69801ebe2e"

wifi:
  networks:

  - ssid: "thegreyfox"
    password: ""
  - ssid: "shop2"
    password: "mine0313"
  - ssid: "fireplace"  
    password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Tft3 Fallback Hotspot"
    password: "mqSdhRBOcqy6"

captive_portal:

    
time:
  - platform: homeassistant
    id: esptime
sensor:
  - platform: homeassistant
    id: outside_temperature
    entity_id: sensor.office_outside_temp_2
    internal: true

  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.big_room_temperature
    internal: true
color:
  - id: my_light_red
    hex: FFFF00  #yello
  - id: my_black
    hex: '000000'  # Hex code for black

font:
# - file: 'Roboto.ttf'
#    id: font1
#   size: 26

  - file:
      type: gfonts
      family: Roboto
      weight: 900
    id: font1
    size: 26     

spi:
 clk_pin: 18
 mosi_pin: 23
 miso_pin: 19
 
# Example configuration entry

display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: 14
    dc_pin: 27 
    reset_pin: 33
    data_rate: 20Mhz   
    spi_mode: mode0
    lambda: |-
      it.fill(id(my_black));   
      it.print(20, 0, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Hello WIZARD!");

      it.print(10, 30, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Time:");
      char buffer[15];
      auto time = id(esptime).now();
      int hour = time.hour % 12; // convert to 12-hour format
      hour = hour ? hour : 12; // handle midnight and noon
      sprintf(buffer, "%d:%02d", hour, time.minute);
      it.print(90, 30, id(font1), id(my_light_red), TextAlign::TOP_LEFT,  buffer);

      it.printf(25, 55, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "In:");
      if (id(inside_temperature).has_state()) {
      it.printf(55, 45, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "%.1f°", id(inside_temperature).state);
      }
#      it.print(0, 20, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "time");
# Example configuration entry
# touchscreen:
#   platform: xpt2046
#   id: my_touchscreen
#   cs_pin: 12
#   interrupt_pin: 16
#   update_interval: 50ms
#   threshold: 400
#   calibration_x_min: 0
#   calibration_x_max: 3860
#   calibration_y_min: 0
#   calibration_y_max: 3860
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/esp32tft3.yaml...
INFO Detected timezone 'America/New_York'
INFO Starting log output from 192.168.1.215 using esphome API
INFO Successfully connected to esp32tft3 @ 192.168.1.215 in 0.084s
INFO Successful handshake with esp32tft3 @ 192.168.1.215 in 0.112s
[19:31:10][I][app:102]: ESPHome version 2023.12.5 compiled on Jan 10 2024, 18:54:31
[19:31:10][C][wifi:573]: WiFi:
[19:31:10][C][wifi:405]:   Local MAC: EC:62:60:A2:6C:18
[19:31:10][C][wifi:410]:   SSID: [redacted]
[19:31:10][C][wifi:411]:   IP Address: 192.168.1.215
[19:31:10][C][wifi:413]:   BSSID: [redacted]
[19:31:10][C][wifi:414]:   Hostname: 'esp32tft3'
[19:31:10][C][wifi:416]:   Signal strength: -58 dB ▂▄▆█
[19:31:10][C][wifi:420]:   Channel: 11
[19:31:10][C][wifi:421]:   Subnet: 255.255.255.0
[19:31:10][C][wifi:422]:   Gateway: 192.168.1.1
[19:31:10][C][wifi:423]:   DNS1: 192.168.1.1
[19:31:10][C][wifi:424]:   DNS2: 8.8.8.8
[19:31:10][C][logger:443]: Logger:
[19:31:10][C][logger:444]:   Level: DEBUG
[19:31:10][C][logger:445]:   Log Baud Rate: 115200
[19:31:10][C][logger:447]:   Hardware UART: UART0
[19:31:10][C][spi:067]: SPI bus:
[19:31:10][C][spi:068]:   CLK Pin: GPIO18
[19:31:10][C][spi:069]:   SDI Pin: GPIO19
[19:31:10][C][spi:070]:   SDO Pin: GPIO23
[19:31:10][C][spi:072]:   Using HW SPI: SPI
[19:31:10][C][ili9xxx:069]: ili9xxx
[19:31:10][C][ili9xxx:069]:   Rotations: 0 °
[19:31:10][C][ili9xxx:069]:   Dimensions: 240px x 320px
[19:31:10][C][ili9xxx:070]:   Width Offset: 0
[19:31:10][C][ili9xxx:071]:   Height Offset: 0
[19:31:10][C][ili9xxx:080]:   Color mode: 8bit 332 mode
[19:31:10][C][ili9xxx:086]:   Data rate: 20MHz
[19:31:10][C][ili9xxx:088]:   Reset Pin: GPIO33
[19:31:10][C][ili9xxx:089]:   CS Pin: GPIO14
[19:31:10][C][ili9xxx:090]:   DC Pin: GPIO27
[19:31:10][C][ili9xxx:092]:   Swap_xy: NO
[19:31:10][C][ili9xxx:093]:   Mirror_x: YES
[19:31:10][C][ili9xxx:094]:   Mirror_y: NO
[19:31:10][C][ili9xxx:099]:   Update Interval: 1.0s
[19:31:10][C][homeassistant.time:010]: Home Assistant Time:
[19:31:10][C][homeassistant.time:011]:   Timezone: 'EST5EDT,M3.2.0,M11.1.0'
[19:31:10][C][psram:020]: PSRAM:
[19:31:10][C][psram:021]:   Available: NO
[19:31:10][C][captive_portal:088]: Captive Portal:
[19:31:10][C][mdns:115]: mDNS:
[19:31:10][C][mdns:116]:   Hostname: esp32tft3
[19:31:10][C][ota:097]: Over-The-Air Updates:
[19:31:10][C][ota:098]:   Address: esp32tft3.local:3232
[19:31:10][C][ota:101]:   Using Password.
[19:31:10][C][api:139]: API Server:
[19:31:10][C][api:140]:   Address: esp32tft3.local:6053
[19:31:10][C][api:142]:   Using noise encryption: YES
[19:31:10][C][homeassistant.sensor:030]: Homeassistant Sensor 'outside_temperature'
[19:31:10][C][homeassistant.sensor:030]:   State Class: ''
[19:31:10][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[19:31:10][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[19:31:10][C][homeassistant.sensor:031]:   Entity ID: 'sensor.office_outside_temp_2'
[19:31:10][C][homeassistant.sensor:030]: Homeassistant Sensor 'inside_temperature'
[19:31:10][C][homeassistant.sensor:030]:   State Class: ''
[19:31:10][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[19:31:10][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[19:31:10][C][homeassistant.sensor:031]:   Entity ID: 'sensor.big_room_temperature'
[19:31:11][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:31:11][W][component:215]: Components should block for at most 20-30ms.
[19:31:12][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:31:12][W][component:215]: Components should block for at most 20-30ms.
[19:31:13][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:31:13][W][component:215]: Components should block for at most 20-30ms.
[19:31:16][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:31:16][W][component:215]: Components should block for at most 20-30ms.
[19:31:16][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:31:16][W][component:215]: Components should block for at most 20-30ms.
[19:31:16][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:31:16][W][component:215]: Components should block for at most 20-30ms.
[19:31:17][W][component:214]: Component display took a long time for an operation (0.11 s).
[19:31:17][W][component:215]: Components should block for at most 20-30ms.
```

illigtr, 2024-01-11:
This is probably related to a ticket I opened as well. See issue 5324.

nielsnl68, 2024-01-11:
hello @Grey-Lancaster could you do me a favor and test me your display with https://esphome.io/components/display/#troubleshooting
And show the screenshot.

Grey-Lancaster, 2024-01-11:
@nielsnl68 Thanks, I will try as soon as I can.  I unplugged all the wires from my esp32 to plug a d1 mini in for something else.  I will try tonight.
@illigtr No, the display did what I told it to had a background color, word (text) it just never gave the info for my sensor ,like my oled display does.  It does show the time though

This is from the oled I do not get it with the lilon tft
[14:19:43][D][homeassistant.sensor:024]: 'sensor.office_outside_temp_2': Got state 56.10
[14:19:43][D][sensor:093]: 'outside_temperature': Sending state 56.10000  with 1 decimals of accuracy

Grey-Lancaster, 2024-01-12:
The problem is not the display it works though it has all those connection erros.  It does not advance to reading the sensors
[image]

[image]

In should show my inside temp.  This never happens with the ili9xxx, this from oled

[image]

clydebarrow, 2024-01-14:
> I do not get it with the lilon tft
> [14:19:43][D][homeassistant.sensor:024]: 'sensor.office_outside_temp_2': Got state 56.10
> [14:19:43][D][sensor:093]: 'outside_temperature': Sending state 56.10000 with 1 decimals of accuracy

So the problem is that you are not getting data from HomeAssistant. Since your display lambda has a test for `has_state()` if there is no data, nothing will be displayed.

There should be a line in the log like this:

```
[21:25:45][D][api:102]: Accepted 192.168.0.107
```
This indicates that HomeAssistant has properly connected to the device. That is probably not happening.

Check in HomeAssistant that your device is properly registered, if necessary remove it and re-add, check the entity ids. The issue is almost certainly unrelated to the display driver, it's just related to your device not talking properly to HomeAssistant.

Grey-Lancaster, 2024-01-15:
Time is actually working so it must be connected.  It is the temp sensors that are not coming in?

Grey-Lancaster, 2024-01-15:
Well, I be Darn.  Obviously I had been fingering with this in espphome with no luck.  All of a sudden I saw a new device notification.  When I went through that and added it to Esphome it worked !?!?!

[image]

Thanks for looking at this,
Grey

Grey-Lancaster, 2024-01-15:
Sad though, this Lolin TFT display works with esp8266 d1 mini's as well as esp32 Mini's which are plug n play. The espHome ili9xxx driver requires a board with psram

closed 2024-01-15 by Grey-Lancaster

clydebarrow, 2024-01-15:
> The espHome ili9xxx driver requires a board with psram

That will no longer be a requirement with LVGL as it can draw into buffers less than full screen size. Still WIP but will get there before too long.

Grey-Lancaster, 2024-01-15:
@clydebarrow That's great as having a board that just plugs into the Lilon is much better than all these lose jumper wires !