esphome/issues #5362: OTA on_state_change misses a semicolon after "lambda:"-keyword

Michael-Beubl, 2024-01-11:
### The problem

Hi everyone,

on the esphome-website under the OTA-Documentation (https://esphome.io/components/ota) i found a not working example text.
In the example text to the on_state_change function is a semicolon missing at the end of line with the "lambda:" keyword.
In the current form i wasnt able to compile my .yaml file with this example inside of it.
With a semicolon at the end of this line everything worked fine.

Thanks, have a nice day.

### Which version of ESPHome has the issue?

2023.12.05

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

https://esphome.io/components/ota

### What platform are you using?

ESP32-IDF

### Board

ESP32-C3-SuperMini (esp32-c3-devkitm-1

### Component causing the issue

OTA state_on_change

### Example YAML snippet

```yaml
ota:
  on_state_change:
    then:
      - if:
          condition:
            lambda: return state == ota::OTA_STARTED
          then:
            - logger.log: "OTA start"
```

### Anything in the logs that might be useful for us?

```txt
***\esp32-c3-test>esphome run esp32-c3-test-1.yaml
INFO ESPHome 2023.12.5
INFO Reading configuration esp32-c3-test-1.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing esp32-c3-test-1 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
[... 86 lines cut ...]
Compiling .pioenvs\esp32-c3-test-1\app_trace\host_file_io.o
Compiling .pioenvs\esp32-c3-test-1\app_trace\gcov\gcov_rtio.o
Archiving .pioenvs\esp32-c3-test-1\esp-idf\app_trace\libapp_trace.a
Compiling .pioenvs\esp32-c3-test-1\app_update\esp_ota_ops.o
Compiling .pioenvs\esp32-c3-test-1\app_update\esp_app_desc.o
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_common.o
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_common_loader.o
Archiving .pioenvs\esp32-c3-test-1\esp-idf\app_update\libapp_update.a
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_clock_init.o
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_flash.o
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_mem.o
<unicode string>: In lambda function:
<unicode string>:75:39: error: expected ';' before '}' token
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_random.o
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_random_esp32c3.o
Compiling .pioenvs\esp32-c3-test-1\bootloader_support\src\bootloader_utility.o
*** [.pioenvs\esp32-c3-test-1\src\main.o] Error 1
============================================= [FAILED] Took 149.64 seconds =============================================
```

### Additional information

As a beginner with a not so small .yaml-file this took me quite some time to figure out.

Have a nice Day.

ssieb, 2024-01-11:
You could make a PR for the docs.

M4dMikel, 2024-01-11:
Thanks for your Advice.

I tried so here:
https://github.com/esphome/esphome-docs/pull/3530

As a first-timer i appreciate your Feedback.

closed 2024-01-11 by ssieb