esphome/issues #5363: Issue with with ADC and ESP32-C6

lucas2lab, 2024-01-11:
### The problem

Impossible to launch the build using the ADC component on an ESP32-C6. The exact same pin using a digital read is compiled, but using the ADC module I get the error as displayed in the log. It seems the ADC for the C6 is wrongly declared in esphome or I may setup it wrongly. Could you please assist me ?

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

esp32-c6

### Component causing the issue

adc

### Example YAML snippet

```yaml
esphome:
  name: esp32-c6-adc-test 

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  framework:
    type: esp-idf
    version: 5.1.2
    platform_version: 6.5.0

# Enable logging
logger:

ota:
  password: "123456"

#binary_sensor:
#  - name: "Binary 1"
#    platform: gpio
#    pin: GPIO0

sensor:
  - name: "ADC 1"
    platform: adc
    pin: GPIO0
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
PS C:\Users\User\Projects\ESPHomeTest> esphome compile .\helloworld.yaml
INFO ESPHome 2023.12.5
INFO Reading configuration .\helloworld.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\User\anaconda3\envs\ESPHome\Scripts\esphome.exe\__main__.py", line 7, in <module>
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\__main__.py", line 1019, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\config.py", line 1066, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\config.py", line 920, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\config.py", line 908, in _load_config
    result = validate_config(config, command_line_substitutions)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\config.py", line 836, in validate_config
    result.run_validation_steps()
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\config.py", line 143, in run_validation_steps
    task.step.run(self)
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\config.py", line 717, in run
    self.comp.final_validate_schema(conf)
  File "C:\Users\User\anaconda3\envs\ESPHome\Lib\site-packages\esphome\components\adc\sensor.py", line 41, in final_validate_config
    in ESP32_VARIANT_ADC2_PIN_TO_CHANNEL[variant]
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'ESP32C6'
PS C:\Users\User\Projects\ESPHomeTest>
```

ssieb, 2024-01-11:
This is fixed in dev, but even after you get past this, it still won't compile due to other issues.  Support for the C6 hasn't really been added yet.

lucas2lab, 2024-01-12:
Thank you very much. Indeed the ADC compilation is not passing.

ssieb, 2024-01-12:
I was going to leave this open since it does need to be fixed at some point.

rgnyldz, 2024-05-05:
I get a similar error only about the adc2_channel_t . I'm new and guessing that it wants us to use adc2 ?

```
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/multisensor-c6/src/main.o
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from src/esphome/components/adc/adc_sensor.h:11:
/data/cache/platformio/packages/framework-espidf/components/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
src/esphome/components/adc/adc_sensor.h:26:21: error: 'adc2_channel_t' has not been declared
   26 |   void set_channel2(adc2_channel_t channel) {
      |                     ^~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.h:63:3: error: 'adc2_channel_t' does not name a type; did you mean 'adc_channel_t'?
   63 |   adc2_channel_t channel2_{ADC2_CHANNEL_MAX};
      |   ^~~~~~~~~~~~~~
      |   adc_channel_t
src/esphome/components/adc/adc_sensor.h:66:3: error: 'esp_adc_cal_characteristics_t' does not name a type
   66 |   esp_adc_cal_characteristics_t cal_characteristics_[SOC_ADC_ATTEN_NUM] = {};
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.h: In member function 'void esphome::adc::ADCSensor::set_channel1(adc1_channel_t)':
src/esphome/components/adc/adc_sensor.h:24:5: error: 'channel2_' was not declared in this scope; did you mean 'channel1_'?
   24 |     channel2_ = ADC2_CHANNEL_MAX;
      |     ^~~~~~~~~
      |     channel1_
src/esphome/components/adc/adc_sensor.h:24:17: error: 'ADC2_CHANNEL_MAX' was not declared in this scope; did you mean 'ADC1_CHANNEL_MAX'?
   24 |     channel2_ = ADC2_CHANNEL_MAX;
      |                 ^~~~~~~~~~~~~~~~
      |                 ADC1_CHANNEL_MAX
src/esphome/components/adc/adc_sensor.h: In member function 'void esphome::adc::ADCSensor::set_channel2(int)':
src/esphome/components/adc/adc_sensor.h:27:5: error: 'channel2_' was not declared in this scope; did you mean 'channel1_'?
   27 |     channel2_ = channel;
      |     ^~~~~~~~~
      |     channel1_
*** [.pioenvs/multisensor-c6/src/main.o] Error 1
========================== [FAILED] Took 3.88 seconds ==========================
```

Here is my yaml file (sorry its long duo to LD2411s);

```
substitutions:
  name: multisensor-c6
  friendly_name: Multisensor C6
  uart_tx_pin: GPIO17
  uart_rx_pin: GPIO16

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  includes:
    - LD2411s.h
  project:
    name: esphome.web
    version: '1.0'
  on_boot:
    priority: 250
    then:
      - uart.write: [0xFD, 0xFC, 0xFB, 0xFA, 0x04, 0x00, 0xFF, 0x00, 0x01, 0x00, 0x04, 0x03, 0x02, 0x01]
      - uart.write: [0xFD, 0xFC, 0xFB, 0xFA, 0x02, 0x00, 0x73, 0x00, 0x04, 0x03, 0x02, 0x01]
      - uart.write: [0xFD, 0xFC, 0xFB, 0xFA, 0x04, 0x00, 0xFE, 0x00, 0x01, 0x00, 0x04, 0x03, 0x02, 0x01]

esp32:
[... 183 lines cut ...]
          data: !lambda
                  int minmot = id(minmotion).state;
                  int maxmot = id(maxmotion).state;
                  int minpr = id(minpresence).state;
                  int maxpr = id(maxpresence).state;
                  int unocctm = id(unocc_time).state;
                  uint16_t minmvalue = static_cast<uint16_t>(minmot);
                  uint8_t minmhighByte = static_cast<uint8_t>(minmvalue >> 8);
                  uint8_t minmlowByte = static_cast<uint8_t>(minmvalue & 0xFF);
                  uint16_t maxmvalue = static_cast<uint16_t>(maxmot);
                  uint8_t maxmhighByte = static_cast<uint8_t>(maxmvalue >> 8);
                  uint8_t maxmlowByte = static_cast<uint8_t>(maxmvalue & 0xFF);
                  uint16_t minpvalue = static_cast<uint16_t>(minpr);
                  uint8_t minphighByte = static_cast<uint8_t>(minpvalue >> 8);
                  uint8_t minplowByte = static_cast<uint8_t>(minpvalue & 0xFF);
                  uint16_t maxpvalue = static_cast<uint16_t>(maxpr);
                  uint8_t maxphighByte = static_cast<uint8_t>(maxpvalue >> 8);
                  uint8_t maxplowByte = static_cast<uint8_t>(maxpvalue & 0xFF);
                  uint16_t unoccvalue = static_cast<uint16_t>(unocctm * 10);
                  uint8_t unocchighByte = static_cast<uint8_t>(unoccvalue >> 8);
                  uint8_t unocclowByte = static_cast<uint8_t>(unoccvalue & 0xFF);
                  std::vector<uint8_t> data = {0xFD, 0xFC, 0xFB, 0xFA, 0x20, 0x00, 0x67, 0x00, 0x00, 0x00, maxmlowByte, maxmhighByte, 0x00, 0x00, 0x01, 0x00, minmlowByte, minmhighByte, 0x00, 0x00, 0x02, 0x00, maxplowByte, maxphighByte, 0x00, 0x00, 0x03, 0x00, minplowByte, minphighByte, 0x00, 0x00, 0x04, 0x00, unocclowByte, unocchighByte, 0x00, 0x00, 0x04, 0x03, 0x02, 0x01};
                  return data;
      - uart.write: [0xFD, 0xFC, 0xFB, 0xFA, 0x04, 0x00, 0xFE, 0x00, 0x01, 0x00, 0x04, 0x03, 0x02, 0x01]
```

QuickJack, 2024-06-06:
I am facing the same issue.

The datasheet of ESP32-C6 (https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf) shows in chapter 2.9.9 that it has only one ADC. However, the original ESP32 (https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) shows in chapter 2.2 that it has two.

There is an issue when using WiFi together with ADC2 on the original ESP32 (s. https://www.bitfoic.com/components/how-to-resolve-the-wifi-and-adc2-sharing-dilemma?id=206). This is also described in https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf in chapter 29.3.1.

jharris4, 2024-06-14:
I’m consistently getting the adc2_channel_t error as well if I try to add a sensor with platform adc. Compiles and works fine if I don’t use adc.

WouterJN, 2024-07-30:
When will this be solved? Are there any work arrounds that still work with 2024.7?

My current code;

`esphome:
  name: livingroom-plant-watering-unit
  friendly_name: LivingRoom Plant Watering Unit

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 8MB
  framework:
    type: esp-idf
    version: 5.2.2
    platform_version: 6.6.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_USE_MINIMAL_MDNS: y

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  - platform: esphome
    password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Livingroom-Plant-Watering-Unit"
    password: "xxxx"

captive_portal:

sensor:
  - platform: adc
    pin: GPIO01
    name: Soil Moisture Value
    attenuation: 11db
    id:  soil_moisture
    accuracy_decimals: 0
    unit_of_measurement: "%"
    update_interval: 10s
    icon: mdi:flower
    filters:
      - calibrate_linear:
        - 1.795 -> 0.0
        - 1.43 -> 100.0
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15
      - median:
          window_size: 15
          send_every: 7
          send_first_at: 3
`

I also tried by adding the following code but that also doesnt work anylonger;

`external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: adc
    components: [ adc ]
    refresh: 1min
`

It would be great if you could help @ssieb

jkremlacek, 2024-08-01:
I quickly checked sourced and if I understand the issue correctly, current adc support in esphome is based on deprecated solution provided by esp-idf which is missing support for C6 and as it is deprecated it will most likely never be added.

@ssieb Is there any rough time range when migration to ADC oneshot would be done on esphome side?

(normally I would offer my help, but I haven't touched C++ seriously for 6+ years and I lack knowledge in both deprecated and new solution of esp-idf - I tried bruteforce fixing it, but after 2hrs I got into deadend)

Source (deprecation info):
https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/migration-guides/release-5.x/5.0/peripherals.html#adc

lboue, 2024-08-05:
> I quickly checked sources and if I understand the issue correctly, current adc support in esphome is based on deprecated solution provided by esp-idf which is missing support for C6 and as it is deprecated it will most likely never be added.
> 
> @ssieb Is there any rough time range when migration to ADC oneshot would be done on esphome side?
> 
> (normally I would offer my help, but I haven't touched C++ seriously for 6+ years and I lack knowledge in both deprecated and new solution of esp-idf - I tried bruteforce fixing it, but after 2hrs I got into deadend)
> 
> Source (deprecation info): https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/migration-guides/release-5.x/5.0/peripherals.html#adc

For ADC Oneshot Mode Driver:

**Doc**
[ESP-IDF Programming Guide: Analog to Digital Converter (ADC) Oneshot Mode Driver](https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32c6/api-reference/peripherals/adc_oneshot.html)

**Examples**

Here are the examples for the ADC module with ESP-IDF v5.3
* [ADC Single Read exemple](https://github.com/espressif/esp-idf/tree/v5.3/examples/peripherals/adc/oneshot_read)

lboue, 2024-08-05:
@lucas2lab @jkremlacek
I've spent several hours coding and I have a static example that works but is limited to GPIO0.
Could you test it with the following configuration?

**ESPHome config**

```
external_components:
  - source:
      # https://github.com/lboue/esphome/blob/adc_oneshot/esphome/components/adc
      type: git
      url: https://github.com/lboue/esphome
      ref: adc_oneshot
      #ref: ESP32_C6_ADC2
    components: [ adc ]
    refresh: 0s

sensor:
  - name: "ADC1_CH0 GPIO0"
    platform: adc
    pin: GPIO0
    update_interval: 15s
```

jkremlacek, 2024-08-07:
Worked like a charm for me. 

I tried it with a voltage divider using 2 47k resistors on 5v input and ignoring the innacuracy of the ADC within the ESP (about 0.1 V) it measured correctly. 

---

Note for others, in case you would be replicating my usecase you have to change the sensor definition in order to adjust the measurement for the voltage divider using `filters` and `multiply`:

```
sensor:
  - name: "ADC1_CH0 GPIO0"
    platform: adc
    pin: GPIO0
    update_interval: 15s
    filters:
      - multiply: 2
```

dcolens, 2024-08-13:
@lboue thanks for the fix, I tried it too with a FireBeetle 2 ESP32-C6 and it is working fine, any chance to get a patch that works for other GPIOs :-D ?

naknut, 2024-08-25:
@lboue I could not get you fix to work for me. This is my current configuration:

```
esphome:
  name: soil-mosture-lab
  friendly_name: Soil Mosture Lab

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 8MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: "5.2.1"
    platform_version: 6.6.0
    sdkconfig_options:
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_USE_MINIMAL_MDNS: y
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y

external_components:
  - source:
      # https://github.com/lboue/esphome/blob/adc_oneshot/esphome/components/adc
      type: git
      url: https://github.com/lboue/esphome
      ref: adc_oneshot
    components: [ adc ]
    refresh: 0s

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "msi5g7riSsnTaf19fahduGUr9q8CzdIBaLA2K2CpB/w="

ota:
  - platform: esphome
    password: "515825743d9cdf6e99109b287a490fa1"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Soil-Mosture-Lab"
    password: "yewabOCCJh7q"

captive_portal:

sensor:
  - name: "ADC1_CH0 GPIO0"
    platform: adc
    pin: GPIO0
    update_interval: 15s
```

and this is the output I get when flashing:

```
INFO ESPHome 2024.8.0
INFO Reading configuration /config/esphome/soil-mosture-lab.yaml...
INFO Updating https://github.com/lboue/esphome@adc_oneshot
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING This board is unknown. Make sure the chosen chip component is correct.
[... 19 lines cut ...]
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/soil-mosture-lab/src/esphome/components/api/api_connection.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()':
/data/cache/platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/13.2.0/bits/basic_string.h:792:(.text._ZN7esphome3api13APIConnection16send_sensor_infoEPNS_6sensor6SensorE+0x134): undefined reference to `esphome::sensor::Sensor::get_accuracy_decimals()'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /data/cache/platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/13.2.0/bits/basic_string.h:792:(.text._ZN7esphome3api13APIConnection16send_sensor_infoEPNS_6sensor6SensorE+0x160): undefined reference to `esphome::sensor::Sensor::get_state_class()'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/soil-mosture-lab/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/data/build/soil-mosture-lab/src/esphome/core/controller.cpp:29:(.text._ZN7esphome10Controller16setup_controllerEb+0x60): undefined reference to `esphome::sensor::Sensor::add_on_state_callback(std::function<void (float)>&&)'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/soil-mosture-lab/src/main.o: in function `esphome::Component::set_component_source(char const*)':
/data/build/soil-mosture-lab/src/esphome/core/component.h:149:(.text._Z5setupv+0x510): undefined reference to `esphome::sensor::Sensor::Sensor()'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/soil-mosture-lab/src/main.o: in function `esphome::api::APIServer* esphome::Application::register_component<esphome::api::APIServer>(esphome::api::APIServer*)':
/data/build/soil-mosture-lab/src/esphome/core/application.h:191:(.text._Z5setupv+0x524): undefined reference to `vtable for esphome::adc::ADCSensor'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /data/build/soil-mosture-lab/src/esphome/core/application.h:191:(.text._Z5setupv+0x528): undefined reference to `vtable for esphome::adc::ADCSensor'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/soil-mosture-lab/src/main.o: in function `setup()':
/data/build/soil-mosture-lab/src/main.cpp:144:(.text._Z5setupv+0x530): undefined reference to `vtable for esphome::adc::ADCSensor'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /data/build/soil-mosture-lab/src/main.cpp:144:(.text._Z5setupv+0x534): undefined reference to `vtable for esphome::adc::ADCSensor'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /data/build/soil-mosture-lab/src/main.cpp:145:(.text._Z5setupv+0x53c): undefined reference to `vtable for esphome::adc::ADCSensor'
[... cut ...]
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/soil-mosture-lab/src/main.o: in function `esphome::adc::ADCSensor::ADCSensor()':
/data/build/soil-mosture-lab/src/esphome/components/adc/adc_sensor.h:36:(.text._Z5setupv+0x648): undefined reference to `esphome::adc::ADCSensor::set_sample_count(unsigned char)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/soil-mosture-lab/firmware.elf] Error 1
========================= [FAILED] Took 23.75 seconds =========================
```

eugene-polyakov, 2024-08-31:
Hi folks - based upon @lboue work, added GPIO selection and re-implemented as a custom component, fits my purposes better.

Still pretty raw, didn't touch ADC2 -- it's not available with WiFi on anyways I believe so please stick to ADC1 pins for now.

[](https://github.com/eugene-polyakov/esphome_custom_components/tree/main/adc_oneshot)
working example - [](https://github.com/eugene-polyakov/esphome_custom_components/blob/main/oneshot.yaml)

Luis-Sarabando, 2024-10-02:
Can confirm the above solution worked for me on a M5nanoC6 (esp32-c6-devkitc-1), so adc is working!

```esp32:
  board: esp32-c6-devkitc-1
  flash_size: 8MB
  variant: esp32c6
  framework:
    type: esp-idf
#Can be dev, latest, recommended, or check https://github.com/espressif/esp-idf/releases
    version: "5.3.1"
    #version: "5.3.0"
#check platform_version @ https://github.com/platformio/platform-espressif32/releases/
    platform_version: 6.9.0
    #platform_version: 6.8.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y

#This is to make sure adc compiles or else it will throw an error
external_components:
  - source:
      # https://github.com/lboue/esphome/blob/adc_oneshot/esphome/components/adc
      type: git
      url: https://github.com/lboue/esphome
      ref: adc_oneshot
    components: [ adc ]
    refresh: 0s
    
    sensor:
  - name: "ADC 1"
    platform: adc
    pin: GPIO5
    update_interval: 5s
    ```

Luis-Sarabando, 2024-10-11:
Update to my post above, adc works but the values don't match up, they dont change, which leads me to belive its not actually working @kdpuvvadi 

As ill inject a voltage and the measured in esp doesnt even change.

PhracturedBlue, 2024-10-12:
I put together a working ADC for the ESP32-C6 here: https://github.com/PhracturedBlue/c6_adc

It is just a stop-gap, and it doesn't support features like auto-attenuation or averaging multiple samples per iteration.  but it seems to work for me on all chanels (0-6).  The default update rate is 60sec, but that can be changed in the usual manner.

The config would look something like:
```
esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: "5.3.0"
    platform_version: 6.8.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y

external_components:
  - source: github://PhracturedBlue/c6_adc

sensor:
  - platform: c6_adc
    name: adc5
    pin: GPIO5
    attenuation: 0dB  # optional.  defaults to 12dB
```

@Luis-Sarabando FYI

kdpuvvadi, 2024-10-13:
There seems to be issue with your component @PhracturedBlue 

```
In file included from src/esphome/components/c6_adc/c6_adc.cpp:2:
src/esphome/components/c6_adc/c6_adc.h:3:10: fatal error: esphome/components/voltage_sampler/voltage_sampler.h: No such file or directory
    3 | #include "esphome/components/voltage_sampler/voltage_sampler.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\test\src\esphome\components\c6_adc\c6_adc.cpp.o] Error 1
```

PhracturedBlue, 2024-10-13:
what version of esphoome are you using?  I used  2024.8.3 with the exact configuration above, and it builds fine for me

kdpuvvadi, 2024-10-14:
I was on `2024.9.2`. Downgraded to `2024.8.3`. Getting the same error. 

Removed the esphome completetly and installed 9.2 and still same issue. Just for the sake of it tried on windows workstation (also wsl2), debian 12 box and macos Sequoia. Getting same error on all.

PhracturedBlue, 2024-10-14:
Sorry about that.  it was due to a change I made when adding support for the ct_clamp.  it didn't show up when the ct_clamp was also included.  The issue should be fixed now.  you need to either remove your external_components directory, or add `refresh: 0s` to the c6_adc external-component in your yaml (see exmaple above) to force a git-pull.  I don't recommend leaving `refresh: 0s` in your production yaml file as it slows down the validator.

leon-v, 2024-11-29:
In case anyone finds this while trying to get a C2 working, I have re-added the C2 pins in a new fork:
```
external_components:
  - source:
      # https://github.com/leon-v/esphome/blob/adc_one_shot_C2/esphome/components/adc
      type: git
      url: https://github.com/leon-v/esphome
      ref: adc_one_shot_C2
    components: [ adc ]
    refresh: 0s
```

Seems to be working OK.

Full yaml:
```
esphome:
  name: esp32c2-test
  friendly_name: esp32c2-test

esp32:
  board: esp32-c2-devkitm-1
  variant: esp32c2
  flash_size: 2MB
  framework:
    type: esp-idf
    version: '5.3.1'
    platform_version: 'https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53'
    sdkconfig_options:
      CONFIG_XTAL_FREQ_26: y

#This is to make sure adc compiles or else it will throw an error
external_components:
  - source:
      # https://github.com/leon-v/esphome/blob/adc_one_shot_C2/esphome/components/adc
      type: git
      url: https://github.com/leon-v/esphome
      ref: adc_one_shot_C2
    components: [ adc ]
    refresh: 0s

logger:
  level: DEBUG

sensor:

  - platform: adc
    pin: GPIO0
    name: "ADC 1"
    id: adc_1
    update_interval: 2s

  # - platform: adc
  #   pin: GPIO2
  #   name: "ADC 2"
  #   id: adc_2
  #   update_interval: 60s
```

mattmeye, 2024-12-22:
> @lboue thanks for the fix, I tried it too with a FireBeetle 2 ESP32-C6 and it is working fine, any chance to get a patch that works for other GPIOs :-D ?

@dcolens and @kdpuvvadi 
Do you have a working solution for the DF Robot Firebeetle 2 ESP32-C6? If so, what does it look like? Unfortunately, I get an incorrect voltage number. I suspect a problem with the attenuation, which lboue's solution not support.

bouttier, 2025-03-13:
I can successfully read DFRobot Firebeetle 2 ESP32-C6 battery voltage thanks to [#7942](https://github.com/esphome/esphome/pull/7942):
```
external_components:
  - source: github://pr#7942
    components: [ "adc" ]

sensors:
  - platform: adc
    pin: GPIO0
    id: battery_voltage
    name: "Battery voltage"
    attenuation: 12db
    samples: 20
    filters:
     - multiply: 2  # 2×1MΩ voltage divider
```

eric-schmieg, 2025-04-12:
I was able to get 2 ADC sensors polling with a seeed c6 with [PhracturedBlue/c6_adc](github://PhracturedBlue/c6_adc).

Thanks for making this @PhracturedBlue

closed 2025-12-28 by ssieb