esphome/issues #5364: fastled_clockless won't compile

psitech, 2024-01-11:
### The problem

A fastled_clockless sketch won't compile without errors.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.1.2

### What platform are you using?

ESP8266

### Board

ESP32-C3 Supermini

### Component causing the issue

light

### Example YAML snippet

```yaml
esphome:
  name: rgbledstrip
  friendly_name: RGB LED strip

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: "boFycI7HZ13/b0buThFBC2nrYHQ/upyh/eJeR2d4Xkc="

ota:
  password: "d215d40850361e4bdd3ba49dfdcc7d3c"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

# ---------- this compiles fine ----------------------
# light:
#  - platform: neopixelbus
#    type: GRB
#    variant: WS2812
#    pin: GPIO6
#    num_leds: 144
#    name: "LED strip"
#    default_transition_length: '0s'
#    effects:
[... 124 lines cut ...]
                                                    ^
In file included from .piolibdeps/rgbledstrip/FastLED/platforms/esp/32/clockless_rmt_esp32.h:97,
                 from .piolibdeps/rgbledstrip/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from .piolibdeps/rgbledstrip/FastLED/platforms.h:36,
                 from .piolibdeps/rgbledstrip/FastLED/FastLED.h:52,
                 from .piolibdeps/rgbledstrip/FastLED/colorutils.cpp:7:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/driver/include/driver/rmt.h:930:11: note: declared here
 esp_err_t rmt_set_pin(rmt_channel_t channel, rmt_mode_t mode, gpio_num_t gpio_num)
           ^~~~~~~~~~~
*** [.pioenvs/rgbledstrip/libb07/FastLED/colorutils.cpp.o] Error 1
========================= [FAILED] Took 15.85 seconds =========================
```

### Additional information

Compile errors from log above:
.piolibdeps/rgbledstrip/FastLED/platforms/esp/32/fastpin_esp32.h:16:17: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1_w1ts'; did you mean 'out_w1ts'?

.piolibdeps/rgbledstrip/FastLED/platforms/esp/32/fastpin_esp32.h:21:17: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1_w1tc'; did you mean 'out_w1tc'?

.piolibdeps/rgbledstrip/FastLED/platforms/esp/32/fastpin_esp32.h:26:17: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1'; did you mean 'out'?

.piolibdeps/rgbledstrip/FastLED/platforms/esp/32/fastpin_esp32.h:32:31: error: no match for 'operator^=' (operand types are 'volatile gpio_dev_s::<unnamed union>' and 'long unsigned int')

.piolibdeps/rgbledstrip/FastLED/platforms/esp/32/fastpin_esp32.h:51:34: error: cannot convert 'volatile gpio_dev_s::<unnamed union>*' to '_ESPPIN<PIN, MASK>::port_ptr_t' {aka 'volatile long unsigned int*'} in return

ssieb, 2024-01-11:
It needs a newer fastled library.  There are already issues open for this.  But you should use the RMT component anyway.

closed 2024-01-11 by ssieb

psitech, 2024-01-11:
Thanks for your reply. A newer FastLED lib seems obvious.
Wasn't aware of using the RMT component as preferred - I will look into that.

psitech, 2024-01-12:
RMT component works fine.
For reference here is the working code:

```
esphome:
  name: rgbledstrip
  friendly_name: RGB LED strip

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: "boFycI7HZ13/b0buThFBC2nrYHQ/upyh/eJeR2d4Xkc="

ota:
  password: "d215d40850361e4bdd3ba49dfdcc7d3c"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: GPIO6
    num_leds: 144
    rmt_channel: 0
    chipset: ws2812
    name: "LED strip"
    effects:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 144
```