esphome/issues #5365: sen5x does not report negative Temperatures

helmarw, 2024-01-12:
### The problem

whenever the enviromental sensor reaches temperatures below 0 Celsius the reported value jumps to +327 Celsius 
anybody else has this problem ?

[image]

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.1.2

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

sen54

### Example YAML snippet

```yaml
- platform: sen5x
    id: sen54
    pm_1_0:
      name: " PM <1µm Weight concentration"
      id: pm_1_0
      accuracy_decimals: 1
    pm_2_5:
      name: " PM <2.5µm Weight concentration"
      id: pm_2_5
      accuracy_decimals: 1
    pm_4_0:
      name: " PM <4µm Weight concentration"
      id: pm_4_0
      accuracy_decimals: 1
    pm_10_0:
      name: " PM <10µm Weight concentration"
      id: pm_10_0
      accuracy_decimals: 1
    temperature:
      name: "Temperature"
      accuracy_decimals: 1
    humidity:
      name: "Humidity"
      accuracy_decimals: 0
    voc:
      name: "VOC"
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    nox:
      name: "NOx"
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    temperature_compensation:
      offset: 0
      normalized_offset_slope: 0
      time_constant: 0
    acceleration_mode: low
    store_baseline: true
    address: 0x69
    update_interval: 10s
```

ssieb, 2024-01-12:
Please try the PR.

closed 2024-01-12 by kbx81

helmarw, 2024-01-12:
wow thnx, will give it a try :)