esphome/issues #5366: Entity name is doubled

mbo18, 2024-01-12:
### The problem

I've added several new devices to ESPHome recently and when in the configuration name = none, the entity name in HA is doubled.

From the doc (here: https://esphome.io/components/switch/): 
>Note
If you have a [friendly_name](https://esphome.io/components/esphome#esphome-configuration-variables) set for your device and you want the switch to use that name, you can set name: None.

With the Yaml below, this is the entities created in HA:
When name is None (example with the switch): `switch.bureau_plug_bureau_plug` when I expect `switch.bureau_plug`
When name is not None (example with the binary sensor): binary_sensor.bureau_plug_status which is fine

[image]

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.2

### What platform are you using?

ESP8266

### Board

All

### Example YAML snippet

```yaml
esphome:
  name: bureau-plug
  friendly_name: "Bureau Plug"

esp8266:
  board: esp01_1m

binary_sensor:
  - platform: status
    name: "Status"

switch:
  - platform: gpio
    id: relay
    name: None
    device_class: outlet
    pin: GPIO12
    on_turn_on:
      - light.turn_on: state_led
    on_turn_off:
      - light.turn_off: state_led
```

kaechele, 2024-01-18:
Same here.

Possibly related:
- https://github.com/esphome/issues/issues/4848
- https://community.home-assistant.io/t/how-is-friendly-name-supposed-to-behave/542570

mbo18, 2024-05-18:
Not stale