esphome/issues #5369: WaveShare 4.20 V2 not displaying

RacoJo, 2024-01-12:
### The problem

Hello,
I wanted to know is someone achieve to display from a 4.20in V2 (it's writtent v2 in a sticker on the back and rev2.2).

Because it's working well when I upload an example from WaveShare with arduino IDE, but not with esphome.

Is this because 4.20in-bv2 is not my version ? I tried :  

4.20in-bv2 
4.20in-bv2 with busy inverted : true
4.20in
4.20in with busy inverted : true

So many different pins configuration also, but it's not working...

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12

### What platform are you using?

ESP8266

### Board

d1 mini

### Example YAML snippet

```yaml
'''
esphome:
  name: einktest
  friendly_name: einktest

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dashboard-Eink1 Fallback Hotspot"
    password: 

captive_portal:

spi:
  clk_pin: D5
  mosi_pin: D7

display:
  - platform: waveshare_epaper
    cs_pin: D8
    dc_pin: D3
    update_interval: 00s
    busy_pin: 
      number: D2
      inverted: true
    reset_pin: D4
    model: 4.20in
    reset_duration: 2ms
    lambda: |-
      it.rectangle(140,0,140,150);
'''
```

MrMDavidson, 2024-02-08:
Experiencing the same issue - if I change the model to be 2.90inv2 I get rendering as expected in the subsection of the display that the 2.90inv2 would take up (in appropriate rotation) with the rest of the display corrupted.

MrMDavidson, 2024-02-10:
Hey @RacoJo I've got a draft PR open that may resolve this: https://github.com/esphome/esphome/pull/6209 It seems to work for my device but I cannot guarantee it's without issues or won't damage your display, etc, etc.

MChartier, 2025-08-11:
This issue persists. I'm able to use my 4.2" Waveshare e-ink display only when including @MrMDavidson's PR and setting the model to 4.20in-v2 - it doesn't work otherwise.

wissencat1, 2026-10-12:
Hello,

the issue is now present again because the epaper spi library does not account for this display and the patch from

[https://github.com/esphome/esphome/pull/6209](url)

does not compile anymore correctly and the suggested solution for that does not work.

its sad to see that now my display has no funktion anymore and my projekt is now on hold to find another 4.2 inch one that works for it. :(

wissencat1, 2026-10-12:
@MrMDavidson sadly your fix does not compile anymore and I can't get the Display to work with the epaper spi Library even when I use the Controller and Dimension configuration.

I am Not so Deep into the GitHub external component stuff but is there any way I can still use it? 

Right now it's Not Computing anymore sadly. Does that have to do with the closure of the Post?